openrazer/openrazer #353: Razer Blade Stealth 13.3 ripple effect issue

jonmanes, 2017-08-08:
[ripple_issue_pcaps.zip](https://github.com/terrycain/razer-drivers/files/1209348/ripple_issue_pcaps.zip)
The ripple effect does not work. As you suspected it uses the last of the single key setting the board had set. 

The ripple sngl required selecting effect / none twice before it actually went back to no effect so you may see that in the attached pcap.

jonmanes, 2017-08-08:
[RBS_ripple_windows.zip](https://github.com/terrycain/razer-drivers/files/1209402/RBS_ripple_windows.zip)
This pcap is the working feature running from the windows side.

Alkeryn, 2017-10-10:
Other than that is there any other issue with the laptop
(i just bought it but didn't got it yet)

jonmanes, 2017-10-10:
Working very well for me.

I'm running Ubuntu 17.04 if that helps.

Alkeryn, 2017-10-12:
Thanks, i receive my puppy in few days, i'm gonna put arch on it :P
That will be my first high end linux laptop so kinda hyped xD

jonmanes, 2017-10-13:
Enjoy it! I love mine

Alkeryn, 2017-10-16:
Well i got my laptop but i can't get openrazer to work polychromatic says that the device is not recognized 
though the id is on the supported list
Running arch

Alkeryn, 2017-10-16:
also modeprobe razerkbd don't find the module

z3ntu, 2017-10-16:
@Alkeryn Please post the output of `sudo dkms status`.

Alkeryn, 2017-10-16:
I fixed it, forgot to install linux-header
though since then it force my layout into qwerty instead of azerty
even though i have setxkbmap fr in my autostart

z3ntu, 2018-01-28:
@jonmanes Did you try with the newest version (v2.2.0)?

PR #514 (2018-01-30): Fix ripple effect on non-standard keyboards
Until now, ripple only worked on keyboards with a (6, 22) matrix.
Blade keyboards have different sizes (e.g. (6, 16) or (6, 25)) and they
should be supported.

~~This is not tested yet but it could work.~~

Fixes #166 and fixes #353
closed 2018-01-30 by z3ntu