openrazer/openrazer #354: Installation script error

jfeeley97, 2017-08-09:
Hi, I'm running on Linux Mint 18.2 and I can't seem to get the razer drivers installed.

I've emboldened where I think the issue is occurring but don't know where to go from here.
I've also tried sudo apt-get -f install already too.

```
sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following package was automatically installed and is no longer required:
  libfftw3-3
Use 'sudo apt autoremove' to remove it.
The following NEW packages will be installed:
  python3-razer razer-daemon razer-doc razer-kernel-modules-dkms
0 upgraded, 4 newly installed, 0 to remove and 175 not upgraded.
Need to get 0 B/114 kB of archives.
After this operation, 854 kB of additional disk space will be used.
Selecting previously unselected package python3-razer.
(Reading database ... 204589 files and directories currently installed.)
Preparing to unpack .../python3-razer_1.1.15-0ubuntu1_all.deb ...
Unpacking python3-razer (1.1.15-0ubuntu1) ...
Selecting previously unselected package razer-kernel-modules-dkms.
Preparing to unpack .../razer-kernel-modules-dkms_1.1.15-0ubuntu1_amd64.deb ...
Unpacking razer-kernel-modules-dkms (1.1.15-0ubuntu1) ...
Selecting previously unselected package razer-daemon.
Preparing to unpack .../razer-daemon_1.1.15-0ubuntu1_amd64.deb ...
Unpacking razer-daemon (1.1.15-0ubuntu1) ...
Selecting previously unselected package razer-doc.
Preparing to unpack .../razer-doc_1.1.15-0ubuntu1_all.deb ...
Unpacking razer-doc (1.1.15-0ubuntu1) ...
Processing triggers for man-db (2.7.5-1) ...
Setting up python3-razer (1.1.15-0ubuntu1) ...
Setting up razer-kernel-modules-dkms (1.1.15-0ubuntu1) ...
Removing old razer_chroma_driver-1.0.0 DKMS files...
Loading new razer_chroma_driver-1.0.0 DKMS files...
Error! DKMS tree already contains: razer_chroma_driver-1.0.0
You cannot add the same module/version combo more than once.
**dpkg: error processing package razer-kernel-modules-dkms (--configure):
 subprocess installed post-installation script returned error exit status 3**
dpkg: dependency problems prevent configuration of razer-daemon:
 razer-daemon depends on razer-kernel-modules-dkms; however:
  Package razer-kernel-modules-dkms is not configured yet.

dpkg: error processing package razer-daemon (--configure):
 dependency problems - leaving unconfigured
Setting up razer-doc (1.1.15-0ubuntu1) ...
Errors were encountered while processing:
 razer-kernel-modules-dkms
 razer-daemon
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

lah7, 2017-08-09:
An older version of the driver is still installed, this command should reveal this:

    sudo dkms status razer_chroma_driver/1.0.0

Remove all the packages to drop the dependency problems, then remove the leftover driver.

    sudo apt remove python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
    sudo dkms uninstall razer_chroma_driver/1.0.0

Check it's definitely uninstalled before installing the packages again:

    sudo dkms status razer_chroma_driver/1.0.0
    sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc

z3ntu, 2017-09-29:
Closing as there was no message since August 9th

closed 2017-09-29 by z3ntu