openrazer/openrazer #356: Razer Nostromo: keys not working

chrisvannooten, 2017-08-10:
Hello there.
When trying to use my Razer Nostromo none of the keys seem to work (but strangely enough the scroll wheel does).
I am using Xubuntu 17.04 and everything is up to date, restarted the system after installation of the drivers.
I also reconnected the device through a separate USB port to see if that fixes anything but to no avail. 
The device is definitely mounted as per the razer mount logs:
Aug 10 17:07:55 Xeon razer_mount[418]: Driver razerkbd
Aug 10 17:07:55 Xeon razer_mount[419]: Driver razerkbd
Aug 10 17:07:55 Xeon razer_mount[427]: Device_ID 0003:1532:0111.0006
Aug 10 17:07:55 Xeon razer_mount[428]: Device_ID 0003:1532:0111.0007
Aug 10 17:07:55 Xeon razer_mount[429]: Unbinding 0003:1532:0111.0006 from hid-generic
Aug 10 17:07:55 Xeon razer_mount[430]: Unbinding 0003:1532:0111.0007 from hid-generic
Aug 10 17:07:55 Xeon razer_mount[432]: Binding 0003:1532:0111.0007 to razerkbd
Aug 10 17:07:55 Xeon razer_mount[436]: Binding 0003:1532:0111.0006 to razerkbd
Aug 10 17:07:55 Xeon razer_mount[490]: Finished binding 0003:1532:0111.0007
Aug 10 17:07:55 Xeon razer_mount[491]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0111.0007/
Aug 10 17:07:55 Xeon razer_mount[497]: Finished binding 0003:1532:0111.0006
Aug 10 17:07:55 Xeon razer_mount[498]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0111.0006/
Aug 10 17:07:56 Xeon razer_mount[500]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0111.0007/
Aug 10 17:07:56 Xeon razer_mount[501]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0111.0006/

And according to the logs in ~/.local/share/openrazer/logs/razer.log, it definitely initialized the commands (ill put fewer snippets):
2017-08-10 18:54:18 | razer.device0                  | INFO     | Initialising device.0 RazerNostromo
2017-08-10 18:54:18 | razer.device0                  | INFO     | Initialising device.0 RazerNostromo
everything seemed fine after that, no errors to be found.

I added the user to the plugdev group and secureboot is disabled.

I installed both Polychromatic controller and razerCommander, both show the Nostromo but do not allow key editing of any kind, the only option I have available is changing the brightness.
Anyone know what to do?

Kind regards,
Chris Van Nooten

chrisvannooten, 2017-08-10:
Found the issue, it lies with the Polychromatic package, after purging it I could use the Nostromo just fine but after reinstall if I opened it it would immediately stop working after 1 key.
Apologies for the confusion.

closed 2017-08-10 by chrisvannooten

lah7, 2017-08-10:
I'm the author of Polychromatic, and this doesn't really make much sense to me. :confused: 

However, since it interfaces with the daemon, I'm wondering if this is triggering a mode on the device so the keybindings are under the daemon's control. AFAIK macros are being re-written in the daemon and some front-ends (like Polychromatic) do not have support for managing macros yet.

This would be the case if the keys started working again after the daemon is stopped. (`razer-daemon -s`)

chrisvannooten, 2017-08-10:
@lah7 I just installed the daily build(version 0.3.9.4-dev) instead and that one doesnt have the issue aldo again all I see in terms of options is adjusting the brightness (is it even possible to edit the keys?). 
So whatever it is, it seems to be only present in the package provided by the stable PPA.

lah7, 2017-08-10:
Thanks for the info, I'll build a new stable build.

> adjusting the brightness (is it even possible to edit the keys?).

That's all the device is "capable" of I'm afraid. :disappointed: These are the only `true` capabilities that daemon tells the front-ends:

```
"brightness": true
"serial": true
"name": true
"macro_logic": true
"type": true
"firmware_version": true
```

`macro_logic` isn't supported in Polychromatic at the moment. I think razerCommander does (with the current daemon code) though. AFAIK, @terrycain is working on rewriting macros, so support in Polychromatic would be added then.

chrisvannooten, 2017-08-10:
Ah Ill patiently wait until then.
Best of luck with the build and thank you guys for all the work you provide!