openrazer/openrazer #357: Drivers broken after any system update

ghetolay, 2017-08-12:
So I'm using polychromatic and so your daemon on Ubuntu 17.04, it's working fine until my system updates. Every time I update my system (`apt upgrade`) it breaks the drivers. On next reboot `razer-daemon` can't find my device anymore. 

The current solution is to reinstall the kernel module  `apt install --reinstall razer-kernel-modules-dkms` and after a reboot everything works fine again.

It's been going for some time so I would like to find what's the problem but I don't really know how to proceed. What can I do before/after updates to track down this bug ?

z3ntu, 2017-08-12:
1. It's not really possible to break after a non-kernel upgrade.
2. For kernel upgrades we use dkms for building the kernel module
3. What does `sudo dkms status` output?

ghetolay, 2017-08-12:
`dkms status` gave me an error : 
>Error! Could not locate dkms.conf file.

So I found out I had old anbox directories inside `/var/lib/dkms` something probably went wrong when I deleted anbox. Now I get the correct output of : 
> razer_chroma_driver, 1.0.0, 4.10.0-32-generic, x86_64: installed

I may have fixed the problem thks to you.
I'll let some system updates pass and see if it's resolved.

z3ntu, 2017-08-12:
Great!

ghetolay, 2017-08-31:
Ok severals update passed by without any problem.
There was definitively something wrong with my dkms modules and wasn't related to the razer drivers.
Thks for the help !

closed 2017-08-31 by ghetolay