openrazer/openrazer #358: dbus-user-session dependency causes issues with chrome

MannyC, 2017-08-15:
This isn't exactly the driver's fault, but at the moment on Ubuntu 16.04 installing the dbus-user-session package causes gnome-keyring not to unlock on boot, which causes Chrome to take ages to start and not unlock your passwords.  

As dbus-user-session is not a default package it's likely that the driver package will end up installing it.  If it breaks chrome it will probably appear to the user as if it was the driver package that caused the issue.

Uninstalling dbus-user-session "fixes" the issue, but I've yet to test whether razer-drivers will continue to work (currently uninstalled on my system).

This is probably pretty common at the moment, so may be worth documenting on the website, or at least here in this issue.

gnome-keyring issue tracker has more info here: https://bugs.launchpad.net/ubuntu/+source/gnome-keyring/+bug/1689825

EricMCarroll, 2017-08-23:
I can confirm that removing razer-daemon, polychromatic and dbus-user-session resolves the gnome-keyring and chrome issue in 16.04.3.

Right now removing dbus-user-session forces the removal of its dependencies, namely razer-daemon and polychromatic.

z3ntu, 2017-08-23:
That's how much I love Ubuntu. Having millions of bug reports that never get any attention from people that fix stuff... Unfortunately we can't really do much if the distro is broken :/

aldencolerain, 2017-08-25:
@z3ntu I'm in the same boat trying to run Ubuntu and the razor drivers.  Is dbus-user-session a hard dependency are there any alternatives or workarounds anyone has gotten to work?

z3ntu, 2017-08-26:
It would be nice to have. Another method of autostarting the daemon would be putting the razer-daemon.desktop file in the repository into /etc/xdg/autostart/ and removing the dbus-user-session package. I don't know if this can be achieved with dpkg packages but maybe...

terrycain, 2017-08-26:
> It would be nice to have. Another method of autostarting the daemon would be putting the razer-daemon.desktop file in the repository into /etc/xdg/autostart/ and removing the dbus-user-session package. I don't know if this can be achieved with dpkg packages but maybe...

I thought about this the other day, worse case we could just used xdg for all of ubuntu till we can drop 16.04

Zerophase, 2017-09-19:
I'll have to figure out how my driver is getting loaded. I removed udev from my startup, and every thing still gets bound correctly every boot.

terrycain, 2017-09-19:
Wird

ptrvilya, 2017-10-02:
There's a workaround:
`gnome-keyring-daemon &` 
[From [askubuntu](https://askubuntu.com/questions/913342/ubuntu-16-04-some-applications-are-slow-to-start)]

z3ntu, 2017-10-03:
@EchoCloud and rest here:
Please try the debs and reboot (note: they are only for 16.04!)
[1604debs.tar.gz](https://github.com/openrazer/openrazer/files/1352396/1604debs.tar.gz)

They are based on the contents in the PR #402

vlykad, 2017-10-18:
I installed the daily version from ppa:openrazer/daily on Ubuntu 16.04.3 and dbus-user-session was installed too. Is this expected?

SsnL, 2017-10-18:
Also interested in the correct way to install these.

KazWolfe, 2017-10-18:
I found that installing the drivers will install `dbus-user-session`, but you can just uninstall it immediately afterwards without anything complaining:

    sudo apt remove dbus-user-session

Note that this may not work for you depending on your config, but it's worth a shot.

z3ntu, 2017-10-18:
Found the issue why launchpad is failing my check: `make[1]: lsb_release: Command not found`. Will fix that.

z3ntu, 2017-10-18:
Should be fixed in 980462d

vlykad, 2017-10-19:
It works well for me now. Thank you!

z3ntu, 2017-11-14:
v2.1 is out

closed 2017-11-14 by z3ntu