openrazer/openrazer #359: Not usable on Razer Blade Stealth 13.3

royalbaecker, 2017-08-17:
Hi,

i installed all drivers and everything as mentioned.

Polychromatic says it finds a Razer Device, but the service cant find it.

i checked through all Issues, and followed the steps given to people with similiar problems, but nothing is working.

if anybody could help me, that would be super nice.

if logs or anything is needed, let me know.

Thank you so much!!

z3ntu, 2017-08-17:
Can you post the output of lsusb?

royalbaecker, 2017-08-17:
of course! thank you so much for the help.

here it is:

db@BladeStealth:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 04f3:2379 Elan Microelectronics Corp. 
Bus 001 Device 005: ID 1532:022d Razer USA, Ltd 
Bus 001 Device 004: ID 1bcf:2c9a Sunplus Innovation Technology Inc. 
Bus 001 Device 003: ID 0cf3:e300 Atheros Communications, Inc. 
Bus 001 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

royalbaecker, 2017-08-18:
here is the output:

[]

z3ntu, 2017-08-18:
The first command didn't output anything either?

terrycain, 2017-08-18:
022d isnt released yet

terrycain, 2017-08-18:
it'll will be out in next package

royalbaecker, 2017-08-18:
well, it outputs these 2 symbols, nothing more.

ahh okay, so that means all i need to do is wait and with the next update all should work fine.
i can live with that.

if you need any help, logs or anything, im available to help!

royalbaecker, 2017-08-18:
Update, i found and installed your files here:
https://github.com/terrycain/razer-drivers/issues/340

Now it works fine!!! 

Thank you very much!!!

terrycain, 2017-09-01:
1.1.16 is being built, will be out soon

closed 2017-09-01 by terrycain