openrazer/openrazer #360: Blade Pro 2015

3cpo, 2017-08-18:
Hi,

Is there any track pad support for Blade Pro 2015?  I've been using:
 * Razer USB keyboard device driver
 *  Copyright (c) 2014 Milan Votava (votava@mageo . cz)
 *  Copyright (c) 2013 Zach Carlson (FxChiP@Gmail . com)
 * Bases on:	usb_skeleton.c v2.2 by Greg Kroah-Hartman
Above does not support track pad graphics.  Is there anything available that supports track pad graphics and added buttons?
Many Thanks!

terrycain, 2017-08-18:
Nothing supports it yet not this project

z3ntu, 2017-08-18:
The output of lsusb please

3cpo, 2017-08-18:
Here you go:

lsusb
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 8087:07dc Intel Corp. 
Bus 003 Device 015: ID 1532:0116 Razer USA, Ltd 
Bus 003 Device 002: ID 0bda:57a2 Realtek Semiconductor Corp.                                                                     
Bus 003 Device 016: ID 25a7:fa23                                                                                                 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub    

I have a rule to load the driver that was referenced in my earlier post:
ACTION=="add", DRIVER=="usbhid", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0116", RUN+="/usr/local/razer-0.01/razer-bind-udev.sh

terrycain, 2017-09-01:
Can you get me some screenshots from synapse showing what the trackpad can do

3cpo, 2017-09-11:
Hi terrycain,

Sorry for the delay.  Here are some screenshots.  Thanks for looking at this!
[image]
[image]
[image]
[image]
[image]

z3ntu, 2017-10-20:
@3cpo We need screenshots from the Windows Application called "Razer Synapse" as that'll show us what hardware we are dealing with. The "Prerequisites" section on [this wiki page](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) will tell you more.

zackteo, 2018-06-10:
Is there still no support for Blade Pro 2015?

z3ntu, 2018-06-10:
@zackteo No, as we currently don't have the necessary information about the device that we need.

* Read [this](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) because we need packet captures.
* Screenshots of all available options on Razer Synapse are needed.
* And lastly, the product & model number of the device for documentation reasons please.

zackteo, 2018-06-10:
@z3ntu Hello, thanks for your super prompt response! Here are the screenshots from Razer Synapse :)
The last image is the window that appears when you click on apps. Synapse was bugging out (and the apps button didn't do anything) so I got that image from https://www.techspot.com/review/995-razer-blade-pro-2015/page2.html

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Do I also need to download wireshark on linux and proceed with phase 2? [Based on this](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol)

z3ntu, 2018-06-10:
The idea is having a Linux host with Wireshark capturing the usb traffic going in and out of the virtual machine where the usb device is passed through. If you just have Windows outside of a VM, you can also try capturing it directly in Windows, but I don't know if this'll work.

zackteo, 2018-06-14:
In that case, since I am dual booting Ubuntu and windows 10. I just download wireshark on windows 10 and run it there ?

z3ntu, 2018-06-14:
That could work, yes.