openrazer/openrazer #361: Razer Chroma HDK (Hardware Development Kit) - 4x16 LED strip driver

CalcProgrammer1, 2017-08-18:
Razer now makes an LED strip box similar to the NZXT Hue+.  I'm integrating it into my keyboard visualizer program on Windows with the latest version of the Chroma SDK.  I will try to capture some packets soon.  So far I know it is a USB device, has 4 LED strip ports capable of 16 LEDs per strip, can detect which strips are connected, and looks like it probably can detect how many LEDs are in a strip (if you cut the strip).  It has the standard Chroma patterns (wave, spectrum cycle, static, breathing, etc) and should support custom effects as well.

z3ntu, 2017-08-18:
Lol this came out a few days ago and you already have it 😁

terrycain, 2017-08-18:
I was gonna get one but its properly overpriced. Yeah I would guess it follows the standard protocol so should be an easy add once I've got captures.

Dublerq, 2018-01-01:
Once an open driver is fot that device is made, it should be easily doable to make a DIY Chroma HDK with arduino and cheaper led stripes. That would be awesome :D

pnomolos, 2019-08-31:
How can I help to get this working?

z3ntu, 2019-09-24:
@pnomolos There's a pull request open at #904

pnomolos, 2019-09-24:
@z3ntu The basics appear to be working for me - I only have two strips installed but was able to use `matrix_effect_breath` so far.  Will play around with it more.

z3ntu, 2019-09-25:
Yes, partial rows are really buggy, so always send full rows^^

commit 692e36be (2020-03-12): Add Razer Chroma HDK (Hardware Development Kit) (#1003)

Closes #361
closed 2020-03-12 by lah7