openrazer/openrazer #362: Configuration profiles.... (Naga Hex V2)

geremi1, 2017-08-20:
Are my configuration profiles of Razer Synapse going to work with this? I'd want something that handle my lightening effects, buttons configurations (all 7 to the left and 2 on top) and macros.

(I'm actually on MacOS El Captain and want to change to Ubuntu 16.04 LTS)

lah7, 2017-08-20:
Right now, no, not yet.

Razer Synapse (`.razerchroma`) files will need to be converted/parsed to a format that the daemon can process (this has been requested for my frontend -- https://github.com/lah7/polychromatic/issues/69 -- but I haven't got round to working on this yet :disappointed:).

I think profile management would be more something that front-ends would focus on then the driver/daemon (this repo), which just aims to support as many devices as possible and provide libraries for scripts/front-ends to make Razer devices do stuff.

Macros are WIP on the daemon side, but there is a V1 implementation already ([razerCommander](https://github.com/GabMus/razerCommander) supports this for keyboards, not sure about keypads). Hardware effects work though.

geremi1, 2017-08-20:
So if I reconfigure all my button tweaks and lighting effects, all's going to be as smart as on OS X?

closed 2018-02-28 by stale[bot]