openrazer/openrazer #363: Razer Naga 2014 (RZ01-0104) on RHEL7/CentOS7

Jifl, 2017-08-21:
Hi,

I've been trying to get a Naga 2014 working on CentOS7. At the moment, I'm just trying to get the basic kernel driver working, and have successfully built a driver, albeit with RPM .spec file modifications which are off-topic for this issue, and no daemon due to python3 dependency problems.

Anyway, when the mouse is plugged in without the driver installed, it works fine as a generic mouse. Once I install the razermouse driver and unplug/replug the mouse, then although /var/log/messages show it recognising okay and getting unmapped from the hid-generic driver and mapped to the razermouse driver, basic mouse operation - moving the pointer and clicking buttons - don't work at all. Pressing the side buttons (which are a HID keyboard) do work, as does setting stuff like the backlight with backlight_led_state in sysfs so it seems razermouse can talk to the mouse, but it doesn't appear to be considered a HID driver or something?

I've tried unbinding the mouse driver from razermouse and binding it back to hid-generic, but that doesn't help. Only uninstalling the razermouse driver and then replugging the mouse restores basic mouse operation.

What am I missing, or what can I do to investigate further?

Thanks, Jifl

terrycain, 2017-09-01:
Thats very weird. I cant thing of anything that would cause that, unless possibly selinux is getting involved but I dont think it should

Jifl, 2017-09-03:
I disabled SElinux, but no change, not very suprisingly.

Another little bit of info: when plugged in with the razermouse driver, an entry does appear at /dev/input/mouse1 (mouse0 is a different mouse). But if I do "cat /dev/input/mouse1" I get no output at all if I wiggle the mouse. But I do get output with mouse0 when I wiggle that other mouse, or indeed with the Naga and mouse1 if I use the hid-generic driver rather than razermouse.

Can you suggest any way I can debug what is happening? I'm not familiar enough with the HID system so there might be something simple to try.

Thanks, Jifl

terrycain, 2017-09-03:
run 
`ls -l /sys/bus/hid/drivers/razermouse/`
`ls -l /sys/bus/hid/devices/`

Jifl, 2017-09-05:
Thanks for the reply Terry. Here you go:

```
[root@b ~]# ls -l /sys/bus/hid/drivers/razermouse/
total 0
lrwxrwxrwx. 1 root root    0 Sep  5 02:05 0003:1532:0040.002C -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:1532:0040.002C
lrwxrwxrwx. 1 root root    0 Sep  5 02:05 0003:1532:0040.002D -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.1/0003:1532:0040.002D
lrwxrwxrwx. 1 root root    0 Sep  5 02:05 0003:1532:0040.002E -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.2/0003:1532:0040.002E
--w-------. 1 root root 4096 Sep  5 02:05 bind
lrwxrwxrwx. 1 root root    0 Sep  5 02:06 module -> ../../../../module/razermouse
--w-------. 1 root root 4096 Sep  5 02:06 new_id
--w-------. 1 root root 4096 Sep  5 02:05 uevent
--w-------. 1 root root 4096 Sep  5 02:06 unbind
[root@b ~]# ls -l /sys/bus/hid/devices/
total 0
lrwxrwxrwx. 1 root root 0 Sep  5 02:06 0003:046D:C506.000A -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6.3/3-6.3:1.0/0003:046D:C506.000A
lrwxrwxrwx. 1 root root 0 Sep  5 02:06 0003:058F:6364.0008 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-12/3-12:1.1/0003:058F:6364.0008
lrwxrwxrwx. 1 root root 0 Sep  5 02:02 0003:062A:5918.002A -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.1/3-3.1:1.0/0003:062A:5918.002A
lrwxrwxrwx. 1 root root 0 Sep  5 02:02 0003:062A:5918.002B -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.1/3-3.1:1.1/0003:062A:5918.002B
lrwxrwxrwx. 1 root root 0 Sep  5 02:06 0003:10D5:0001.0006 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6.1/3-6.1:1.0/0003:10D5:0001.0006
lrwxrwxrwx. 1 root root 0 Sep  5 02:06 0003:10D5:0001.0007 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6.1/3-6.1:1.1/0003:10D5:0001.0007
lrwxrwxrwx. 1 root root 0 Sep  5 02:05 0003:1532:0040.002C -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:1532:0040.002C
lrwxrwxrwx. 1 root root 0 Sep  5 02:05 0003:1532:0040.002D -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.1/0003:1532:0040.002D
lrwxrwxrwx. 1 root root 0 Sep  5 02:05 0003:1532:0040.002E -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.2/0003:1532:0040.002E
lrwxrwxrwx. 1 root root 0 Sep  5 02:06 0003:1A81:2001.000B -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6.4/3-6.4:1.0/0003:1A81:2001.000B
lrwxrwxrwx. 1 root root 0 Sep  5 02:06 0003:1A81:2001.000C -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6.4/3-6.4:1.1/0003:1A81:2001.000C
lrwxrwxrwx. 1 root root 0 Sep  5 02:06 0003:1A81:2001.000D -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6.4/3-6.4:1.2/0003:1A81:2001.000D
lrwxrwxrwx. 1 root root 0 Sep  5 02:06 0003:1A81:2001.000E -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6.4/3-6.4:1.3/0003:1A81:2001.000E
lrwxrwxrwx. 1 root root 0 Sep  5 02:06 0003:7545:0897.0009 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-14/3-14:1.3/0003:7545:0897.0009
[root@b ~]# 
```

terrycain, 2017-09-05:
Judging by that, in theory the driver part should work

ekistner, 2017-09-06:
Hi,
I have the exact same issue with a Razer Naga Hex v2 on a Centos 7.3.
So it doesn't seems to be an orphan bug, or maybe there is only two of us.

Let me know if I can help.

z3ntu, 2017-09-06:
Also see #248

Jifl, 2017-09-07:
Thanks z3ntu. I'm aware of the python3 issue, so in the context of this issue, we should focus only on the basic kernel driver support. I've edited the title of this issue accordingly.

Terry, in trying to debug this further, I have found something interesting. I read that I could use the "hid-recorder" tool to look at raw HID events. The interesting thing is that when I run it and choose the mouse device, the mouse instantly springs back to life! FAOD, this is with the razermouse driver, not the hid-generic driver. I don't even have to move the mouse to generate events - if I just run hid-recorder, choose the mouse device and instantly Ctrl-C it, the mouse works.

So clearly there is something that hid-recorder is doing which causes the mouse to reset or wake up or who knows what in some way. Does this help point the way towards something to look at further? When I run hid-recorder it prints this:
```D: 0
R: 61 05 01 09 06 a1 01 05 07 19 e0 29 e7 15 00 25 01 75 01 95 08 81 02 81 01 19 00 2a ff 00 15 00 26 ff 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00 25 01 75 01 95 03 91 02 95 05 91 01 c0
N: Razer Razer Naga 2014
P: usb-0000:00:14.0-8/input2
I: 3 1532 0040
^C
No events where recorded.
```
So clearly it's probing the razermouse HID device in some way, and something there is what gets it working.

terrycain, 2017-09-07:
That's really weird. Its definitely a centos problem then I think. I would look into what hid-recorder does to the device file during its startup. I would but I'm just too busy at the moment

Jifl, 2017-09-10:
Curiouser and curiouser.... from looking at hid-recorder and then my own test program, all that's required to "wake up" the mouse is to open() the /dev/hidraw file for the mouse, e.g. /dev/hidraw2. I don't even need any ioctl()'s. Just opening the device file. Perhaps anyone who is using the daemon finds that the daemon is opening the hidraw file, so they never see this problem? And it's only because the daemon isn't working on CentOS (yet) that we see this?

Jifl, 2017-09-17:
Perhaps someone can try it and see if this can be reproduced elsewhere: kill razer-daemon process, then unplug mouse, perhaps "rmmod razermouse" too just to be sure, and then plug mouse back in again and see if you get the same symptoms.

z3ntu, 2017-09-29:
See #399 for EL7 packages.

moozhub, 2017-09-29:
I don't seem to be having this problem with the builds that I have.. 

```
Sep 29 04:08:24 localhost kernel: usb 1-1: USB disconnect, device number 2
Sep 29 04:08:25 localhost acpid: input device has been disconnected, fd 9
Sep 29 04:08:25 localhost acpid: input device has been disconnected, fd 10
Sep 29 04:08:25 localhost kernel: usb 1-2: USB disconnect, device number 3
Sep 29 04:08:25 localhost acpid: input device has been disconnected, fd 12
Sep 29 04:08:25 localhost kernel: usbhid 1-2:1.0: Razer Device disconnected
Sep 29 04:08:25 localhost kernel: usbhid 1-2:1.1: Razer Device disconnected
Sep 29 04:08:25 localhost kernel: usbhid 1-2:1.2: Razer Device disconnected
Sep 29 04:10:17 localhost kernel: usb 1-2: new full-speed USB device number 7 using xhci_hcd
Sep 29 04:10:17 localhost kernel: usb 1-2: New USB device found, idVendor=1532, idProduct=0060
Sep 29 04:10:17 localhost kernel: usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Sep 29 04:10:17 localhost kernel: usb 1-2: Product: Razer Lancehead Tournament Edition
Sep 29 04:10:17 localhost kernel: usb 1-2: Manufacturer: Razer
Sep 29 04:10:17 localhost kernel: input: Razer Razer Lancehead Tournament Edition as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:1532:0060.0008/input/input30
Sep 29 04:10:17 localhost kernel: hid-generic 0003:1532:0060.0008: input,hidraw0: USB HID v1.11 Mouse [Razer Razer Lancehead Tournament Edition] on usb-0000:00:14.0-2/input0
Sep 29 04:10:17 localhost kernel: input: Razer Razer Lancehead Tournament Edition as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/0003:1532:0060.0009/input/input31
Sep 29 04:10:17 localhost kernel: hid-generic 0003:1532:0060.0009: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer Lancehead Tournament Edition] on usb-0000:00:14.0-2/input1
Sep 29 04:10:17 localhost kernel: input: Razer Razer Lancehead Tournament Edition as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/0003:1532:0060.000A/input/input32
Sep 29 04:10:17 localhost kernel: hid-generic 0003:1532:0060.000A: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer Lancehead Tournament Edition] on usb-0000:00:14.0-2/input2
Sep 29 04:10:17 localhost mtp-probe: checking bus 1, device 7: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2"
Sep 29 04:10:17 localhost mtp-probe: bus: 1, device: 7 was not an MTP device
Sep 29 04:10:21 localhost kernel: usb 1-1: new full-speed USB device number 8 using xhci_hcd
Sep 29 04:10:21 localhost kernel: usb 1-1: New USB device found, idVendor=1532, idProduct=0216
Sep 29 04:10:21 localhost kernel: usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Sep 29 04:10:21 localhost kernel: usb 1-1: Product: Razer BlackWidow X Chroma
Sep 29 04:10:21 localhost kernel: usb 1-1: Manufacturer: Razer
Sep 29 04:10:21 localhost kernel: input: Razer Razer BlackWidow X Chroma as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:1532:0216.000B/input/input33
Sep 29 04:10:21 localhost kernel: hid-generic 0003:1532:0216.000B: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer BlackWidow X Chroma] on usb-0000:00:14.0-1/input0
Sep 29 04:10:21 localhost kernel: input: Razer Razer BlackWidow X Chroma as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:1532:0216.000C/input/input34
Sep 29 04:10:21 localhost kernel: hid-generic 0003:1532:0216.000C: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer BlackWidow X Chroma] on usb-0000:00:14.0-1/input1
Sep 29 04:10:21 localhost kernel: input: Razer Razer BlackWidow X Chroma as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:1532:0216.000D/input/input35
Sep 29 04:10:21 localhost kernel: hid-generic 0003:1532:0216.000D: input,hidraw5: USB HID v1.11 Mouse [Razer Razer BlackWidow X Chroma] on usb-0000:00:14.0-1/input2
Sep 29 04:10:21 localhost mtp-probe: checking bus 1, device 8: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1"
Sep 29 04:10:21 localhost mtp-probe: bus: 1, device: 8 was not an MTP device
Sep 29 04:10:21 localhost razer_mount[7344]: Driver razerkbd
Sep 29 04:10:21 localhost razer_mount[7345]: Driver razerkbd
Sep 29 04:10:21 localhost razer_mount[7344]: Device_ID 0003:1532:0216.000B
Sep 29 04:10:21 localhost razer_mount[7345]: Device_ID 0003:1532:0216.000C
Sep 29 04:10:21 localhost razer_mount[7347]: Driver razerkbd
Sep 29 04:10:21 localhost razer_mount[7345]: Unbinding 0003:1532:0216.000C from hid-generic
Sep 29 04:10:21 localhost razer_mount[7344]: Unbinding 0003:1532:0216.000B from hid-generic
Sep 29 04:10:21 localhost razer_mount[7347]: Device_ID 0003:1532:0216.000D
Sep 29 04:10:21 localhost razer_mount[7347]: Unbinding 0003:1532:0216.000D from hid-generic
Sep 29 04:10:21 localhost razer_mount[7345]: Binding 0003:1532:0216.000C to razerkbd
Sep 29 04:10:21 localhost kernel: input: Razer Razer BlackWidow X Chroma as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:1532:0216.000C/input/input36
Sep 29 04:10:21 localhost acpid: input device has been disconnected, fd 12
Sep 29 04:10:21 localhost acpid: input device has been disconnected, fd 10
Sep 29 04:10:21 localhost razer_mount[7344]: Binding 0003:1532:0216.000B to razerkbd
Sep 29 04:10:21 localhost kernel: input: Razer Razer BlackWidow X Chroma as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:1532:0216.000B/input/input37
Sep 29 04:10:21 localhost razer_mount[7347]: Binding 0003:1532:0216.000D to razerkbd
[... cut ...]
Sep 29 04:10:21 localhost razer_mount[7344]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0216.000B/
Sep 29 04:10:21 localhost razer_mount[7344]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0216.000B/
Sep 29 04:10:22 localhost razer_mount[7347]: Finished binding 0003:1532:0216.000D
Sep 29 04:10:22 localhost razer_mount[7347]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0216.000D/
Sep 29 04:10:22 localhost razer_mount[7347]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0216.000D/
```

Jifl, 2017-10-02:
Hi moozhub, it looks like your hardware is sufficiently different that I'm not sure we can really draw any conclusions - in particular it uses the razerkbd driver for the mouse, not razermouse which is what the Naga uses.

But if anyone out there is definitely using the razermouse driver, and can see what happens when they plug it in *without* razer daemon running, that might be a useful data point I think.

moozhub, 2017-10-02:
Sorry, you're right I didn't even realize that.. Luckily I had another compatible razer device laying around that will use the razermouse driver.. this was all with openrazer-daemon shut off and razermouse module unloaded.

```
Oct 01 22:57:00 localhost.localdomain kernel: usb 1-2: new full-speed USB device number 17 using xhci_hcd
Oct 01 22:57:00 localhost.localdomain kernel: usb 1-2: New USB device found, idVendor=1532, idProduct=004c
Oct 01 22:57:00 localhost.localdomain kernel: usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Oct 01 22:57:00 localhost.localdomain kernel: usb 1-2: Product: Razer Diamondback
Oct 01 22:57:00 localhost.localdomain kernel: usb 1-2: Manufacturer: Razer
Oct 01 22:57:00 localhost.localdomain kernel: input: Razer Razer Diamondback as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:1532:004C.0026/input/input84
Oct 01 22:57:00 localhost.localdomain kernel: hid-generic 0003:1532:004C.0026: input,hidraw0: USB HID v1.11 Mouse [Razer Razer Diamondback] on usb-0000:00:14.0-2/input0
Oct 01 22:57:00 localhost.localdomain kernel: input: Razer Razer Diamondback as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/0003:1532:004C.0027/input/input85
Oct 01 22:57:00 localhost.localdomain kernel: hid-generic 0003:1532:004C.0027: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer Diamondback] on usb-0000:00:14.0-2/input1
Oct 01 22:57:00 localhost.localdomain kernel: input: Razer Razer Diamondback as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/0003:1532:004C.0028/input/input86
Oct 01 22:57:00 localhost.localdomain kernel: hid-generic 0003:1532:004C.0028: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer Diamondback] on usb-0000:00:14.0-2/input2
Oct 01 22:57:00 localhost.localdomain mtp-probe[25703]: checking bus 1, device 17: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2"
Oct 01 22:57:00 localhost.localdomain mtp-probe[25703]: bus: 1, device: 17 was not an MTP device
Oct 01 22:57:00 localhost.localdomain razer_mount[25722]: Driver razermouse
Oct 01 22:57:00 localhost.localdomain razer_mount[25723]: Driver razermouse
Oct 01 22:57:00 localhost.localdomain razer_mount[25724]: Device_ID 0003:1532:004C.0026
Oct 01 22:57:00 localhost.localdomain razer_mount[25726]: Device_ID 0003:1532:004C.0028
Oct 01 22:57:00 localhost.localdomain razer_mount[25727]: Driver razermouse
Oct 01 22:57:00 localhost.localdomain razer_mount[25728]: Unbinding 0003:1532:004C.0026 from hid-generic
Oct 01 22:57:00 localhost.localdomain razer_mount[25729]: Unbinding 0003:1532:004C.0028 from hid-generic
Oct 01 22:57:00 localhost.localdomain razer_mount[25730]: Device_ID 0003:1532:004C.0027
Oct 01 22:57:00 localhost.localdomain razer_mount[25732]: Unbinding 0003:1532:004C.0027 from hid-generic
Oct 01 22:57:00 localhost.localdomain razer_mount[25733]: Binding 0003:1532:004C.0026 to razermouse
Oct 01 22:57:00 localhost.localdomain kernel: input: Razer Razer Diamondback as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:1532:004C.0026/input/input87
Oct 01 22:57:00 localhost.localdomain razer_mount[25734]: Binding 0003:1532:004C.0028 to razermouse
Oct 01 22:57:00 localhost.localdomain kernel: input: Razer Razer Diamondback as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/0003:1532:004C.0028/input/input88
Oct 01 22:57:00 localhost.localdomain razer_mount[25735]: Binding 0003:1532:004C.0027 to razermouse
Oct 01 22:57:00 localhost.localdomain kernel: input: Razer Razer Diamondback as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/0003:1532:004C.0027/input/input89
Oct 01 22:57:00 localhost.localdomain kernel: razermouse 0003:1532:004C.0026: input,hidraw0: USB HID v1.11 Mouse [Razer Razer Diamondback] on usb-0000:00:14.0-2/input0
Oct 01 22:57:00 localhost.localdomain kernel: razermouse 0003:1532:004C.0028: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer Diamondback] on usb-0000:00:14.0-2/input2
Oct 01 22:57:00 localhost.localdomain razer_mount[25738]: Finished binding 0003:1532:004C.0026
Oct 01 22:57:00 localhost.localdomain kernel: razermouse 0003:1532:004C.0027: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer Diamondback] on usb-0000:00:14.0-2/input1
Oct 01 22:57:00 localhost.localdomain razer_mount[25740]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:004C.0026/
Oct 01 22:57:00 localhost.localdomain razer_mount[25742]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:004C.0026/
Oct 01 22:57:00 localhost.localdomain razer_mount[25746]: Finished binding 0003:1532:004C.0028
Oct 01 22:57:00 localhost.localdomain razer_mount[25747]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:004C.0028/
Oct 01 22:57:00 localhost.localdomain razer_mount[25749]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:004C.0028/
Oct 01 22:57:00 localhost.localdomain razer_mount[25753]: Finished binding 0003:1532:004C.0027
Oct 01 22:57:00 localhost.localdomain razer_mount[25754]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:004C.0027/
Oct 01 22:57:00 localhost.localdomain razer_mount[25756]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:004C.0027/
Oct 01 22:57:08 localhost.localdomain kernel: usb 1-2: USB disconnect, device number 17
Oct 01 22:57:08 localhost.localdomain kernel: usbhid 1-2:1.0: Razer Device disconnected
Oct 01 22:57:08 localhost.localdomain acpid[946]: input device has been disconnected, fd 11
Oct 01 22:57:08 localhost.localdomain kernel: usbhid 1-2:1.1: Razer Device disconnected
[... cut ...]
Oct 01 22:57:14 localhost.localdomain razer_mount[25854]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:004C.002A/
Oct 01 22:57:29 localhost.localdomain kernel: usb 1-2: USB disconnect, device number 18
Oct 01 22:57:29 localhost.localdomain kernel: usbhid 1-2:1.0: Razer Device disconnected
Oct 01 22:57:29 localhost.localdomain acpid[946]: input device has been disconnected, fd 10
Oct 01 22:57:29 localhost.localdomain kernel: usbhid 1-2:1.1: Razer Device disconnected
Oct 01 22:57:29 localhost.localdomain kernel: usbhid 1-2:1.2: Razer Device disconnected
```

z3ntu, 2017-10-20:
I'm going to close this issue as we don't support CentOS/RedHat.

closed 2017-10-20 by z3ntu

philippeback, 2017-11-26:
`sudo yum install hid-replay` for hid-recorder on CentOS / epel

philippeback, 2017-11-26:
Running hid-recorder once and selecting the keyboard indeed makes it work on CentOS 7.3