openrazer/openrazer #364: Razer Imperator 2012 -> (reopen) #51

zor1984qq, 2017-08-22:
[https://github.com/terrycain/razer-drivers/issues/51](url)
Have the same issue under manjaro, I completely installed openrazer-meta from AUR without any problems, the driver module seems to compiled normally for KMS but I don't see to any driver load properly on the system startup in dmesg.

dmesg | grep razer shows nothing, only if Razer used first latter in upper case:
```
[    4.099105] input: Razer Razer Imperator as /devices/pci0000:00/0000:00:0b.0/usb2/2-2/2-2:1.0/0003:1532:0017.0004/input/input5
[    4.099181] hid-generic 0003:1532:0017.0004: input,hidraw3: USB HID v1.11 Mouse [Razer Razer Imperator] on usb-0000:00:0b.0-2/input0
[    4.135815] input: Razer Razer Imperator as /devices/pci0000:00/0000:00:0b.0/usb2/2-2/2-2:1.1/0003:1532:0017.0005/input/input6
[    4.190057] hid-generic 0003:1532:0017.0005: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer Imperator] on usb-0000:00:0b.0-2/input1
```
Tried to restart daemon and pushing retry button in polychromatic doesn't help. In fact razer-daemon itself works normally but I don't know from where it starts it causing a lot frustration since I use systemd. If I use AUR prepared command to create symlink for systemd launch service doesn't start and can't connect to Dbus even if I edit systemd service file to ExecPreStart=/usr/bin/razer-daemon -s or -r keys but these is not primary problem.
> -r, --respawn
>               Stop any existing daemon first, if one is running.
>        -s, --stop
>               Gracefully stop the existing daemon.
The main problem seems to me that kernel module wouldn't load at all or driver didn't applied. I checked udev it has a file in /lib/udev/rules.d/99-razer.rules but it seems that it not installed by AUR packages by default (maybe some permissions problem?) I just copied these file from github and paste it with default root rights in /etc/udev/rules.d/99-razer.rules so it is show up then in /lib/udev/rules.d/99-razer.rules as I thought.

dmesg bash colored output: [https://pastebin.com/Jkht3uTF](url)
sudo journalctl -b | grep -i razer_mount (these does not show anything just clear)
razer.log file [https://pastebin.com/Lh6b5By1](url)
```
uname -a
Linux PC_NAME 4.9.43-1-MANJARO #1 SMP PREEMPT Sun Aug 13 20:28:47 UTC 2017 x86_64 GNU/Linux
core/linux49-headers 4.9.43-1 [installed]
    Header files and scripts for building modules for Linux49 kernel
razer-daemon -Fv
Daemon already exists. Please stop that one. (polychromatic deal with these in one button shot)
```
I don't have secure boot my PC is from 2009 mobo STRIKER II NSE ASUS BIOS 0901
According to an ArchWiki I did the following (but its not related here cause its not working even when I have no these file there):
```
cat /etc/X11/xorg.conf.d/01-mouse.conf 
Section "InputDevice"
    Identifier  "Mouse"
    Driver  "mouse"
    Option  "Device" "/dev/input/mice"
```

What else can I do to determine if something goes wrong or fix these?

z3ntu, 2017-08-22:
Run sudo modprobe razerkbd

zor1984qq, 2017-08-22:
Still not recognized. I tried to load razercore, razermouse, razerkbd (not found razer_mount on my PC after full installation of meta package from AUR or even if I manually install separate parts needed for razer-daemon work, it works now and restarts properly without systemd, but still no detection of my old mouse).

How can I check if I run appropriate kernel module and its loads for me from openrazer-drivers package? On install I saw that AUR is compiled module for my 4.9 linux kernel, maybe its just not load it for me and I need to write something in /etc/default/grub config?

Dmesg and razer.log shows no errors...and I use internal ACPI functions from BIOS:
```
GRUB_CMDLINE_LINUX_DEFAULT="quiet splash"
GRUB_CMDLINE_LINUX="acpi_osi=Linux acpi=force clocksource=hpet hpet=force divider=10 radeon.dpm=1 radeon.audio=1"
```

z3ntu, 2017-08-22:
Can you post the output of "pacman -Q | grep razer"?

zor1984qq, 2017-08-22:
Yes sure here it is:
pacman -Q | grep razer
openrazer-meta 1.1.15-1
python-razer 1.1.15-1
razer-daemon 1.1.15-1
razer-driver-dkms 1.1.15-1

z3ntu, 2017-09-09:
Did you solve the issue?

z3ntu, 2017-10-04:
Closing as there was no response for ~1 month.

closed 2017-10-04 by z3ntu

mirh, 2021-07-03:
Perhaps the mouse may even have been bought in 2012, or idk, but 1532:0017 is #488

z3ntu, 2021-07-03:
@mirh If your mouse has the USB ID 1532:0017 then follow that issue instead of this one please.