openrazer/openrazer #365: Abyssus 1800 support

Kliesen, 2017-08-22:
Hello, I have an Abyssus 1800 razer mice and I wanted to know if it is possible to add support for it, it doesn't have any buttons for dpi change or lightning, everything is controlled with synapse (dpi and blue razer light)
```
$ lsusb -d 1532:0020 -v

Bus 003 Device 002: ID 1532:0020 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0020 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      41
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8

```
```
$ ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 140 ago 22 12:25 .
drwxr-xr-x 4 root root 420 ago 22 12:25 ..
lrwxrwxrwx 1 root root   9 ago 22 12:25 usb-Razer_Razer_Abyssus_1800-event-mouse -> ../event2
lrwxrwxrwx 1 root root   9 ago 22 12:25 usb-Razer_Razer_Abyssus_1800-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 ago 22 12:25 usb-Razer_Razer_Abyssus_1800-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 ago 22 12:25 usb-SINO_WEALTH_USB_KEYBOARD-event-if01 -> ../event5
lrwxrwxrwx 1 root root   9 ago 22 12:25 usb-SINO_WEALTH_USB_KEYBOARD-event-kbd -> ../event4

```
I can add synapse  screenshots but I'm still creating my dual boot system.

terrycain, 2017-09-01:
Yeah screenshots would be god

Kliesen, 2017-09-02:
Sorry for the late response, I had some troubles with my HDD. here are the screens

http://prntscr.com/ggc5yl
http://prntscr.com/ggc65a
http://prntscr.com/ggc68z

terrycain, 2017-09-03:
Read [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol) because we'll need packet captures.

* 1 Packet capture of sliding the dpi from min to max a few times
* 1 Packet capture of changing the poll rate between all the options
* 1 Packet capture of toggling the logo light on and off a few times

Kliesen, 2017-09-09:
Here are the packet captures, I used a secondary mouse for option changing, so it only has the option changes. I changed the 3 options a few times each

[abyssus1800pcap.tar.gz](https://github.com/openrazer/openrazer/files/1290071/abyssus1800pcap.tar.gz)

Bielcito, 2017-10-10:
Hello, I've an abyssus 1800 too, there are anything that I can do for help to add support for this mouse?

z3ntu, 2017-10-28:
Please test the code in the branch `feature_abyssus1800`. If you don't know how, just say it (and the distro you are using) and I'll send you instructions.

Kliesen, 2017-11-09:
I may need some instruction, I'm not very experienced in compiling. I'm using debian 9

z3ntu, 2017-11-09:
Run that in your terminal:
```
git clone https://github.com/openrazer/openrazer -b feature_abyssus1800
cd openrazer
./scripts/build_debs.sh
```
If there's an error, look into the log file that's shown in the output and it should tell you that some command wasn't found, which you have to install.

Bielcito, 2017-11-17:
I make the steps you say z3ntu, and this files is created on dist folder:

openrazer-daemon_2.0.0-0ubuntu1_amd64.deb
openrazer-doc_2.0.0-0ubuntu1_all.deb
openrazer-kernel-modules-dkms_2.0.0-0ubuntu1_amd64.deb
openrazer-meta_2.0.0-0ubuntu1_all.deb
python3-openrazer_2.0.0-0ubuntu1_all.deb

what should i do?

z3ntu, 2017-11-17:
Run `sudo dpkg -i *.deb && sudo apt install -f`

Bielcito, 2017-11-18:
Everything works fine.
Test with Polychromatic Controller after restart Ubuntu.
I'm not sure about Polling Rate... No visual changes, but probably is working too.
Thank you! You're the best.

terrycain, 2017-11-18:
Cool if its working, I'll merge it in the main codebase the.

terrycain, 2017-11-18:
Has been merged, will be released soonish.

closed 2017-11-18 by terrycain

williamcanin, 2017-12-12:
How to apply this solution in another distro (Fedora)?

Bielcito, 2017-12-12:
This is because the changes aren't released yet.

z3ntu, 2017-12-12:
A new release will be out in the next couple of days. Please by patient.

z3ntu, 2019-04-10:
@williamcanin @rolivegab @BlazeKl Did you try DPI on the Abyssus 1800 and has it ever worked?

BlazeKl, 2019-04-10:
it did work when I tested it with polychromatic and razergenie both dpi and blue light

z3ntu, 2019-04-10:
What firmware version does your mouse has? 
```
cat /sys/bus/hid/drivers/razermouse/*1532:0020*/firmware_version
```

BlazeKl, 2019-04-10:
it says v1.3

commit f299cc37 (2017-11-18): Add Razer Abyssus 1800

Closes #365
commit 5a8350b1 (2017-11-18): Add Razer Abyssus 1800

Closes #365