openrazer/openrazer #367: Razer Kraken 7.1 Chromas not being detected by Polychromatic.

BenColligan, 2017-08-24:
I have already asked about this issue on Ask Ubuntu so I am just going to copy and paste my initial question.

> 

Long story short, I am trying to install the non native Razer Drivers for my headset so I can mess around more with my headset setting (like bass boost for example) I am using Ubuntu 16.04 LTS and would really love it if someone helped, thanks.

On the installation document on the git-hub page of Polychromatic it tells you the exact USB VID:PID's of which products are supported, so I typed in 'lsusb | grep '1532:'' into the command line like it says and it comes up with;

Bus 004 Device 004: ID 1532:0504 Razer USA, Ltd

Which is great because it says the product with this USB VID:PID (Kraken 7.1 Chroma's) are supported so I am not really sure why It is not getting recognized when I launch the Polychromatic Controller. There is no the daemon could be broken surely because Its a super fresh install.

Update: I just found the troubleshooting guide and went through it all but nothing changed, the only semi hopefully response I got from putting a certain command into the command was the command sudo gpasswd -a benjamin plugdev which gave me the message adding benjamin to group plugdev. Not sure if thats normal or it does it every time I enter that command.

z3ntu, 2017-08-24:
Did you reboot since running the command with the plugdev stuff? If no, do so and try again.

BenColligan, 2017-08-24:
Yep. One thing I just realized is that when I do 'sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc' in the terminal it says process not found. I assume that means my daemon is not running, how would I go about starting it up again?

z3ntu, 2017-08-24:
Can you make a screenshot of that error and upload it here?

BenColligan, 2017-08-24:
Yup, sure. Here you go. http://i.imgur.com/ciaKfOn.png

z3ntu, 2017-08-24:
The daemon is called `razer-daemon` instead of `razer-service` for a few releases now. Please use that name

z3ntu, 2017-08-24:
And where did you get the "razer-service" name from?

BenColligan, 2017-08-24:
Oh, I was browsing the YouTube comment section of this video hoping to maybe find a fix or something I had forgot to do. https://www.youtube.com/watch?v=3JrqJA8d5BE
Want me to try the same again but razer-daemon instead of razer-service?

z3ntu, 2017-08-24:
Ah ^^ Yeah run `razer-daemon -Fv` and tell me if there's a success message (that says "Serving DBus") at the end

BenColligan, 2017-08-24:
So I kinda dun-goofed, I copy and pasted the original command and then changed the things but it auot ran it without the -Fv tag and I closed the terminal but from what I saw it was a list of Razer Products and I did see something DBus in there but it was not Serving DBus I am pretty sure. I opened up another Terminal and ran it again and it came up with process not found again. Here is a screenshot of it and its the last line which I add the -Fv tag. http://i.imgur.com/14oM0wX.png

terrycain, 2017-08-24:
Change && to ;

BenColligan, 2017-08-24:
Okie doke, I will post the results one moment.

BenColligan, 2017-08-24:
http://i.imgur.com/undefined.png and Yes the line right below the last one says...

2017-08-24 17:00:50 | razer                          | INFO     | Serving DBus

z3ntu, 2017-08-24:
Does polychromatic show the device?

terrycain, 2017-08-24:
also run

```
ls -l /sys/bus/hid/drivers/razer*/0*/
```

BenColligan, 2017-08-24:
What do you mean z3ntu? If you mean when I open Polychromatic then no it says no devices detected but considering it shows my headset in the sound setting on Ubuntu.
I will run that command now terrycain and post the results.

BenColligan, 2017-08-24:
http://i.imgur.com/PKpa33H.png No such directory? o.O Wut.

terrycain, 2017-08-24:
ok try
```
sudo modprobe razerkraken
lsmod | grep razer
```

BenColligan, 2017-08-24:
Here is the outcome of both commands, I have an inkling that my drivers are not installed properly?

http://i.imgur.com/y0SeTJm.png

z3ntu, 2017-08-24:
Look info wiki for uefi secure boot, you have to turn that off, you can google something with "dkms secure boot"

BenColligan, 2017-08-24:
Ok, do you disable UEFI secure boot via the BIOS? From googling it thats the only way I see of turning it off.

terrycain, 2017-08-24:
yeah its a bios setting somewhere

BenColligan, 2017-08-24:
Ok, I will be right back with UEFI secure boot turned off and see if it changed anything while launching the program.

BenColligan, 2017-08-24:
So I am typing on my phone now. I have deleted the PK key and changed the OS type to 'other OS' and not Windows and I have come up with a window before even booting into Ubuntu with the options. Ubuntu, advanced options for Ubuntu and systme setup. I will provide screenshots one second.

BenColligan, 2017-08-24:
http://i.imgur.com/YPgrZfl.jpg
This is what comes up before I get to the Ubuntu login screen.

BenColligan, 2017-08-24:
And thus is what it looks like on the part of my BIOS were I turn off Secure Boot. Should I change switching OS type to Windows UEFI mode and not other OS?

BenColligan, 2017-08-24:
http://i.imgur.com/p4uxMcD.jpg

BenColligan, 2017-08-24:
So I tried switching to Windows UEFI mode in my BIOS instead other OS but it is giving me the same thing. Its bringing me to a screen with three options and I pick Ubuntu but it brings me to a black purple screen.

BenColligan, 2017-08-24:
I am stuck on a purple screen after disabling Secure Boot, any ideas?

z3ntu, 2017-08-24:
Sorry, but I have zero experience with mainboards and secure boot as I've never used it. :/

terrycain, 2017-08-24:
what happens if you re-enable secure boot?

BenColligan, 2017-08-24:
OK. So. With secure boot off I go straight to a page saying please unlock the disk but I am not able to enter a password and if I enable secure boot then I come to the page which says Ubuntu, Advanced options or system setup.

terrycain, 2017-08-24:
did you have windows on there with bitlocker?

terrycain, 2017-08-24:
odd, ive never come across anything like this

BenColligan, 2017-08-24:
No, this drive was wiped before installing Ubuntu. There is another option on the area which I disabled secure boot, it is the OS type. Should I try switching it to other OS instead of Windows UEFI

BenColligan, 2017-08-24:
If I pick Other OS I just get a purple screen...

terrycain, 2017-08-24:
it shouldnt matter, on mine i literally just turned secure boot off and it booted like nothing had changed

terrycain, 2017-08-24:
how fresh is the installation, as the only think im pretty sure would work is an install once secure boot is off, but your seeing grub so it cant be too bad

BenColligan, 2017-08-24:
Of course something weird had happen the first time I timed of Secure Boot so I went the super awkward way and reinstalled Ubuntu with secure boot off. Now I am going to reinstall the Razer drivers and Polychromatic and come back with some results and the steps I went through exactly.

BenColligan, 2017-08-24:
Ok so, I have reinstalled Polychromatic and the drivers and these are all the commands I used.

sudo add-apt-repository ppa:terrz/razerutils
sudo apt update
sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
sudo gpasswd -a benjamin plugdev
sudo add-apt-repository ppa:lah7/polychromatic
sudo apt update
sudo apt install polychromatic

I am sure these are correct so I hope it works -_-

terrycain, 2017-08-24:
yeah looks about right

BenColligan, 2017-08-24:
Well, it didn't. I see on the 'Device not recognized' page it says. Use 'modprobe' to bind a driver to a device. How would I do that? o.O

terrycain, 2017-08-24:
```
sudo modprobe razerkraken
lsmod | grep razer
```

BenColligan, 2017-08-24:
Right. sudo modprobe razerkraken came back with nothing and lsmod | grep razer came back with;

razerkraken            20480  0
hid                   118784  3 hid_generic,usbhid,razerkraken

terrycain, 2017-08-24:
yeah if modprobe is successful it returns nothing.
ok so now
```
ls -l /sys/bus/hid/drivers/razer*/0*/
```

BenColligan, 2017-08-24:
ls -l /sys/bus/hid/drivers/razer*/0*/ came back with;

ls: cannot access '/sys/bus/hid/drivers/razer*/0*/': No such file or directory

BenColligan, 2017-08-24:
I remember during the install it said it downloaded somewhere else...

terrycain, 2017-08-24:
odd.

ls -l /sys/bus/hid/drivers/

BenColligan, 2017-08-24:
Perfect! Came back with;

total 0
drwxr-xr-x 2 root root 0 Aug 24 20:02 hid-generic
drwxr-xr-x 2 root root 0 Aug 24 20:53 razerkraken

terrycain, 2017-08-24:
ls -l /sys/bus/hid/drivers/razerkraken/

BenColligan, 2017-08-24:
Comes back with;

total 0
--w------- 1 root root 4096 Aug 24 20:59 bind
lrwxrwxrwx 1 root root    0 Aug 24 20:59 module -> ../../../../module/razerkraken
--w------- 1 root root 4096 Aug 24 20:59 new_id
--w------- 1 root root 4096 Aug 24 20:53 uevent
--w------- 1 root root 4096 Aug 24 20:59 unbind

terrycain, 2017-08-24:
ok so the headphones havent been bound to the driver. Did you reboot since you installed it?
Otherwise unplug them, plug them back in then re-run `ls -l /sys/bus/hid/drivers/razerkraken/` that "should" fix it

BenColligan, 2017-08-24:
Oh drat, yeh I forgot to reboot. After rebooting should I run ls -l /sys/bus/hid/drivers/razerkraken/ again?

terrycain, 2017-08-24:
yeah

BenColligan, 2017-08-24:
After restarting it comes back with this; 

total 0
lrwxrwxrwx 1 root root    0 Aug 24 21:04 0003:1532:0504.0006 -> ../../../../devices/pci0000:00/0000:00:12.0/usb4/4-5/4-5:1.3/0003:1532:0504.0006
--w------- 1 root root 4096 Aug 24 21:04 bind
lrwxrwxrwx 1 root root    0 Aug 24 21:09 module -> ../../../../module/razerkraken
--w------- 1 root root 4096 Aug 24 21:09 new_id
--w------- 1 root root 4096 Aug 24 21:04 uevent
--w------- 1 root root 4096 Aug 24 21:09 unbind

terrycain, 2017-08-24:
"Should" work now

BenColligan, 2017-08-24:
Awesome, lemme test real quick.

BenColligan, 2017-08-24:
Worked! Thank you man! You are beyond helpful. Just one last thing, for my birthday next month I am getting a Blackwidow X Chroma and a Deathadder Chroma. If I wear to do it for them it would be these commands right? What would I put instead of razerkraken for both of them?

sudo modprobe razerblackwidow _and_ razerdeathadder _(I think)_
lsmod | grep razer
ls -l /sys/bus/hid/drivers/
ls -l /sys/bus/hid/drivers/razerblackwidow _and_razerdeathadder/

terrycain, 2017-08-24:
nah, those command were just for me to diagnose things. You should just be able to plug them in and they'll magically work.

BenColligan, 2017-08-24:
Oh gotcha ;) Thank you again!

terrycain, 2017-08-24:
Np :smile:

closed 2017-08-24 by terrycain