openrazer/openrazer #368: Deathstalker Classic 1532:0118

jwjacobs88, 2017-08-28:
Would love Deathstalker Classic support

Bus 001 Device 016: ID 1532:0118 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0118 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

z3ntu, 2017-08-28:
* Please post the output of
`lsusb -d 1532:0118 -v`
and
`ls -lah /dev/input/by-id/`
* Read [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol) because we probably need packet captures.
* Screenshots of all available options on Razer Synapse are needed.
* And lastly, the product & model number of the device for documentation reasons please.

jwjacobs88, 2017-08-28:
**ls -lah /dev/input/by-id/ relevant bits:**
`lrwxrwxrwx. 1 root root   9 Aug 28 10:31 usb-Razer_Razer_DeathStalker-event-kbd -> ../event9
lrwxrwxrwx. 1 root root  10 Aug 28 10:31 usb-Razer_Razer_DeathStalker-if01-event-kbd -> ../event10
lrwxrwxrwx. 1 root root  10 Aug 28 10:31 usb-Razer_Razer_DeathStalker-if02-event-mouse -> ../event11
lrwxrwxrwx. 1 root root   9 Aug 28 10:31 usb-Razer_Razer_DeathStalker-if02-mouse -> ../mouse2
`
**Screenshots from WIndows Razer Synapse:**
[http://imgur.com/a/76M3r](http://imgur.com/a/76M3r)

**Model #**: RZ03-0080
**Product #**: RZ03-00800100-R3U1

I'll read through the pack capture documentation later today. Thanks!

terrycain, 2017-09-01:
Yep need packet captures of
* switching between brightness and pulsate a few times
* changing brightness up and down
* changing game mode on and off a few times

BahamutGm, 2019-12-17:
### Device Info

* Name: Razer DeathStalker
* Product Number: RZ03-00800100-R3U1
* Model Number: RZ03-0080

### Device Information
```
Please post the output of lsusb -d 1532:0118 -v

Bus 003 Device 003: ID 1532:0118 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0118 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer DeathStalker
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

```
Please post the output of
    ls -lah /dev/input/by-id/
drwxr-xr-x 2 root root 200 dez 10 09:54 .
drwxr-xr-x 4 root root 540 dez 10 09:54 ..
lrwxrwxrwx 1 root root   9 dez 10 09:54 usb-Razer_Razer_DeathAdder_Essential-event-mouse -> ../event4
lrwxrwxrwx 1 root root   9 dez 10 09:54 usb-Razer_Razer_DeathAdder_Essential-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 dez 10 09:54 usb-Razer_Razer_DeathAdder_Essential-if02-event-kbd -> ../event6
lrwxrwxrwx 1 root root   9 dez 10 09:54 usb-Razer_Razer_DeathAdder_Essential-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 dez 10 09:54 usb-Razer_Razer_DeathStalker-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 dez 10 09:54 usb-Razer_Razer_DeathStalker-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 dez 10 09:54 usb-Razer_Razer_DeathStalker-if02-event-mouse -> ../event9
lrwxrwxrwx 1 root root   9 dez 10 09:54 usb-Razer_Razer_DeathStalker-if02-mouse -> ../mouse1

```

Vinidia, 2022-10-12:
bump

I have the same keyboard. If you need more info, please let me know.

rsmithiii, 2023-03-31:
> Yep need packet captures of
> 
> * switching between brightness and pulsate a few times
> * changing brightness up and down
> * changing game mode on and off a few times

Here's a pcapng file (inside zip) from my Razer DeathStalker of these things.
[razer_deathstalker_usbpcap.zip](https://github.com/openrazer/openrazer/files/11127099/razer_deathstalker_usbpcap.zip)

Please let me know if you'd like me to redo these in any specific way.

z3ntu, 2023-04-22:
@rsmithiii Packet captures look good.

Looks like a pretty standard old device. transaction_id=0xff

* `razer_chroma_standard_set_led_brightness` with `BACKLIGHT_LED`
* `razer_chroma_standard_set_led_effect`, also `BACKLIGHT_LED` with pulsate or static effect.
* Also macro_led and game_mode_led is supported

Let me know if you have any questions or troubles implementing this.

PR #2085 (2023-07-16): Add support for Razer Deathstalker Essential
Closes #368.

Adds support for Razer Deathstalker/Deathstalker Essential.
Tested in an Ubuntu VM with Polychromatic.
Brightness control, Pulsate and static effects, Game Mode LED, and Macro LED control are all functional.
closed 2023-07-16 by z3ntu