openrazer/openrazer #369: Razer Naga Classic Support

systmrrr, 2017-08-29:
[image]

Razer Naga Classic Support Please :D

z3ntu, 2017-08-29:
The output of `lsusb | grep 1532' please

systmrrr, 2017-08-29:
;)

systmrrr, 2017-08-29:
i will send screenshot

z3ntu, 2017-08-29:
@z3ntu
> Secondly, is this mouse really called "Razer Naga Classic" because according to my spreadsheet it's called the Razer Naga 2014. Because the 0040 mouse is already supported by the driver.

@systmrrr 
> But can not assign tasks to keys

(just that the conversation is complete)

systmrrr, 2017-08-30:
[image]

terrycain, 2017-09-01:
Yeah key bindings are a work in progress and long overdue, when I get time to finish the new macro functionality this will work.

closed 2017-09-01 by terrycain

systmrrr, 2017-09-01:
thank you. good work

Adi118, 2018-04-20:
@terrycain is there option to remap keys in razer mices yet ?

terrycain, 2018-04-20:
@Adi118 Nope not yet