openrazer/openrazer #370: Razer Naga Classic Support

z3ntu, 2017-08-29:
GitHub kinda died so I recreated this issue:

GitHub, [29.08.17 21:38]
🐛 New issue razer-drivers#369 Razer Naga Classic Support (https://github.com/terrycain/razer-drivers/issues/369)
by @systmrrr (https://github.com/systmrrr)

[image] 
 
Razer Naga Classic Support Please :D

Reply to this message to post a comment on GitHub.

GitHub, [29.08.17 21:39]
💬 New comment on razer-drivers#369 Razer Naga Classic Support (https://github.com/terrycain/razer-drivers/issues/369#issuecomment-325756371)
by @z3ntu (https://github.com/z3ntu)

The output of `lsusb | grep 1532' please

Reply to this message to post a comment on GitHub.

GitHub, [29.08.17 21:41]
💬 New comment on razer-drivers#369 Razer Naga Classic Support (https://github.com/terrycain/razer-drivers/issues/369#issuecomment-325756917)
by @systmrrr (https://github.com/systmrrr)

Bus 002 Device 003: ID 1532:0040 Razer USA, Ltd 

Reply to this message to post a comment on GitHub.

z3ntu, 2017-08-29:
@systmrr First of all, GitHub let your issue disappear 😄 
Secondly, is this mouse really called "Razer Naga Classic" because according to my spreadsheet it's called the Razer Naga 2014. Because the 0040 mouse is already supported by the driver.

systmrrr, 2017-08-29:
But can not assign tasks to keys

z3ntu, 2017-08-29:
Okay #369 is back 😃 Let's talk more there

closed 2017-08-29 by z3ntu