openrazer/openrazer #371: Deamon problem

m1roh, 2017-08-30:
Hi, I installed the drivers for my Ornata Chroma, but when I try to configure it with Razercommander or Polychromatic nothing happend at all. The keyboard is properly detected and I can activate the game mode with the 2 softs. So I think the problem come with the deamon, also when I look to the log.file of Polychromatic, the programm is actually send orders to the keyboard but they're not intrepreted. Is there a solution? Or have I to re-install Windows?

terrycain, 2017-08-31:
whats in the log of ~/.razer-service/logs

m1roh, 2017-09-01:
Hi, here is the content of the file : 
`2017-08-30 18:04:49 | razer                          | INFO     | Initialising Daemon (v1.1.15). Pid: 6659
2017-08-30 18:04:49 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2012 --------------------- (1532:010D)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerExpert ----------------- (1532:0202)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderChroma ------------------- (1532:0043)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderElite -------------------- (1532:005C)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerImperator -------------------------- (1532:002F)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerKrakenClassic ---------------------- (1532:0501)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wired --------------------- (1532:0024)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wireless ------------------ (1532:0025)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
2017-08-30 18:04:49 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
[... cut ...]
2017-09-01 09:59:58 | razer.screensaver              | DEBUG    | Received screensaver inactive signal
2017-09-01 09:59:58 | razer.device0                  | INFO     | Resuming RazerOrnataChroma
2017-09-01 09:59:58 | razer.device0                  | DEBUG    | DBus call set_brightness`

I'm definitely not an expert, but everything seems to be ok... So I don't understand at all what happens. You have to know that used to work when my laptop was on dual-boot with Win 10, but I made a clean install with only ubuntu 16.04 and now it doesn't work anymore.

terrycain, 2017-09-01:
How clean is the install? as I have a feeling if you try 17.* it would work better, nothing should have changed to break it.

m1roh, 2017-09-01:
> How clean is the install?
What do you mean?  I totally erased the hard drive before installing Ubuntu...

> as I have a feeling if you try 17.
Ok I download it now, I'll give you a feedback after install it...

m1roh, 2017-09-01:
Just made the fresh and clean install of ubuntu 17.04 and it still doesn't work. Again the log file : 
`2017-09-01 11:53:08 | razer                          | INFO     | Initialising Daemon (v1.1.15). Pid: 2500
2017-09-01 11:53:08 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2012 --------------------- (1532:010D)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerExpert ----------------- (1532:0202)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderChroma ------------------- (1532:0043)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderElite -------------------- (1532:005C)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerImperator -------------------------- (1532:002F)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerKrakenClassic ---------------------- (1532:0501)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wired --------------------- (1532:0024)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wireless ------------------ (1532:0025)
2017-09-01 11:53:08 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
[... cut ...]
2017-09-01 11:55:39 | razer.device0                  | DEBUG    | DBus call get_brightness
2017-09-01 11:55:48 | razer.device0                  | DEBUG    | DBus call set_static_effect
2017-09-01 11:55:48 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerOrnataChroma:SI1704F12003442, 'setStatic', dbus.Byte(0), dbus.Byte(255), dbus.Byte(0))
2017-09-01 11:55:48 | razer.device0                  | DEBUG    | DBus call get_brightness
2017-09-01 11:55:48 | razer.device0                  | DEBUG    | DBus call get_brightness`

terrycain, 2017-09-01:
Output of 
`ls -l /sys/bus/hid/drivers/razer*/0*/`
and
`id`

m1roh, 2017-09-01:
`m1roh@m1roh-HP:/$ ls -l /sys/bus/hid/drivers/razer*/0*/
/sys/bus/hid/drivers/razerkbd/0003:1532:021E.0001/:
total 0
-r--r--r-- 1 root plugdev 4096 sept.  1 11:50 country
lrwxrwxrwx 1 root plugdev    0 sept.  1 11:50 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 sept.  1 11:50 hidraw
drwxr-xr-x 3 root plugdev    0 sept.  1 11:50 input
-rw-rw---- 1 root plugdev 4096 sept.  1 11:50 key_alt_f4
-rw-rw---- 1 root plugdev 4096 sept.  1 11:50 key_alt_tab
-rw-rw---- 1 root plugdev 4096 sept.  1 11:50 key_super
-r--r--r-- 1 root plugdev 4096 sept.  1 11:50 modalias
drwxr-xr-x 2 root plugdev    0 sept.  1 11:50 power
-r--r--r-- 1 root plugdev 4096 sept.  1 11:50 report_descriptor
lrwxrwxrwx 1 root plugdev    0 sept.  1 11:50 subsystem -> ../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 sept.  1 11:50 uevent

/sys/bus/hid/drivers/razerkbd/0003:1532:021E.0002/:
total 0
-r--r--r-- 1 root plugdev 4096 sept.  1 11:50 country
lrwxrwxrwx 1 root plugdev    0 sept.  1 11:50 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 sept.  1 11:50 hidraw
drwxr-xr-x 3 root plugdev    0 sept.  1 11:50 input
-rw-rw---- 1 root plugdev 4096 sept.  1 11:50 key_alt_f4
-rw-rw---- 1 root plugdev 4096 sept.  1 11:50 key_alt_tab
-rw-rw---- 1 root plugdev 4096 sept.  1 11:50 key_super
-r--r--r-- 1 root plugdev 4096 sept.  1 11:50 modalias
drwxr-xr-x 2 root plugdev    0 sept.  1 11:50 power
-r--r--r-- 1 root plugdev 4096 sept.  1 11:50 report_descriptor
lrwxrwxrwx 1 root plugdev    0 sept.  1 11:50 subsystem -> ../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 sept.  1 11:50 uevent

/sys/bus/hid/drivers/razerkbd/0003:1532:021E.0003/:
total 0
-r--r--r-- 1 root plugdev 4096 sept.  1 11:50 country
-rw-rw---- 1 root plugdev 4096 sept.  1 11:53 device_mode
-r--r----- 1 root plugdev 4096 sept.  1 11:50 device_serial
-r--r----- 1 root plugdev 4096 sept.  1 11:50 device_type
lrwxrwxrwx 1 root plugdev    0 sept.  1 11:50 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
-r--r----- 1 root plugdev 4096 sept.  1 11:50 firmware_version
-rw-rw---- 1 root plugdev 4096 sept.  1 11:50 game_led_state
drwxr-xr-x 3 root plugdev    0 sept.  1 11:50 hidraw
drwxr-xr-x 3 root plugdev    0 sept.  1 11:50 input
-rw-rw---- 1 root plugdev 4096 sept.  1 11:50 macro_led_effect
-rw-rw---- 1 root plugdev 4096 sept.  1 11:50 macro_led_state
-rw-rw---- 1 root plugdev 4096 sept.  1 11:53 matrix_brightness
--w--w---- 1 root plugdev 4096 sept.  1 11:50 matrix_custom_frame
--w--w---- 1 root plugdev 4096 sept.  1 11:50 matrix_effect_breath
--w--w---- 1 root plugdev 4096 sept.  1 11:50 matrix_effect_custom
--w--w---- 1 root plugdev 4096 sept.  1 11:50 matrix_effect_none
--w--w---- 1 root plugdev 4096 sept.  1 11:50 matrix_effect_reactive
--w--w---- 1 root plugdev 4096 sept.  1 11:53 matrix_effect_spectrum
--w--w---- 1 root plugdev 4096 sept.  1 11:50 matrix_effect_starlight
--w--w---- 1 root plugdev 4096 sept.  1 11:55 matrix_effect_static
--w--w---- 1 root plugdev 4096 sept.  1 11:50 matrix_effect_wave
-r--r--r-- 1 root plugdev 4096 sept.  1 11:50 modalias
drwxr-xr-x 2 root plugdev    0 sept.  1 11:50 power
-r--r--r-- 1 root plugdev 4096 sept.  1 11:50 report_descriptor
lrwxrwxrwx 1 root plugdev    0 sept.  1 11:50 subsystem -> ../../../../../../../../bus/hid
-rw-rw---- 1 root plugdev 4096 sept.  1 11:50 test
-rw-r--r-- 1 root plugdev 4096 sept.  1 11:50 uevent
-r--r----- 1 root plugdev 4096 sept.  1 11:50 version
m1roh@m1roh-HP:/$ id
uid=1000(m1roh) gid=1000(m1roh) groupes=1000(m1roh),4(adm),24(cdrom),27(sudo),30(dip),46(plugdev),121(lpadmin),131(sambashare)
`

terrycain, 2017-09-01:
```
cd /sys/bus/hid/drivers/razerkbd/0003:1532:021E.0003/
echo -n '64' > ./matrix_brightness
echo -n '128' > ./matrix_brightness
echo -n '255' > ./matrix_brightness
```
That "should" change the brightness of the kb

m1roh, 2017-09-01:
unfortunately not...

terrycain, 2017-09-01:
Odd,

```
cat ./device_mode
cat ./device_serial
cat ./device_type
cat ./firmware_version
```

m1roh, 2017-09-01:
`m1roh@m1roh-HP:/sys/bus/hid/drivers/razerkbd/0003:1532:021E.0003$ cat ./device_mode
3:0
m1roh@m1roh-HP:/sys/bus/hid/drivers/razerkbd/0003:1532:021E.0003$ cat ./device_serial
SI1704F12003442
m1roh@m1roh-HP:/sys/bus/hid/drivers/razerkbd/0003:1532:021E.0003$ cat ./device_type
Razer Ornata Chroma
m1roh@m1roh-HP:/sys/bus/hid/drivers/razerkbd/0003:1532:021E.0003$ cat ./firmware_version
v1.0
`

terrycain, 2017-09-01:
odd, I'll check my ornata chroma over the weekend if i can

m1roh, 2017-09-01:
Ok thank you for your answers, the time you spend to try solve my problem and your work... And excuse me for my english, I'm a french guy...

m1roh, 2017-09-02:
I finally solved the problem. In fact the last time I turned off my computer before deleting Windows, I was logged on my razer account with my "non-lighting profile". So it's look like something, somewhere, remembered that. So I just pluggged my keyboard on a windows computer, installed the official driver, logged on my razer synapse account and move on my "lighting profile", then I reppluged this on my ubuntu laptop and... magic happens ! It works !!!!! One more time I really want to thank you, I close the issue... :+1: :+1: :+1:

closed 2017-09-02 by m1roh