openrazer/openrazer #374: Daemon is not responding!

LinoBla, 2017-09-01:
I have installed everything as described,"echo -n "\xFF\x5D\x5D\x19" > matrix_effect_static" also works, but I can't start the "razerCommander". This returns the error "ERROR: the daemon is not responding!
Try running `killall razer-service && razer-service` or rebooting. If this doesn't work, please fill an issue!"..."

After a "killall razer-service && razer-service" also didn't bring success.

terrycain, 2017-09-01:
`killall razer-daemon`
Then
`razer-daemon -Fv`
It will run in foreground mode, what does it output?

LinoBla, 2017-09-01:
lino@19:~/$ killall razer-daemon
razer-daemon: Kein Prozess gefunden
lino@19:~/$razer-daemon -Fv
razer-daemon: Befehl nicht gefunden.
lino@19:~/$Downloads/razer-drv/razerCommander-master$

terrycain, 2017-09-01:
Are you running the latest version

LinoBla, 2017-09-01:
I think so, yes.

lino@19:/$ apt list | grep razer

WARNING: apt does not have a stable CLI interface yet. Use with caution in scripts.

python3-razer/trusty,now 1.1.11-0ubuntu1 all [installed]
razer-daemon/trusty,now 1.1.11-0ubuntu1 amd64 [installed]
razer-doc/trusty,now 1.1.11-0ubuntu1 all [installed]
razer-kernel-modules-dkms/trusty,now 1.1.11-0ubuntu1 amd64 [installed]
razercommander/now 1.1.0 all [installed,local]

terrycain, 2017-09-01:
1.1.15 is out, please install that reboot and see if the issue still occurs

LinoBla, 2017-09-01:
I'm gonna try it.

z3ntu, 2017-09-01:
I think you're using Ubuntu 14.04?

z3ntu, 2017-09-24:
Closing as the last response was 23 days ago.

closed 2017-09-24 by z3ntu