openrazer/openrazer #375: Razer Naga Chroma side keys not working and throwing errors

Marahin, 2017-09-01:
# What's wrong? 

After the update, when the daemon is on, the side keys don't work. 

# Logs

```
2017-09-01 21:28:55 | razer.device1                  | DEBUG    | DBus call get_firmware
2017-09-01 21:28:55 | razer.device1                  | DEBUG    | DBus call version
2017-09-01 21:28:55 | razer.device1                  | DEBUG    | DBus call has_matrix
2017-09-01 21:28:55 | razer.device1                  | DEBUG    | DBus call has_matrix
2017-09-01 21:28:55 | razer.device0                  | DEBUG    | DBus call get_logo_brightness
2017-09-01 21:28:55 | razer.device0                  | DEBUG    | DBus call get_scroll_brightness
2017-09-01 21:28:58 | razer.device1                  | DEBUG    | DBus call get_brightness
2017-09-01 21:28:59 | razer.device0                  | DEBUG    | DBus call get_logo_brightness
2017-09-01 21:28:59 | razer.device0                  | DEBUG    | DBus call get_scroll_brightness
2017-09-01 21:28:59 | razer.device0                  | DEBUG    | DBus call get_poll_rate
2017-09-01 21:30:15 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 11
2017-09-01 21:31:18 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 11
2017-09-01 21:31:34 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 9
2017-09-01 21:31:35 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 9
2017-09-01 21:31:35 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 10
2017-09-01 21:31:38 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 10
2017-09-01 21:31:38 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 11
2017-09-01 21:31:39 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 11
2017-09-01 21:31:39 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 12
2017-09-01 21:31:41 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 12
2017-09-01 21:31:41 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 13
2017-09-01 21:31:43 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 13
2017-09-01 21:31:51 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 9
2017-09-01 21:31:53 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 9
2017-09-01 21:31:55 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 9
2017-09-01 21:31:57 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 9
```

# Possible cause

@terrycain has brought up an idea of what could be the reason [here](https://github.com/openrazer/openrazer/issues/59#issuecomment-326673985).

# Environment

Ubuntu 17.04 64-bit. 

# Misc.

Everything else (setting the DPI, brightness, etc.) works marvelous, thank you for the amazing work and bringing compatibility to Linux! :-) :+1:

terrycain, 2017-09-03:
Kill the daemon with pkill razer-daemon

Clone the repo, there is a file `scripts/misc/event_reader.py`, run that like
`scripts/misc/event_reader.py /dev/input/by-id/usb-Razer_Razer_Naga_Chroma-if01-event-kbd`
press all the number buttons and save the output
`scripts/misc/event_reader.py /dev/input/by-id/usb-Razer_Razer_Naga_Chroma-if02-event-kbd`
and do the same with that command

It should react when you press the number buttons.

Marahin, 2017-09-03:
I will do that in the nearest future.

MorningStarGG, 2017-09-11:
This is currently happening for me as well. I also use Polychromatic and when I go to the mouse config. "There was a problem communicating with the daemon for this device." - I was using a Razer Naga 2014 and had no issues, but with the Naga Chroma, I have both of these issues. I'm guessing they are somehow related?

wenzler, 2017-12-19:
Ran the event_reader.py on Ubuntu16.04 and it shows no output for both if01 and if02 when pressing the side buttons.
Killing the openrazer-daemon gives the default 1,2,3...

z3ntu, 2017-12-19:
@wenzler You could try to apply the following diff: https://github.com/openrazer/openrazer/commit/d1e683ac7cbaf0a70c6be1efc33d142f6c26ac72 to the installed files. A new release containing this should be out in the next couple of days.

wenzler, 2017-12-19:
Thanks @z3ntu - this works ok-ish (excl tilt wheel) with https://github.com/apocatarsis/Naga_KeypadMapper
Yet, looking at the patch I'm wondering how the key mapping is supposed to be done after all. 
In xorg/xmodmap? I’m not a bit wiser after a day of duckduck-ing :/

z3ntu, 2018-01-28:
d1e683a is in a release.

closed 2018-01-28 by z3ntu