openrazer/openrazer #376: Gigabyte Aero 15W

Queatz, 2017-09-04:
Any chance of support for the Gigabyte Aero 15W?

terrycain, 2017-09-04:
I guess we can give it a shot, for starters output of `lsusb` and screenshots of whatever windows program controls it

Queatz, 2017-09-04:
Sweet.

`lsusb` yields

```
jacob@debian:~$ lsusb
Bus 002 Device 006: ID 0bda:0316 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 1bcf:2c6b Sunplus Innovation Technology Inc. 
Bus 001 Device 004: ID 04d9:8008 Holtek Semiconductor, Inc. 
Bus 001 Device 003: ID 8087:0a2b Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

I'll see if I can find some screenshots from the Windows controller.  I remember it had a simple pattern selector along with a single color chooser.

terrycain, 2017-09-04:
you'll most likely need the windows app, as were going to need to reverse engineer how it works

also output of `lspci`

Queatz, 2017-09-04:
Got it, shouldn't be too hard to get Windows up again.

`lspci`

```
jacob@debian:~$ lspci
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 05)
00:01.0 PCI bridge: Intel Corporation Skylake PCIe Controller (x16) (rev 05)
00:02.0 VGA compatible controller: Intel Corporation Device 591b (rev 04)
00:08.0 System peripheral: Intel Corporation Skylake Gaussian Mixture Model
00:14.0 USB controller: Intel Corporation Sunrise Point-H USB 3.0 xHCI Controller (rev 31)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-H Thermal subsystem (rev 31)
00:16.0 Communication controller: Intel Corporation Sunrise Point-H CSME HECI #1 (rev 31)
00:17.0 SATA controller: Intel Corporation Sunrise Point-H SATA Controller [AHCI mode] (rev 31)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-H PCI Express Root Port #1 (rev f1)
00:1c.2 PCI bridge: Intel Corporation Sunrise Point-H PCI Express Root Port #3 (rev f1)
00:1c.4 PCI bridge: Intel Corporation Sunrise Point-H PCI Express Root Port #5 (rev f1)
00:1f.0 ISA bridge: Intel Corporation Sunrise Point-H LPC Controller (rev 31)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-H PMC (rev 31)
00:1f.3 Audio device: Intel Corporation Device a171 (rev 31)
00:1f.4 SMBus: Intel Corporation Sunrise Point-H SMBus (rev 31)
01:00.0 VGA compatible controller: NVIDIA Corporation GP106M [GeForce GTX 1060] (rev ff)
02:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 16)
03:00.0 Network controller: Intel Corporation Wireless 8265 / 8275 (rev 78)
```

terrycain, 2017-09-04:
Hmm, I didnt see the keyboard listed there, we shall find it the old fashioned way. 

`ls -lah /dev/input/*`

Queatz, 2017-09-04:
Looks like it's in there

```
jacob@debian:~$ ls -lah /dev/input/*
crw-rw---- 1 root input 13, 64 Sep  3 22:10 /dev/input/event0
crw-rw---- 1 root input 13, 65 Sep  3 22:10 /dev/input/event1
crw-rw---- 1 root input 13, 74 Sep  3 22:10 /dev/input/event10
crw-rw---- 1 root input 13, 75 Sep  3 22:10 /dev/input/event11
crw-rw---- 1 root input 13, 76 Sep  3 22:10 /dev/input/event12
crw-rw---- 1 root input 13, 77 Sep  3 22:10 /dev/input/event13
crw-rw---- 1 root input 13, 78 Sep  3 22:10 /dev/input/event14
crw-rw---- 1 root input 13, 79 Sep  3 22:10 /dev/input/event15
crw-rw---- 1 root input 13, 80 Sep  3 22:10 /dev/input/event16
crw-rw---- 1 root input 13, 81 Sep  3 22:10 /dev/input/event17
crw-rw---- 1 root input 13, 82 Sep  3 22:10 /dev/input/event18
crw-rw---- 1 root input 13, 66 Sep  3 22:10 /dev/input/event2
crw-rw---- 1 root input 13, 67 Sep  3 22:10 /dev/input/event3
crw-rw---- 1 root input 13, 68 Sep  3 22:10 /dev/input/event4
crw-rw---- 1 root input 13, 69 Sep  3 22:10 /dev/input/event5
crw-rw---- 1 root input 13, 70 Sep  3 22:10 /dev/input/event6
crw-rw---- 1 root input 13, 71 Sep  3 22:10 /dev/input/event7
crw-rw---- 1 root input 13, 72 Sep  3 22:10 /dev/input/event8
crw-rw---- 1 root input 13, 73 Sep  3 22:10 /dev/input/event9
crw-rw-r-- 1 root input 13,  0 Sep  3 22:10 /dev/input/js0
crw-rw---- 1 root input 13, 63 Sep  3 22:10 /dev/input/mice
crw-rw---- 1 root input 13, 32 Sep  3 22:10 /dev/input/mouse0
crw-rw---- 1 root input 13, 33 Sep  3 22:10 /dev/input/mouse1

/dev/input/by-id:
total 0
drwxr-xr-x 2 root root 120 Sep  3 22:10 .
drwxr-xr-x 5 root root 560 Sep  3 22:10 ..
lrwxrwxrwx 1 root root  10 Sep  3 22:10 usb-27C40-06N20-L01S65Q549_HD_WebCam-event-if00 -> ../event17
lrwxrwxrwx 1 root root  10 Sep  3 22:10 usb-HOLTEK_USB-HID_Keyboard_AP0000000003-event-kbd -> ../event16
lrwxrwxrwx 1 root root  10 Sep  3 22:10 usb-HOLTEK_USB-HID_Keyboard_AP0000000003-if02-event-mouse -> ../event18
lrwxrwxrwx 1 root root   6 Sep  3 22:10 usb-HOLTEK_USB-HID_Keyboard_AP0000000003-if02-mouse -> ../js0

/dev/input/by-path:
total 0
drwxr-xr-x 2 root root 140 Sep  3 22:10 .
drwxr-xr-x 5 root root 560 Sep  3 22:10 ..
lrwxrwxrwx 1 root root  10 Sep  3 22:10 pci-0000:00:14.0-usb-0:11:1.0-event-kbd -> ../event16
lrwxrwxrwx 1 root root  10 Sep  3 22:10 pci-0000:00:14.0-usb-0:11:1.2-event-mouse -> ../event18
lrwxrwxrwx 1 root root   6 Sep  3 22:10 pci-0000:00:14.0-usb-0:11:1.2-mouse -> ../js0
lrwxrwxrwx 1 root root  10 Sep  3 22:10 pci-0000:00:14.0-usb-0:9:1.0-event -> ../event17
lrwxrwxrwx 1 root root   9 Sep  3 22:10 platform-pcspkr-event-spkr -> ../event7

/dev/input/ckb0:
total 12K
drwxr-xr-x 2 root root 100 Sep  3 22:10 .
drwxr-xr-x 5 root root 560 Sep  3 22:10 ..
-rw-r--r-- 1 root root   1 Sep  3 22:10 connected
-rw-r--r-- 1 root root   4 Sep  3 22:10 pid
-rw-r--r-- 1 root root   7 Sep  3 22:10 version
```

terrycain, 2017-09-04:
Ahh that explains the `Bus 001 Device 004: ID 04d9:8008 Holtek Semiconductor, Inc. `
I googled the `04d9:8008` which normally brings back chinese vendor results but got rubbish for that one.

Yeah nothing much more we can do without being able to poke it with the windows app

terrycain, 2017-09-04:
But, if you ran virtualbox, you could pass through that usb device to the windows vm which would actually be far more helpful and probably easier to set up

terrycain, 2017-09-04:
If possible, its much easier to get the data from a vm :smile:

Queatz, 2017-09-04:
Oh cool, I will install a Windows VM when I get home and see what I can do

Queatz, 2017-09-06:
Hi @terrycain, so I was able to get the Virtual Box up and run Gigabyte's provided software to control the keyboard, however it doesn't seem to work through the VM.  Would it still be helpful in that case?

terrycain, 2017-09-06:
Did you pass the USB keyboard through to the vm, and it showing in device manager?

Queatz, 2017-09-06:
I am not seeing any devices in the Devices menu (it is the laptop's builtin keyboard.) I did install the Guest Additions.

terrycain, 2017-09-06:
install teamviewer 12 on the linux part and ill have a look

Queatz, 2017-09-06:
Installed.  I'm available from 6pm CST today or we can set up a time that works

terrycain, 2017-09-06:
Am in the UK so 6pm CST will be midnight, should be able to do 6pm cst tomorrow

Queatz, 2017-09-06:
Cool, I can also do mornings here ex. 1pm UTC if that is better

terrycain, 2017-09-06:
Can do 1pm utc on friday if thats alright? That would be easier for me

Queatz, 2017-09-06:
Yup, let's do that

terrycain, 2017-09-06:
Have added to calendar. If you have/will get telegram add me on that "@terrycain" and i'll add you to the razer group

z3ntu, 2018-01-28:
I'm gonna close this for now as supporting non-Razer devices isn't really the scope of this project. 😉

closed 2018-01-28 by z3ntu

terrycain, 2018-01-28:
@z3ntu But just for reference I haven't forgot about it :D