openrazer/openrazer #377: Razer Naga 2012 Support

minecrawler, 2017-09-06:
Will the [Razer Naga 2012](https://www2.razerzone.com/eu-en/gaming-mice/razer-naga-2012) work with OpenRazer (it already supports the Razer Naga 2014, so just maybe...)?

[image]

I cannot test it at the moment, however I could not find any notes searching the issues. I don't know the exact differences between Razer Naga 2012 and Razer Naga 2014.

If you tell me what to do and what to test, I can do so tonight, so you might be able to add that device to the supported mice list :)

terrycain, 2017-09-06:
If the USB PID is in the list then yes. Read - https://github.com/openrazer/openrazer#determining-the-device-id

minecrawler, 2017-09-06:
Well, the PID is `002E`. So for now, it is not in the list.

z3ntu, 2017-09-06:
* Please post the output of
`lsusb -d 1532:002e -v`
and
`ls -lah /dev/input/by-id/`
* Read [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol) because we probably need packet captures.
* Screenshots of all available options on Razer Synapse are needed.
* And lastly, the product & model number of the device for documentation reasons please.

minecrawler, 2017-09-06:
| Key | Value |
| --- | --- |
| Product | [Razer Naga 2012](https://www2.razerzone.com/eu-en/gaming-mice/razer-naga-2012) |
| Model No.  | RZ01-0058, Gaming Mouse |
| Product No. | RZ01-00580100 |

Let me post the screenshots first, since I am on Windows atm.

[image]

[image]

[image]

[image]

terrycain, 2017-09-06:
Ok I need packet captures  of 
* Moving dpi from min to max
* logo on and off a few times
* scroll wheel on and off a few times
* poll rate all options

minecrawler, 2017-09-06:
Well, I tried to make sense of the capture files, but they are a mess of too many packets to understand them in a short amount of time :/

I didn't do the full dpi spectrum. I selected the values from 100 to 1000. As for the scroll wheel and logo, I started by turning them off, then on, etc. The Poll rates are 125, 500 and 1000.

Could you check if those captures are good enough for you? Do you need anything else? Is there a way I can further help you get that mouse added? You can download all the files from my [OneDrive folder](https://1drv.ms/f/s!AjlXArG494TzgtwCQ9FP4yGPElTeHQ).

minecrawler, 2017-09-06:
```
# ls -lah /dev/input/by-id/ |grep Naga
lrwxrwxrwx 1 root root   9 Sep  6 19:59 usb-Razer_Razer_Naga-event-mouse -> ../event5
lrwxrwxrwx 1 root root   9 Sep  6  2017 usb-Razer_Razer_Naga-if01-event-kbd -> ../event6
lrwxrwxrwx 1 root root   9 Sep  6  2017 usb-Razer_Razer_Naga-mouse -> ../mouse1
```

```
# lsusb -d 1532:002e -v

Bus 002 Device 003: ID 1532:002e Razer USA, Ltd RZ01-0058 Gaming Mouse [Naga]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x002e RZ01-0058 Gaming Mouse [Naga]
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Naga
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     103
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0009  1x 9 bytes
        bInterval               8
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

crisvuchs, 2017-09-21:
So is there an eta to when the Razer Naga 2012 support will be added?

minecrawler, 2017-09-21:
@crisvuchs since this project is FOSS and people are working on it as a hobby, any feature is added when they have time - either that or you do it yourself ;)

z3ntu, 2017-09-21:
@minecrawler Thanks
@crisvuchs No

BaaRamEwe, 2018-04-19:
I don't know if it is any help mentioning [razercfg](http://bues.ch/cms/hacking/razercfg.html), which can change the DPI, poll rate, and LEDs for the Naga 2012, though not button remapping.

Since it appears to use the [same code](https://github.com/mbuesch/razer/blob/master/librazer/hw_naga.c) between the Naga 2012 and the Naga Hex, which openrazer already supports, maybe the same can be attempted here.

Adi118, 2018-04-20:
@BaaRamEwe I am able to change the DPI poll rate and leds for naga 2014 as well but how do I remap the butttons all they do right now is enter numbers 1-7

BaaRamEwe, 2018-04-20:
@Adi118 Not currently doable using razercfg, according to its page. (My mouse is a Naga 2012, so that's all I can test)
I had thought that openrazer was able to remap its supported mice, including Naga 2014?

Adi118, 2018-04-23:
@BaaRamEwe I couldn't find it, I installed one the drivers and the gui app and all i could do was change color. If you find any links to remap the button let me know

commit f2a6f59a (2020-01-05): Add support for Razer Naga 2012

Fixes #377
closed 2020-01-05 by z3ntu