openrazer/openrazer #378: How to install this on debian?

phspade, 2017-09-07:
Hello any ideas on how to install.. any steps? my OS is Parrot Security which is debian and my KB is Razer Ornata and Razer Blackwidow X Chroma Mercury Edition.

z3ntu, 2017-09-07:
Instructions are on the website: https://openrazer.github.io/#debian

leosegovia, 2017-09-07:
You need to install `openrazer-meta_1.1.16-0ubuntu1_all.deb` from the link provided in the instructions or by running the following instructions:
Install some dependencies first before installing **openrazer-meta**:
- **dbus-user-session**
  `sudo apt-get install dbus-user-session -y`
- **razer-kernel-modules-dkms**
  `wget http://download.opensuse.org/repositories/hardware:/razer/Debian_9.0/amd64/razer-kernel-modules-dkms_1.1.16-0ubuntu1_amd64.deb && sudo dpkg -i razer-kernel-modules-dkms_1.1.16-0ubuntu1_amd64.deb`

After installing those, you should be ready to install **openrazer-meta** by running `sudo dpkg -i openrazer-meta_1.1.16-0ubuntu1_all.deb`

z3ntu, 2017-09-07:
@leosegovia If there are missing dependencies, please let me know. Otherwise the instructions on the website are completely sufficient.

leosegovia, 2017-09-07:
**Debian 8.6**
```
root@somecomputer:~# lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 8.6 (jessie)
Release:        8.6
Codename:       jessie
```

***Installing:***
```
root@somecomputer:~# dpkg -i razer-daemon_1.1.16-0ubuntu1_amd64.deb 
Selecting previously unselected package razer-daemon.
(Reading database ... 30981 files and directories currently installed.)
Preparing to unpack razer-daemon_1.1.16-0ubuntu1_amd64.deb ...
Unpacking razer-daemon (1.1.16-0ubuntu1) ...
dpkg: dependency problems prevent configuration of razer-daemon:
 razer-daemon depends on razer-kernel-modules-dkms; however:
  Package razer-kernel-modules-dkms is not installed.
 razer-daemon depends on python3; however:
  Package python3 is not installed.
 razer-daemon depends on python3-dbus; however:
  Package python3-dbus is not installed.
 razer-daemon depends on python3-gi; however:
  Package python3-gi is not installed.
 razer-daemon depends on python3-pyudev; however:
  Package python3-pyudev is not installed.
 razer-daemon depends on python3-evdev; however:
  Package python3-evdev is not installed.
 razer-daemon depends on python3-setproctitle; however:
  Package python3-setproctitle is not installed.
 razer-daemon depends on python3-notify2; however:
  Package python3-notify2 is not installed.
 razer-daemon depends on gir1.2-gtk-3.0; however:
  Package gir1.2-gtk-3.0 is not installed.
 razer-daemon depends on xautomation; however:
  Package xautomation is not installed.
 razer-daemon depends on xdotool
dpkg: error processing package razer-daemon (--install):
 dependency problems - leaving unconfigured
Processing triggers for man-db (2.7.0.2-5) ...
Errors were encountered while processing:
 razer-daemon
```
Most dependencies can be fixed with simple `apt-get -f install` but **openrazer-meta** still requires **razer-kernel-modules-dkms**, but to correct my instructions, **if you install by adding repo to sources, not issues will be found and instructions are completely correct.**

z3ntu, 2017-09-07:
Yeah sure, dpkg doesn't really care about dependencies (which then gets fixed by `apt-get install -f` as you said), adding the repo is of course the recommended way as you get updates and it's easier to install.

z3ntu, 2017-09-07:
And yes, you also have to install the kernel modules (openrazer-meta is a meta package for the 3 packages built from this project)

closed 2017-09-07 by z3ntu

leosegovia, 2017-09-07:
I'll remove my comment since you are completely correct