openrazer/openrazer #379: razer-daemon trouble

phspade, 2017-09-08:
[image]

i tried to upload this amd64 and i386 of razer-daemon but it always failed. which i dont know why :(

z3ntu, 2017-09-08:
please use the repository and do not invoke dpkg manually but use apt.

closed 2017-09-08 by z3ntu

phspade, 2017-09-08:
i cant add repository thou parrot sec restrict adding a repository..
>`do not invoke dpkg manually but use apt.`

What do u mean by that?

z3ntu, 2017-09-08:
I meant that you should install it via the repository=apt. Anyways then you should download every single Deb manually and install them with dpkg. Then run the apt install -f command to get the rest of the dependencies

z3ntu, 2017-09-08:
You can find all files at http://download.opensuse.org/repositories/hardware:/razer/ (use the correct Debian version please)