openrazer/openrazer #381: DBus interface should see some rework

whot, 2017-09-11:
I'm trying to hook up openrazer to [libratbag](http://github.com/libratbag/libratbag). The purpose of this is to have a generic interface to configure gaming mice (and eventually keyboards). libratbag is designed that we can use other projects like openrazer to provide the actual driver interface, 

The current WIP is available here: https://github.com/whot/libratbag/tree/wip/openrazer-driver
It needs #380 to work though.

There are a few API design issues I found with the DBus interface, it'd be good to get a conversation started on how to fix those. Related documentation are the [DBus API Design Guidelines](https://dbus.freedesktop.org/doc/dbus-api-design.html)

- [ ] The interface names aren't in the recommended URL format but the object paths are. So we have `/org/razer` as object path but `razer.daemon` as interface name. The latter should be `org.razer.daemon`, and then the same for the other interfaces.
- [ ] I don't quite understand why there's `razer.daemon` and `razer.devices` rather than that being one interface, but that's a minor point and can be ignored
- [ ] `razer.devices.getDevices()` returns a serial number string. This string must then be used to construct an object path `/org/razer/device/$serial`. This makes the object path part of the API and hinders future refactoring. Recommended is to return a full object path so that the returned string can directly be used to instantiate the next object.
- [ ] No use of properties. Especially those bits that are constant anyway and cannot be set by the user could be provided as DBus properties. e.g. `razer.device.misc.getDeviceName`. 
- [ ] The `razer.device.dpi` interface uses `int32` for the getters but `uint16` for the setters
- [ ] `razer.device.dpi.getDPI` returns an array, but `setDPI` takes two arguments  - why not return a tuple from `getDPI`?
- [ ] LEDs have the interface name in the method name, e.g. `getLogoEffect` vs `getScrollEffect`. This is unnecessary and makes writing clients harder than need be. [1] There's no benefit from that that I can see, we're in two different interfaces anyway.
- [ ] LEDs are in separate interfaces `razer.device.lighting.logo` and `razer.device.lighting.scroll` and both are on the same device. This would be better solved by having a `razer.device.getLEDs()` function that returns object paths to the entities implementing a more generic `razer.device.lighting` interface.
- [ ] LEDs don't seem to a have a `getColor()` or similar request to get the current colors.
- [ ] The `razer.device.misc` interface seems superfluous. If it doesn't fit into the others, it could just be `razer.device` but it's strange to have subinterfaces but not the parent interface. And having  things like `resumeDevice()` in a `misc` interface. Personally I'd move the direct commands like resume/suspend up to `razer.device` and maybe leave the various information bits in the `misc` interface

I think that's about it, at least that's what I found writing the driver which does the most basic things like handling DPI, poll rate and setting LEDs. The above are all fixable with relatively simple effort but I do realize that any change in the DBus API would break any client relying on the current API.

[1] You can get to all this using introspection, but in the case of libratbag now you're running an XML parser in a process running as root just to get the method name. That's less than ideal. For now, I've resorted to just `snprintf()` the method names.

terrycain, 2017-09-11:
Yeah all makes sense, and if I thought of it at the time, I would of returned the full object path :wink:, I think the bulk of it was written after like 2am in 1 night.

Properties definitely sound good too.

The majority of API uses use the python client already in the repo, so technically no breakage should happen there. The only exception to the rule is @z3ntu but he should be able to cope with the changes.

z3ntu, 2017-09-11:
As long as not everything will be changed it's all right :)

Zerophase, 2017-10-11:
I'm going to start looking into this over the coming week. As long as, I update affected portions of the python client nothing should break, right?

whot, 2017-10-11:
As long as every current client uses the python bindings, yes, that's correct.

Having said that: I looked at this a few weeks back and it's a lot of work. See #385 and #388 as just the start, I eventually gave up because until those are pulled, there's not much point piling more patches on because any changes to those will get exponentially harder.

Current state of my work is here: https://github.com/whot/openrazer/tree/wip/dbus-revamp-v3 but note it's WIP. The biggest issue I ran into was: python-dbus doesn't allow for the same method to be hooked up twice for different objects because it somehow hooks into the properties, etc. So with the current design, it's impossible to e.g. change `getScrollFoo` and `getLogoFoo` to just `getFoo` on both interfaces. To get to that point, we'd have to really reshape a lot of the daemon so it uses separate objects that register themselves, etc. And that's also when I figured out that there was still some hw-specific knowledge in the daemon (e.g. #391) which would have to be cleaned up first.

terrycain, 2017-10-11:
I plan to go through those PRs saturday.

> So with the current design, it's impossible to e.g. change getScrollFoo and getLogoFoo to just getFoo on both interfaces.

Yeah I encountered that long ago, I did have some proof of concept code to do it, but it was at least a year ago so probably lost. It gets pretty nasty when you get to that part of the python dbus.

Zerophase, 2017-10-12:
I'll gladly go through and clean the rest up once those patches get pulled. Only thing is I don't have a Blade to test on.  I only have the Anansi, which lacks a lot of the features needed for the newer keyboards.

closed 2018-03-11 by stale[bot]

perigoso, 2021-10-08:
bumping, this should get a `(not stale)` label