openrazer/openrazer #383: How to save Macro Keys permanent?

sangdth, 2017-09-11:
I've installed latest openrazer, it works great and smoothly. The only thing is, every time I restart, all macro keys I have set with M1 - M5 keys are gone, and I have to record again.

I tried to search google and also this place, but have no result. Could anybody show me how to store them permanent, or even some workaround way?

Thanks.

--
Razer Ultimate classic 2014 keyboard.
Openrazer v1.1.14

sangdth, 2017-09-18:
I knew how to do it. Close this.

closed 2017-09-18 by sangdth

Melvin-Vinvu, 2018-08-08:
Is there another place that I can read your post? Your blog seems to be down.

crazyquark, 2018-10-28:
Could some post some details here? The blog post is a dead link.

blshkv, 2020-07-16:
Could not find the blog post, so spent few hours to come up with this. Hopefully, it would help someone

```
#!/usr/bin/env python3

from openrazer.client import DeviceManager
import openrazer_daemon.misc.macro as _daemon_macro

device_manager = DeviceManager()
//wild guess
device=device_manager.devices[0]

new_macro=[]
new_macro.append(_daemon_macro.MacroKey("T", "0", "DOWN"))
new_macro.append(_daemon_macro.MacroKey("T", "0", "UP"))
new_macro.append(_daemon_macro.MacroKey("e", "505998", "DOWN"))
new_macro.append(_daemon_macro.MacroKey("e", "505998", "UP"))
new_macro.append(_daemon_macro.MacroKey("s", "1594997", "DOWN"))
new_macro.append(_daemon_macro.MacroKey("s", "0", "UP"))
new_macro.append(_daemon_macro.MacroKey("t", "0", "DOWN"))
new_macro.append(_daemon_macro.MacroKey("t", "0", "UP"))
device.macro.add_macro("M5", new_macro)

```