openrazer/openrazer #390: additional_interfaces not filled in for hotplugged devices

whot, 2017-09-18:
951235d2065e41f017b382a14e4fa5ffc62d5d8b added support for additional USB interfaces but that code was only added to `_load_devices` - which is only run on startup. Devices' additional interfaces added at runtime with `_add_devices` are ignored.

z3ntu, 2017-10-20:
Do you have an example where not having that breaks something?

whot, 2017-10-23:
hmm, I can't remember how I found this, sorry. Shouldn't matter though, you get two different behaviours depending on whether the device is already plugged in or plugged in after the daemon is started. That shouldn't happen.

A git grep shows that additional_interfaces is only used in the chroma keyboard's `setGameMode` function. So I suspect it should be reproducible with one of those devices (I don't have one).

whot, 2017-10-23:
doh, wrong button, sorry.

z3ntu, 2018-01-28:
This is apparently also important for the ripple effect / macro / game mode functionality.
The problem is that the `KeyWatcher` doesn't get any event files which it needs for listening to keys. 
```
2018-01-28 18:30:47 | razer.device3                  | DEBUG    | Adding razer.device.lighting.custom.setRippleRandomColour method to DBus
now creating KeyWatcher...
2018-01-28 18:30:47 | razer.device3.keymanager       | WARNING  | No event files for KeyWatcher
2018-01-28 18:30:47 | razer.device3                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2018-01-28 18:30:47 | razer                          | DEBUG    | Emitted Device Added Signal
2018-01-28 18:30:47 | razer                          | DEBUG    | DBus called get_serial_list
2018-01-28 18:30:47 | razer                          | DEBUG    | Device event [add]: /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.2/3-2.2:1.1/0003:1532:0203.003D
adding additional_interfaces: /sys/devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.2/3-2.2:1.1/0003:1532:0203.003D
```
It seems that the code in your patch is called too late where the KeyWatcher was already created.

z3ntu, 2018-01-28:
If you have the time, please take a look at #516

commit 4e32493d (2018-01-28): Work around race condition when hotplugging

The problem is that udev throws 'add' events in a random order but the
current code needs the interface with our sysfs entries first, and then
the other interfaces which get added as 'additional interfaces'. Without
this, in most cases the interfaces get added in the wrong order,
resulting in ripple, game mode, macro functionality etc not working

Now the to-be-added devices get batched up and sorted and only then
really added.

A downside of this commit is that adding new devices is getting delayed by at least 2 seconds (maybe this time can be lowe [... cut]
PR #516 (2018-04-01): Work around race condition when hotplugging
The problem is that udev throws 'add' events in a random order but the
current code needs the interface with our sysfs entries first, and then
the other interfaces which get added as 'additional interfaces'. Without
this, in most cases the interfaces get added in the wrong order,
resulting in ripple, game mode, macro functionality etc not working

Now the to-be-added devices get batched up and sorted and only then
really added.

A downside of this commit is that adding new devices is getting delayed by at least 2 seconds (maybe this time can be lower [... cut]
commit 1f1f548a (2018-04-01): Work around race condition when hotplugging

The problem is that udev throws 'add' events in a random order but the
current code needs the interface with our sysfs entries first, and then
the other interfaces which get added as 'additional interfaces'. Without
this, in most cases the interfaces get added in the wrong order,
resulting in ripple, game mode, macro functionality etc not working

Now the to-be-added devices get batched up and sorted and only then
really added.

A downside of this commit is that adding new devices is getting delayed by at least 2 seconds (maybe this time can be lowe [... cut]
closed 2018-04-01 by z3ntu