openrazer/openrazer #394: Diamondback 3G Support

Styx85, 2017-09-19:
Hello, I have an old DiamondBack 3G  Razer mouse and I want to know if it is possible to add support for it.

> lsusb -d 1532:000d -v
> 
> Bus 008 Device 002: ID 1532:000d Razer USA, Ltd 
> Device Descriptor:
>   bLength                18
>   bDescriptorType         1
>   bcdUSB               2.00
>   bDeviceClass            0 
>   bDeviceSubClass         0 
>   bDeviceProtocol         0 
>   bMaxPacketSize0         8
>   idVendor           0x1532 Razer USA, Ltd
>   idProduct          0x000d 
>   bcdDevice           23.00
>   iManufacturer           1 Razer
>   iProduct                2 Razer Diamondback 3G
>   iSerial                 0 
>   bNumConfigurations      1
>   Configuration Descriptor:
>     bLength                 9
>     bDescriptorType         2
>     wTotalLength           34
>     bNumInterfaces          1
>     bConfigurationValue     1
>     iConfiguration          3 HID-compliant mouse
>     bmAttributes         0xa0
>       (Bus Powered)
>       Remote Wakeup
>     MaxPower              100mA
>     Interface Descriptor:
>       bLength                 9
>       bDescriptorType         4
>       bInterfaceNumber        0
>       bAlternateSetting       0
>       bNumEndpoints           1
>       bInterfaceClass         3 Human Interface Device
>       bInterfaceSubClass      1 Boot Interface Subclass
>       bInterfaceProtocol      2 Mouse
>       iInterface              0 
>         HID Device Descriptor:
>           bLength                 9
>           bDescriptorType        33
>           bcdHID               1.10
>           bCountryCode            0 Not supported
>           bNumDescriptors         1
>           bDescriptorType        34 Report
>           wDescriptorLength      79
>          Report Descriptors: 
>            ** UNAVAILABLE **
>       Endpoint Descriptor:
>         bLength                 7
>         bDescriptorType         5
>         bEndpointAddress     0x81  EP 1 IN
>         bmAttributes            3
>           Transfer Type            Interrupt
>           Synch Type               None
>           Usage Type               Data
>         wMaxPacketSize     0x0008  1x 8 bytes
>         bInterval              10
> can't get device qualifier: Resource temporarily unavailable
> can't get debug descriptor: Resource temporarily unavailable
> Device Status:     0x0000
>   (Bus Powered)
> 

> $ ls -lah /dev/input/by-id/
> insgesamt 0
> drwxr-xr-x 2 root root 140 19. Sep 16:55 .
> drwxr-xr-x 4 root root 440 19. Sep 16:55 ..
> lrwxrwxrwx 1 root root   9 19. Sep 16:55 usb-046a_010d-event-if01 -> ../event1
> lrwxrwxrwx 1 root root   9 19. Sep 16:55 usb-046a_010d-event-kbd -> ../event0
> lrwxrwxrwx 1 root root  10 19. Sep 16:55 usb-046d_0825_760585D0-event-if00 -> ../event15
> lrwxrwxrwx 1 root root   9 19. Sep 16:55 usb-Razer_Razer_Diamondback_3G-event-mouse -> ../event2
> lrwxrwxrwx 1 root root   9 19. Sep 16:55 usb-Razer_Razer_Diamondback_3G-mouse -> ../mouse0

And some jpg's from the driver
[image]
[image]
[image]

As well as the so called "Master Guide"
[DB3GOMG-ENG.pdf](https://github.com/openrazer/openrazer/files/1314841/DB3GOMG-ENG.pdf)

z3ntu, 2017-09-19:
Can you post the product & model number of the device for documentation reasons please?

Styx85, 2017-09-19:
Do you mean the one printed on the back of the mouse?
 It's

> Model Number: RZ01-0002, Gaming Mouse
> Product Code: RZ01-00020600-R3M1

z3ntu, 2017-09-19:
Yes, thanks

terrycain, 2017-11-08:
I could do with some pcaps of the driver if thats possible. https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

Styx85, 2017-11-19:
I tried my very best and put the capture files [here](https://github.com/Styx85/RazerDaimondback3GPacketCapture.git).

Are these correct? Hope that helps.

x2Fer, 2020-06-17:
sorry for unearthing this ancient issue but i was just wondering whether anything else is needed beyond the files styx provided. i've got the same model mouse and was hoping to be able to use this. thanks!

z3ntu, 2020-07-07:
@x2Fer It's using a super old protocol which isn't compatible with the newer devices so our tools don't work with that. Someone will have to manually go through the packets and figure out what's happening.