openrazer/openrazer #397: No dependency loop with yaourt any more

simonszu, 2017-09-26:
Hi,

the website informs me that i should rather use pacaur instead of yaourt on Arch Linux to install `openrazer-meta`, due to an dependency loop.

I tried with a fresh Arch install and up-to-date package, and the installation with yaourt went without any problems. So, maybe i am able to tell you that this problem doesn't occur any more and that the installation with yaourt is possible again.

z3ntu, 2017-09-26:
Was there a yaourt update since [May](https://github.com/openrazer/openrazer/issues/244#issuecomment-299528213)?

simonszu, 2017-09-26:
Yes, yaourt 1.9 was released on July 19th, and the last yaourt buildfile update in the Arch User Repository is from September 19th.

z3ntu, 2017-09-26:
Yaourt is still generally a bad AUR helper but we'll remove it soon from the website. But if there's one more report about yaourt not working, I'll put it back up 😉

closed 2017-09-26 by z3ntu