openrazer/openrazer #398: 2016 Blade keyboard has 16 columns, not 15

calder, 2017-09-27:
On the 2016 Blade keyboard, the right arrow key is row 5 col 15, but the openrazer Python library thinks the keyboard's dimensions are 6x15. This prevents setting the right arrow key's color with
```python
device.fx.advanced.matrix[5,15] = ...
```
and means its color isn't cleared by
```python
d.fx.advanced.matrix.reset()
```

After executing
```python
device.fx.advanced.matrix = openrazer.client.fx.Frame((6,16))
```
both manual color setting and `reset()` work as expected.

terrycain, 2017-09-27:
@calder Thanks, what usb vid:pid you got?

calder, 2017-09-27:
Is that device.serial? If so, BY1708A34902072.

terrycain, 2017-09-27:
@calder Nah the line in lsusb that corrosponds to your device

calder, 2017-09-27:
Ah, this?
```
Bus 001 Device 004: ID 1532:0224 Razer USA, Ltd
```

terrycain, 2017-09-27:
@calder Yeah, will fix on weekend

calder, 2017-09-27:
Thanks @terrycain!

z3ntu, 2017-10-24:
@calder Can you please test https://github.com/openrazer/openrazer/tree/fix_bladematrix ?

commit b50f13e4 (2017-11-19): 0x0224 has 16 columns, not 15

Fixes #398
closed 2017-11-19 by z3ntu