openrazer/openrazer #399: EL7 packages

moozhub, 2017-09-27:
FYI.. I built a EL7 repo with your packages in it, you can take a look [here](https://github.com/moozhub/yum-repo-mooz)

z3ntu, 2017-09-28:
I just tried to add support to the existing spec file for building on OBS (openSUSE Build Service) but Python3 is only support in EPEL apparently which isn't available on OBS without manually copying packages around.

moozhub, 2017-09-28:
I'm not too familiar but I did have to modify the spec file I used slightly and build a few other python packages that weren't included in stock el7 or epel repos, those are also in the repo I built

z3ntu, 2017-09-29:
I just meant that EPEL isn't available where we build the packages (=OBS). If there are more people showing themselves that want/need RHEL support, we'll take a look at it again.

closed 2017-09-29 by z3ntu

z3ntu, 2018-05-07:
Hey @moozhub . As we get quite a few issues about RHEL or other derived distros (centos and scientific linux), we will link your repo on the website with a disclaimer that people shouldn't open issues here about broken packages or whatever :wink: 
For that it would be great if you could update the packages in your repo (we're at v2.3.0 currently, newest spec is at https://github.com/openrazer/OBS-packaging/blob/master/openrazer.spec). Thanks!