openrazer/openrazer #400: Unable to adjust brightness or stop the "breathing" effect on my Blackwidow Ultimate 2012

ChristopherBare, 2017-09-30:
Hello, 
After an update or something going on in my system, I am now unable to  turn off the "breathing" effect or adjust my brightness. I can, however,  activate gaming mode from the polychromatic interface. Does anyone have some ideas of how to fix this issue?

When I ran `dmesg | grep razer` it returned:
```
[   11.016166] razerkbd: loading out-of-tree module taints kernel.
[   11.016199] razerkbd: module verification failed: signature and/or required key missing - tainting kernel
[   11.872302] razerkbd 0003:1532:010D.0001: input,hidraw0: USB HID v1.11 Keyboard [Razer Razer BlackWidow Ultimate] on usb-0000:00:14.0-9.3/input0
[   11.932273] razerkbd 0003:1532:010D.0002: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer BlackWidow Ultimate] on usb-0000:00:14.0-9.3/input1
[   11.932324] razerkbd 0003:1532:010D.0003: input,hidraw2: USB HID v1.11 Mouse [Razer Razer BlackWidow Ultimate] on usb-0000:00:14.0-9.3/input2
[   36.072810] Modules linked in: pci_stub vboxpci(OE) vboxnetadp(OE) vboxnetflt(OE) iptable_filter ip_tables x_tables vboxdrv(OE) joydev binfmt_misc input_leds nls_iso8859_1 nvidia_uvm(POE) mxm_wmi intel_rapl x86_pkg_temp_thermal intel_powerclamp arc4 coretemp kvm_intel snd_hda_codec_hdmi kvm nvidia_drm(POE) uvcvideo nvidia_modeset(POE) irqbypass videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core nvidia(POE) videodev crct10dif_pclmul razerkbd(OE) crc32_pclmul snd_usb_audio media snd_usbmidi_lib ghash_clmulni_intel pcbc snd_hda_codec_realtek snd_hda_codec_generic ath9k snd_hda_intel aesni_intel snd_hda_codec ath9k_common aes_x86_64 ath9k_hw crypto_simd snd_hda_core glue_helper snd_hwdep cryptd snd_pcm ath intel_cstate mac80211 intel_rapl_perf snd_seq_midi snd_seq_midi_event snd_rawmidi 
```
But past that, I'm not too sure what's going on. The log file wan't much help. Just a lot of DEBUG and INFO lines.

z3ntu, 2017-10-01:
Please post the output of `ls -al /sys/bus/hid/drivers/razerkbd/0*/`

ChristopherBare, 2017-10-01:
Here's the output of `ls -al /sys/bus/hid/drivers/razerkbd/0*/`
``` /sys/bus/hid/drivers/razerkbd/0003:1532:010D.0001/:
total 0
drwxr-xr-x 5 root plugdev    0 Sep 30 18:28 .
drwxr-xr-x 5 root root       0 Sep 30 18:28 ..
-r--r--r-- 1 root plugdev 4096 Sep 30 18:28 country
lrwxrwxrwx 1 root plugdev    0 Sep 30 18:28 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 Sep 30 18:28 hidraw
drwxr-xr-x 3 root plugdev    0 Sep 30 18:28 input
-rw-rw---- 1 root plugdev 4096 Sep 30 18:28 key_alt_f4
-rw-rw---- 1 root plugdev 4096 Sep 30 18:28 key_alt_tab
-rw-rw---- 1 root plugdev 4096 Sep 30 18:28 key_super
-r--r--r-- 1 root plugdev 4096 Sep 30 18:28 modalias
drwxr-xr-x 2 root plugdev    0 Sep 30 18:28 power
-r--r--r-- 1 root plugdev 4096 Sep 30 18:28 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Sep 30 18:28 subsystem -> ../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Sep 30 18:28 uevent

/sys/bus/hid/drivers/razerkbd/0003:1532:010D.0002/:
total 0
drwxr-xr-x 5 root plugdev    0 Sep 30 18:28 .
drwxr-xr-x 5 root root       0 Sep 30 18:28 ..
-r--r--r-- 1 root plugdev 4096 Sep 30 18:28 country
lrwxrwxrwx 1 root plugdev    0 Sep 30 18:28 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 Sep 30 18:28 hidraw
drwxr-xr-x 3 root plugdev    0 Sep 30 18:28 input
-rw-rw---- 1 root plugdev 4096 Sep 30 18:28 key_alt_f4
-rw-rw---- 1 root plugdev 4096 Sep 30 18:28 key_alt_tab
-rw-rw---- 1 root plugdev 4096 Sep 30 18:28 key_super
-r--r--r-- 1 root plugdev 4096 Sep 30 18:28 modalias
drwxr-xr-x 2 root plugdev    0 Sep 30 18:28 power
-r--r--r-- 1 root plugdev 4096 Sep 30 18:28 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Sep 30 18:28 subsystem -> ../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Sep 30 18:28 uevent

/sys/bus/hid/drivers/razerkbd/0003:1532:010D.0003/:
total 0
drwxr-xr-x 5 root plugdev    0 Sep 30 18:28 .
drwxr-xr-x 5 root root       0 Sep 30 18:28 ..
-r--r--r-- 1 root plugdev 4096 Sep 30 18:28 country
-rw-rw---- 1 root plugdev 4096 Sep 30 18:31 device_mode
-r--r----- 1 root plugdev 4096 Sep 30 18:28 device_serial
-r--r----- 1 root plugdev 4096 Sep 30 18:28 device_type
lrwxrwxrwx 1 root plugdev    0 Sep 30 18:28 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
-r--r----- 1 root plugdev 4096 Sep 30 18:28 firmware_version
-rw-rw---- 1 root plugdev 4096 Sep 30 18:28 game_led_state
drwxr-xr-x 3 root plugdev    0 Sep 30 18:28 hidraw
drwxr-xr-x 3 root plugdev    0 Sep 30 18:28 input
-rw-rw---- 1 root plugdev 4096 Sep 30 18:28 macro_led_effect
-rw-rw---- 1 root plugdev 4096 Sep 30 18:28 macro_led_state
-rw-rw---- 1 root plugdev 4096 Sep 30 18:28 matrix_brightness
-rw-rw---- 1 root plugdev 4096 Sep 30 18:28 matrix_effect_pulsate
--w--w---- 1 root plugdev 4096 Sep 30 18:28 matrix_effect_static
-r--r--r-- 1 root plugdev 4096 Sep 30 18:28 modalias
drwxr-xr-x 2 root plugdev    0 Sep 30 18:28 power
-r--r--r-- 1 root plugdev 4096 Sep 30 18:28 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Sep 30 18:28 subsystem -> ../../../../../../../../bus/hid
-rw-rw---- 1 root plugdev 4096 Sep 30 18:28 test
-rw-r--r-- 1 root plugdev 4096 Sep 30 18:28 uevent
-r--r----- 1 root plugdev 4096 Sep 30 18:28 version
 ```

z3ntu, 2017-10-20:
What distro are you using?

n4rq5, 2017-10-23:
I had the same or very similar issue today on Ubuntu 16.04 LTS.
It was new keyboard, plugged for the first time and openrazer was installed clean as well.
Had to remove it completely since it prevented Google Chrome from starting at all for example.

z3ntu, 2017-10-23:
@n4rq5 That's issue #358 and it's solved on master / the daily PPA

ChristopherBare, 2017-10-23:
I switched to the daily build to see if that would correct my issue as well. It did not. I am on Ubuntu 16.04 LTS if that is still relevant.

closed 2018-03-22 by stale[bot]