openrazer/openrazer #403: Razer Wolverine Ultimate.

Zerophase, 2017-10-04:
Should we just support the lights for this? Seems like other functionality would be stepping a bit on xpad functionality.  I'm willing to add support once this comes back into the Razer store.

terrycain, 2017-10-04:
If your lucky, the pc will recognise it as a gamepad/joystick. Might not be a bad shout for the frontend developers to add some config stuff if it works out of the box though..

Zerophase, 2017-10-04:
I think it will work mostly out of the box with just some fine tuning for lighting control. I'm just thinking it most likely has a macro feature.

closed 2018-03-03 by stale[bot]

relma2, 2022-05-13:
Hi,
I'd like to reopen this issue because I have a Wolverine Ultimate Controller and would like the RGB lights on it to sync with my mouse

relma2, 2022-05-13:
@z3ntu 
```
Bus 001 Device 011: ID 1532:007e Razer USA, Ltd RC30-030502 Mouse Dock
Bus 001 Device 007: ID 1532:0088 Razer USA, Ltd Gaming Mouse [Basilisk Ultimate, Wireless, Receiver]
Bus 001 Device 013: ID 1532:0a14 Razer USA, Ltd Razer Wolverine Ultimate
```

z3ntu, 2022-05-13:
Ok. Please open a new issue for this device with the "New Device Support" template.