openrazer/openrazer #405: No module named 'openrazer daemon'

snake404, 2017-10-06:
Hi, I have this issue when I try to open the razerGenie, I already installed the dependencies and I followed the instructions of the openrazer repository and I already installed the daemonize library with pip and pip3 but the problem persist.

This is the output of the program:

```
openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since vie 2017-10-06 15:22:36 CDT; 25s ago
     Docs: man:openrazer-daemon(8)
  Process: 4926 ExecStart=/usr/bin/openrazer-daemon -Fv (code=exited, status=1/FAILURE)
 Main PID: 4926 (code=exited, status=1/FAILURE)

oct 06 15:22:36 snake-lenovo systemd[2741]: Starting Daemon to manage razer devices in userspace...
oct 06 15:22:36 snake-lenovo openrazer-daemon[4926]: Traceback (most recent call last):
oct 06 15:22:36 snake-lenovo openrazer-daemon[4926]:   File "/usr/bin/openrazer-daemon", line 9, in <module>
oct 06 15:22:36 snake-lenovo openrazer-daemon[4926]:     from openrazer_daemon.daemon import daemonize
oct 06 15:22:36 snake-lenovo openrazer-daemon[4926]: ModuleNotFoundError: No module named 'openrazer_daemon'
oct 06 15:22:36 snake-lenovo systemd[2741]: openrazer-daemon.service: Main process exited, code=exited, status=1/FAILURE
oct 06 15:22:36 snake-lenovo systemd[2741]: Failed to start Daemon to manage razer devices in userspace.
oct 06 15:22:36 snake-lenovo systemd[2741]: openrazer-daemon.service: Unit entered failed state.
oct 06 15:22:36 snake-lenovo systemd[2741]: openrazer-daemon.service: Failed with result 'exit-code'.
```

z3ntu, 2017-10-06:
What distro do you use?

snake404, 2017-10-06:
Oh sorry, forgot that.

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.3 LTS
Release:	16.04
Codename:	xenial

z3ntu, 2017-10-06:
Did you add the OpenRazer PPA to your system? See [here](https://openrazer.github.io/#ubuntu).

snake404, 2017-10-06:
Yes I did, I tried with other PPA but I added the PPA of your repository again but I got the same error, this is the trace of the installation:

```
sudo add-apt-repository ppa:openrazer/stable
 For users who would like a well-tested and confirmed working driver and daemon for supported devices.
 More info: https://launchpad.net/~openrazer/+archive/ubuntu/stable
Press [ENTER] to continue or ctrl-c to cancel adding it

gpg: keyring `/tmp/tmph9j29469/secring.gpg' created
gpg: keyring `/tmp/tmph9j29469/pubring.gpg' created
gpg: requesting key 7B2AEE37 from hkp server keyserver.ubuntu.com
gpg: /tmp/tmph9j29469/trustdb.gpg: trustdb created
gpg: key 7B2AEE37: public key "Launchpad PPA for OpenRazer" imported
gpg: Total number processed: 1
gpg:               imported: 1  (RSA: 1)
OK

sudo apt install openrazer-meta
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  openrazer-doc openrazer-kernel-modules-dkms
The following packages will be REMOVED:
  razer-doc razer-kernel-modules-dkms
The following NEW packages will be installed:
  openrazer-doc openrazer-kernel-modules-dkms openrazer-meta
0 upgraded, 3 newly installed, 2 to remove and 0 not upgraded.
Need to get 41.3 kB/54.0 kB of archives.
After this operation, 13.3 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://ppa.launchpad.net/openrazer/stable/ubuntu xenial/main amd64 openrazer-kernel-modules-dkms amd64 2.0.0-0ubuntu1~ubuntu16.04.1 [41.3 kB]
Fetched 41.3 kB in 0s (73.2 kB/s)                        
(Reading database ... 248596 files and directories currently installed.)
Removing razer-doc (1.1.16-0ubuntu1) ...
dpkg: razer-kernel-modules-dkms: dependency problems, but removing anyway as you requested:
 openrazer-daemon depends on razer-kernel-modules-dkms.

Removing razer-kernel-modules-dkms (1.1.16-0ubuntu1) ...

-------- Uninstall Beginning --------
Module:  razer_chroma_driver
Version: 1.0.0
Kernel:  4.4.0-96-generic (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

razerkbd.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.4.0-96-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

razermouse.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.4.0-96-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

razerfirefly.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.4.0-96-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

razerkraken.ko:
[... 73 lines cut ...]

razermug.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.4.0-96-generic/updates/dkms/

razercore.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.4.0-96-generic/updates/dkms/

depmod....

DKMS: install completed.
-e 
ADD YOUR USER TO THE PLUGDEV GROUP 'sudo gpasswd -a <USERNAME> plugdev'
Setting up openrazer-doc (2.0.0-0ubuntu1~ubuntu16.04.1) ...
Setting up openrazer-meta (2.0.0-0ubuntu1~ubuntu16.04.1) ...

```
I already added my user to the plugdev group

z3ntu, 2017-10-07:
Please follow the instructions at https://github.com/openrazer/openrazer/releases/tag/v2.0.0 . It seems that you still have our old PPA installed.

snake404, 2017-10-09:
Hi again, sorry for the delay.
I followed your instructions but I got the same error:

```
Traceback (most recent call last):
  File "/usr/bin/openrazer-daemon", line 9, in <module>
    from openrazer_daemon.daemon import daemonize
ModuleNotFoundError: No module named 'openrazer_daemon'

```

Maybe is a specific problem with my linux distribution or with the python version.

z3ntu, 2017-10-10:
Please give me the output of `ls -al /usr/lib/python3*/site-packages/openrazer_daemon*`

snake404, 2017-10-10:
There is no output in that path, but this is the content of the path /usr/lib/python3/dist-packages/openrazer_daemon

```
total 72
drwxr-xr-x  5 root root  4096 oct 10 18:07 .
drwxr-xr-x 14 root root  4096 oct 10 18:07 ..
-rw-r--r--  1 root root 21571 sep 19 12:51 daemon.py
drwxr-xr-x  3 root root  4096 oct 10 18:07 dbus_services
-rw-r--r--  1 root root  5771 sep 19 12:51 device.py
drwxr-xr-x  2 root root  4096 oct 10 18:07 hardware
-rw-r--r--  1 root root    28 sep 19 12:51 __init__.py
-rw-r--r--  1 root root 16239 sep 19 12:51 keyboard.py
drwxr-xr-x  2 root root  4096 oct 10 18:07 misc

```

z3ntu, 2017-10-10:
`python3 -c 'import sys; print(sys.path)'` please?

snake404, 2017-10-10:
Sure
```
python3 -c 'import sys; print(sys.path)'
['', '/usr/local/lib/python36.zip', '/usr/local/lib/python3.6', '/usr/local/lib/python3.6/lib-dynload', '/home/snake/.local/lib/python3.6/site-packages', '/usr/local/lib/python3.6/site-packages']
```

z3ntu, 2017-10-15:
Did you do anything special with python? Because the import paths shouldn't just be in /usr/local/* but also in /usr/lib/. Try reinstall python3 and run the previous command again.

snake404, 2017-10-20:
Sorry for the delay. I already reinstalled python3 but I got the same error and I lost my UI when I uninstalled python3, I had to reinstall my graphical environment. Maybe this issue is an specific error of my pc, I don't know. I tried a lot of things but I got the same error.

Evanjs, 2017-10-23:
This is happening on Gentoo as well.  Happy to provide any information about my install/system that might help.

z3ntu, 2017-10-23:
I am in the process of installing Gentoo in a VM (tbh the handbook is partly pretty bad in terms of readability) so I can try and help then 😉

z3ntu, 2017-10-23:
@Evanjs I didn't manage to install layman but maybe you can help me
```
gentoo-vm ~ # emerge --ask layman

 * IMPORTANT: 11 news items need reading for repository 'gentoo'.
 * Use eselect news read to view new items.

These are the packages that would be merged, in order:

Calculating dependencies... done!
[ebuild  N     ] dev-qt/qtchooser-0_p20151008  USE="{-test}" 
[ebuild  N     ] app-text/poppler-data-0.4.7 
[ebuild  N     ] dev-libs/lzo-2.09  USE="-examples -static-libs" ABI_X86="(64) -32 (-x32)" 
[ebuild  N     ] dev-util/re2c-0.16 
[ebuild  N     ] dev-libs/libatomic_ops-7.6.0  ABI_X86="(64) -32 (-x32)" 
[ebuild  N     ] dev-libs/gobject-introspection-common-1.50.0 
[ebuild  N     ] dev-libs/vala-common-0.34.8 
[ebuild  N     ] app-arch/rpm2targz-9.0.0.5g 
[ebuild  N     ] dev-libs/libunistring-0.9.7  USE="-doc -static-libs" ABI_X86="(64) -32 (-x32)" 
[ebuild  N     ] dev-libs/npth-1.3  USE="-static-libs" 
[ebuild  N     ] media-libs/jbig2dec-0.13-r4  USE="png -static-libs {-test}" 
[ebuild  N     ] sys-libs/mtdev-1.1.5  USE="-static-libs" 
[ebuild  N     ] x11-themes/hicolor-icon-theme-0.15 
[ebuild  N     ] app-eselect/eselect-opengl-1.3.1-r4 
[ebuild  N     ] app-eselect/eselect-qtgraphicssystem-1.1.1 
[ebuild  N     ] app-eselect/eselect-mesa-0.0.10-r1 
[ebuild  N     ] app-eselect/eselect-lib-bin-symlink-0.1.1 
[ebuild  N     ] app-eselect/eselect-pinentry-0.7 
[ebuild  N     ] net-dns/libidn-1.33  USE="nls -doc -emacs -java -mono -static-libs" ABI_X86="(64) -32 (-x32)" 
[ebuild  N     ] dev-libs/libassuan-2.4.3-r1  USE="-static-libs" 
[ebuild  N     ] dev-libs/libksba-1.3.5-r1  USE="-static-libs" 
[ebuild  N     ] dev-libs/elfutils-0.169-r1  USE="bzip2 nls utils -lzma -static-libs {-test} (-threads)" ABI_X86="(64) -32 (-x32)" 
[ebuild  N     ] virtual/libelf-1  ABI_X86="(64) -32 (-x32)" 
[ebuild  N     ] dev-util/scons-2.4.1  USE="-doc" PYTHON_TARGETS="python2_7 (-pypy)" 
[ebuild  N     ] dev-libs/double-conversion-2.0.1  USE="-static-libs" 
[ebuild  N     ] dev-util/ninja-1.7.2  USE="-doc -emacs {-test} -vim-syntax -zsh-completion" 
[ebuild  N     ] dev-python/idna-2.5  PYTHON_TARGETS="python2_7 python3_4 (-pypy) (-pypy3) (-python3_5) (-python3_6)" 
[ebuild  N     ] dev-python/pyasn1-0.2.3  USE="-doc" PYTHON_TARGETS="python2_7 python3_4 (-pypy) (-pypy3) (-python3_5) (-python3_6)" 
[ebuild  N     ] dev-python/asn1crypto-0.22.0  USE="{-test}" PYTHON_TARGETS="python2_7 python3_4 (-pypy) (-pypy3) (-python3_5) (-python3_6)" 
[ebuild  N     ] dev-python/enum34-1.1.6  USE="-doc" PYTHON_TARGETS="python2_7 (-pypy) (-pypy3)" 
[ebuild  N     ] dev-python/ply-3.9  USE="-examples" PYTHON_TARGETS="python2_7 python3_4 (-pypy) (-pypy3) (-python3_5) (-python3_6)"                                                                                                                                                                                      
[ebuild  N     ] dev-python/chardet-3.0.4  USE="{-test}" PYTHON_TARGETS="python2_7 python3_4 (-pypy) (-pypy3) (-python3_5) (-python3_6)"                                                                                                                                                                                  
[ebuild  N     ] dev-python/PySocks-1.6.7  PYTHON_TARGETS="python2_7 python3_4 (-pypy) (-pypy3) (-python3_5) (-python3_6)" 
[ebuild  N     ] dev-python/ipaddress-1.0.18  PYTHON_TARGETS="python2_7 (-pypy)" 
[ebuild  N     ] virtual/python-ipaddress-1.0-r1  PYTHON_TARGETS="python2_7 python3_4 (-pypy) (-pypy3) (-python3_5) (-python3_6)"                                                                                                                                                                                         
[ebuild  N     ] dev-python/pycparser-2.14  USE="{-test}" PYTHON_TARGETS="python2_7 python3_4 (-pypy) (-pypy3) (-python3_5) (-python3_6)"                                                                                                                                                                                 
[ebuild  NS    ] app-text/docbook-xml-dtd-4.4-r2 [4.1.2-r6]
[ebuild  NS    ] app-text/docbook-xml-dtd-4.2-r2 [4.1.2-r6]
[ebuild  NS    ] app-text/docbook-xml-dtd-4.3-r1 [4.1.2-r6]
[ebuild  N     ] virtual/perl-Scalar-List-Utils-1.420.200_rc-r1 
[ebuild     U #] dev-libs/openssl-1.1.0f-r1 [1.0.2l] USE="-bindist* -tls-heartbeat*" 
[ebuild  N     ] virtual/perl-Digest-SHA-5.950.100_rc-r4 
[ebuild  N     ] virtual/perl-Digest-MD5-2.540.0-r3 
[ebuild  N     ] dev-lang/nasm-2.12.01  USE="-doc" 
[ebuild  N     ] virtual/perl-MIME-Base64-3.150.0-r3 
[ebuild  N     ] virtual/perl-libnet-3.80.100_rc-r2 
[ebuild  N     ] dev-perl/Net-SSLeay-1.810.0  USE="-examples (-libressl) -minimal {-test}" 
[ebuild  N     ] dev-perl/Digest-HMAC-1.30.0-r1 
[ebuild  N     ] media-libs/libjpeg-turbo-1.5.1  USE="-java -static-libs" ABI_X86="(64) -32 (-x32)" 
[ebuild  N     ] virtual/jpeg-0-r2  USE="-static-libs" ABI_X86="(64) -32 (-x32)" 
[ebuild  N     ] dev-perl/Authen-SASL-2.160.0-r1  USE="-kerberos" 
[ebuild  N     ] dev-perl/IO-Socket-SSL-2.24.0  USE="-idn" 
[ebuild  N     ] dev-libs/libtasn1-4.12-r1  USE="-doc -static-libs -valgrind" ABI_X86="(64) -32 (-x32)" 
[ebuild  N     ] net-dns/libidn2-2.0.4  USE="-static-libs" ABI_X86="(64) -32 (-x32)" 
[ebuild  N     ] dev-perl/File-BaseDir-0.30.0-r1  USE="{-test}" 
[ebuild  N     ] dev-perl/Error-0.170.240  USE="{-test}" 
[ebuild  N     ] dev-perl/Net-SMTP-SSL-1.30.0  USE="{-test}" 
[... cut ...]
impossible to satisfy simultaneously.  If such a conflict exists in
the dependencies of two different packages, then those packages can
not be installed simultaneously. You may want to try a larger value of
the --backtrack option, such as --backtrack=30, in order to see if
that will solve this conflict automatically.

For more information, see MASKED PACKAGES section in the emerge man
page or refer to the Gentoo Handbook.
```
The first time I ran this command I chose yes and then ran dispatch-conf and pressed U and now the above output is what I get...

Evanjs, 2017-10-23:
Yeah openssl can be a pain on Gentoo. 
[This](https://forums.gentoo.org/viewtopic-t-1061918-start-0.html#8053862) *should* be all you need to do to fix those conflicts.

StevensCol, 2017-11-17:
Hi everyone. I got the same error on running the RazerGenie. In my case i've installed in my pc Fedora 26 1 day ago and the first task I've done has been to install the open razer drivers, following the steps for Fedora 26 mentioned in:

[https://openrazer.github.io/](url)

Next, i've installed the RazerGenie Application following the steps mentioned in:

[https://software.opensuse.org//download.html?project=hardware%3Arazer&package=razergenie](url)

After it's done, i've restarted the pc and then run the RazerGenie App but i got the next error:

`● openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2017-11-17 13:00:17 -05; 1min 59s ago
     Docs: man:openrazer-daemon(8)
  Process: 6381 ExecStart=/usr/bin/openrazer-daemon -Fv (code=exited, status=1/FAILURE)
 Main PID: 6381 (code=exited, status=1/FAILURE)

nov 17 13:00:16 Stevens systemd[1855]: Starting Daemon to manage razer devices in userspace...
nov 17 13:00:17 Stevens python3 /usr/bin/openrazer-daemon -Fv[6381]: detected unhandled Python exception in '/usr/bin/openrazer-daemon'
nov 17 13:00:17 Stevens openrazer-daemon[6381]: Traceback (most recent call last):
nov 17 13:00:17 Stevens openrazer-daemon[6381]:   File "/usr/bin/openrazer-daemon", line 15, in <module>
nov 17 13:00:17 Stevens openrazer-daemon[6381]:     from daemonize import Daemonize
nov 17 13:00:17 Stevens openrazer-daemon[6381]: ModuleNotFoundError: No module named 'daemonize'
nov 17 13:00:17 Stevens systemd[1855]: openrazer-daemon.service: Main process exited, code=exited, status=1/FAILURE
nov 17 13:00:17 Stevens systemd[1855]: Failed to start Daemon to manage razer devices in userspace.
nov 17 13:00:17 Stevens systemd[1855]: openrazer-daemon.service: Unit entered failed state.
nov 17 13:00:17 Stevens systemd[1855]: openrazer-daemon.service: Failed with result 'exit-code'.

`

@z3ntu ca you help me pls?

**Note:** I have a Razer Naga Chroma mouse device.

z3ntu, 2017-11-17:
@StevensCol Sorry, my bad. I missed adding the dependency to the spec file for the latest update.

z3ntu, 2017-11-17:
So running `dnf upgrade` should pull in the latest version with the updated dependencies.

z3ntu, 2017-11-18:
@snake404 @Evanjs Does the issue still happen on the newest version?

StevensCol, 2017-11-20:
Hi, bro @z3ntu , sorry for the delay because i had not been able to work on the pc. The updates works fine and the RazerGenie runs ok without errors, but i do not know how to set the keys bindings for my chroma mouse :(, these apps in the official site only set the light effects.

Thanks so much bro for help me.

pd: sorry fir my bad english.

z3ntu, 2017-11-20:
Keybindings are not yet supported. There should be a bunch of other issues open about that :)

(also still to @snake404 & @Evanjs: Is this issue still valid?)

Evanjs, 2017-11-20:
According to https://github.com/vifino/vifino-overlay/pull/8, as @vifino updated the Gentoo ebuild to include daemonize as a dependency, I think this has been solved for Gentoo users.
That is, as long as they use vifino-overlay, I can't speak for pg_overlay.

At any rate, I think this kind of issue is something for the overlay maintainers (and not the upstream devs) if the problem is simply a missing dependency, at least with Python projects.

TL;DR - No longer valid; Problem has been solved on vifino-overlay.

z3ntu, 2017-11-20:
I'll close this then. Thanks.

closed 2017-11-20 by z3ntu

StevensCol, 2017-11-20:
The news is sad for me, I have several low vision problems and this use to configure keyboard shortcuts and certain accessibility tools. I have to touch a window :( thanks for your collaboration @z3ntu

vifino, 2017-11-20:
Yeah, daemonize is now in my overlay and the python deps of openrazer are handled better thanks to a PR from a user. Gentoo users using my overlay should be fine daemon-wise.

StevensCol, 2017-11-20:
Is there any way to configure the razer chroma mouse keys?

z3ntu, 2017-11-20:
@StevensCol On my DeathAdder Chroma the two side buttons are mouse buttons 8 & 9. If an applications supports mapping mouse buttons to certain actions, it should be possible.

StevensCol, 2017-11-24:
@z3ntu thanks for the recomendation and sorry for the delay, i've still working in this issue, and i've found the xbindkeys program for binding the side keys from my Razer chroma, but, it is a bit akward because the side buttons send the same key events as the key numbers of my laptop. The good news is that my laptop have a numeric keyboard apart from the conventional ones and these send another event type. which allows me to bind the side key buttons of my Razer Chroma. The issues are my Chroma have 12 side buttons and i can only bind the first 10 because the 11 and 12 buttons fires the same event that " ' " and " ¿ " keys, then i can't  to use these because I would not have these two keys available on my keyboard to write. The other little issue is that the number keys on top of my laptop doesn't write the numbers chars because these launch my commands i've assigned.

But this works form me while the open razer driver fixes this issues with the side buttons bindings and can be support this feautre. Thanks a lot bro.

leokassio, 2018-05-14:
Any news on this issue? I am having the same problem on Ubuntu 18.10. 
``from openrazer_daemon.daemon import RazerDaemon, __version__
ModuleNotFoundError: No module named 'openrazer_daemon'``

leokassio, 2018-05-15:
/home/username/Applications/anaconda3/bin/python3

z3ntu, 2018-05-15:
You can replace the shebang (first line) in `/usr/bin/openrazer-daemon` with `#!/usr/bin/python3` for now.

leokassio, 2018-05-15:
It worked. The daemon is on now (just to let me to discover that my DeathAdder 3500 DPI has no support). Anyway, thank you!

Expugnam, 2018-05-26:
hello i am trying for two days now to install openrazer in Fedora 28. Polychromatic says that the daemon took too long to initialize or it has crashed. I have been searching the web for more than 12 hours today and this is my first post on github. (sigh)

commit 8a91796f (2018-07-11): Change shebang to python3 in /usr/bin

Fixes #405 (and a bunch of other issues), where people have a custom python3 in their $PATH which gets preference over /usr/bin/python3 but the site-packages folder where the daemon is installed is only for /usr/bin/python3.