openrazer/openrazer #406: openrazer doesnt install properly

whiteknights2007, 2017-10-08:
after installing openrazer  , I try to probe razerkbd  , but It always gives an error message. Please Help!

I used those commands to install openrazer : 
                                 sudo add-apt-repository ppa:openrazer/stable
                                 sudo apt update
                                 sudo apt install openrazer-meta

This is what I see when I try to probe razerkbd : 

-Blade-Stealth:~$ modprobe razerkbd
modprobe: FATAL: Module razerkbd not found in directory /lib/modules/4.10.0-35-generic

this is what I see in the terminal after installing openrazer : 

-Blade-Stealth:~$ sudo add-apt-repository ppa:openrazer/stabl
[sudo] password for hassan: 
Cannot add PPA: 'ppa:~openrazer/ubuntu/stabl'.
The team named '~openrazer' has no PPA named 'ubuntu/stabl'
Please choose from the following available PPAs:
 * 'daily':  Daily Builds
 * 'stable':  Stable Branch
-Blade-Stealth:~$ sudo apt update
Hit:1 http://ca.archive.ubuntu.com/ubuntu xenial InRelease
Hit:2 http://security.ubuntu.com/ubuntu xenial-security InRelease              
Hit:3 http://ppa.launchpad.net/cyanogen/uchroma/ubuntu xenial InRelease        
Hit:4 http://ca.archive.ubuntu.com/ubuntu xenial-updates InRelease             
Ign:5 http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_17.04  InRelease
Hit:6 https://dl.winehq.org/wine-builds/ubuntu xenial InRelease                
Hit:7 http://ca.archive.ubuntu.com/ubuntu xenial-backports InRelease           
Hit:8 http://ppa.launchpad.net/lah7/polychromatic/ubuntu xenial InRelease      
Hit:9 http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_17.04  Release
Get:10 http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_17.04  Release.gpg [481 B]
Ign:10 http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_17.04  Release.gpg
Hit:11 http://ppa.launchpad.net/openrazer/daily/ubuntu xenial InRelease
Hit:12 http://ppa.launchpad.net/openrazer/stable/ubuntu xenial InRelease       
Ign:13 http://ppa.launchpad.net/realender/winconn/ubuntu xenial InRelease      
Hit:14 http://ppa.launchpad.net/terrz/razerutils/ubuntu xenial InRelease
Hit:15 http://ppa.launchpad.net/ubuntu-desktop/ubuntu-make/ubuntu xenial InRelease
Err:16 http://ppa.launchpad.net/realender/winconn/ubuntu xenial Release        
  404  Not Found
Reading package lists... Done                      
W: GPG error: http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_17.04  Release: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 062F9FD4D6D11CE4
E: The repository 'http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_17.04  Release' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'http://ppa.launchpad.net/realender/winconn/ubuntu xenial Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
Blade-Stealth:~$ sudo apt install openrazer-meta
Reading package lists... Done
Building dependency tree       
Reading state information... Done
openrazer-meta is already the newest version (2.0.0-1~263~ubuntu16.04.1).
The following packages were automatically installed and are no longer required:
  fonts-horai-umefont fonts-play fonts-unfonts-core gnome-exe-thumbnailer
  icoutils libgif7:i386 libgsf-1-114 libgsf-1-common libice6:i386 libmsi0
  libp11-kit-gnome-keyring:i386 libsm6:i386 libwxbase3.0-0v5 libwxgtk3.0-0v5
  libxt6:i386 msitools ocl-icd-libopencl1:i386 odbcinst odbcinst1debian2
  p11-kit-modules:i386 p7zip python-wxgtk3.0 python-wxversion ttf-wqy-microhei
  unixodbc wine-gecko2.21 wine-gecko2.21:i386 wine-mono0.0.8 winetricks
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 51 not upgraded.

z3ntu, 2017-10-08:
I see that you have also tried to install RazerGenie but missed one step which is :
```
wget -nv https://download.opensuse.org/repositories/hardware:razer/xUbuntu_17.04/Release.key -O Release.key
sudo apt-key add - < Release.key
sudo apt-get update
```
Also you still have the old repository in your package lists, so please run:
```
sudo rm /etc/apt/sources.list.d/terrz-ubuntu-razerutils-*
sudo rm /etc/apt/sources.list.d/openrazer-*
sudo add-apt-repository ppa:openrazer/stable
sudo apt update
sudo apt remove openrazer-daemon openrazer-doc openrazer-kernel-modules-dkms openrazer-meta python3-openrazer
sudo apt install openrazer-meta
```

whiteknights2007, 2017-10-08:
yes I tried installing razerGenie , razrCommander , polychromatic . 

so I did what told me but still doesnt work . this is what I see now

-Blade-Stealth:~$ sudo rm /etc/apt/sources.list.d/terrz-ubuntu-razerutils-*
[sudo] password for : 
hassan@hassan-Blade-Stealth:~$ sudo rm /etc/apt/sources.list.d/terrz-ubuntu-razerutils-*
rm: cannot remove '/etc/apt/sources.list.d/terrz-ubuntu-razerutils-*': No such file or directory
-Blade-Stealth:~$ sudo rm /etc/apt/sources.list.d/openrazer-*
-Blade-Stealth:~$ sudo add-apt-repository ppa:openrazer/stable
 For users who would like a well-tested and confirmed working driver and daemon for supported devices.
 More info: https://launchpad.net/~openrazer/+archive/ubuntu/stable
Press [ENTER] to continue or ctrl-c to cancel adding it

gpg: keyring `/tmp/tmpumezxuji/secring.gpg' created
gpg: keyring `/tmp/tmpumezxuji/pubring.gpg' created
gpg: requesting key 7B2AEE37 from hkp server keyserver.ubuntu.com
gpg: /tmp/tmpumezxuji/trustdb.gpg: trustdb created
gpg: key 7B2AEE37: public key "Launchpad PPA for OpenRazer" imported
gpg: Total number processed: 1
gpg:               imported: 1  (RSA: 1)
OK
-Blade-Stealth:~$ sudo apt update
Hit:1 http://ca.archive.ubuntu.com/ubuntu xenial InRelease
Hit:2 http://ppa.launchpad.net/cyanogen/uchroma/ubuntu xenial InRelease        
Get:3 http://security.ubuntu.com/ubuntu xenial-security InRelease [102 kB]     
Ign:4 http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_17.04  InRelease
Get:5 http://ca.archive.ubuntu.com/ubuntu xenial-updates InRelease [102 kB]    
Hit:6 http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_17.04  Release
Hit:7 https://dl.winehq.org/wine-builds/ubuntu xenial InRelease                
Hit:8 http://ppa.launchpad.net/lah7/polychromatic/ubuntu xenial InRelease      
Get:9 http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_17.04  Release.gpg [481 B]
Get:10 http://ca.archive.ubuntu.com/ubuntu xenial-backports InRelease [102 kB] 
Hit:11 http://ppa.launchpad.net/openrazer/stable/ubuntu xenial InRelease       
Ign:9 http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_17.04  Release.gpg
Ign:12 http://ppa.launchpad.net/realender/winconn/ubuntu xenial InRelease      
Hit:13 http://ppa.launchpad.net/ubuntu-desktop/ubuntu-make/ubuntu xenial InRelease
Err:14 http://ppa.launchpad.net/realender/winconn/ubuntu xenial Release
  404  Not Found
Reading package lists... Done                      
W: GPG error: http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_17.04  Release: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 062F9FD4D6D11CE4
E: The repository 'http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_17.04  Release' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'http://ppa.launchpad.net/realender/winconn/ubuntu xenial Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
-Blade-Stealth:~$ sudo apt remove openrazer-daemon openrazer-doc openrazer-kernel-modules-dkms openrazer-meta python3-openrazer
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  fonts-horai-umefont fonts-play fonts-unfonts-core gnome-exe-thumbnailer
  icoutils libblas-common libblas3 libgfortran3 libgif7:i386 libgsf-1-114
  libgsf-1-common libice6:i386 liblapack3 libmsi0
  libp11-kit-gnome-keyring:i386 libsm6:i386 libwxbase3.0-0v5 libwxgtk3.0-0v5
  libxdo3 libxt6:i386 msitools ocl-icd-libopencl1:i386 odbcinst
  odbcinst1debian2 p11-kit-modules:i386 p7zip python-wxgtk3.0 python-wxversion
  python3-evdev python3-numpy python3-pyudev python3-setproctitle
  ttf-wqy-microhei unixodbc wine-gecko2.21 wine-gecko2.21:i386 wine-mono0.0.8
  winetricks xautomation xdotool
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  openrazer-daemon openrazer-doc openrazer-kernel-modules-dkms openrazer-meta
  python3-openrazer
0 upgraded, 0 newly installed, 5 to remove and 51 not upgraded.
After this operation, 397 kB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 227198 files and directories currently installed.)
Removing openrazer-meta (2.0.0-1~263~ubuntu16.04.1) ...
[... 87 lines cut ...]
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.10.0-35-generic/updates/dkms/

razercore.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.10.0-35-generic/updates/dkms/

depmod....

DKMS: install completed.
-e 
ADD YOUR USER TO THE PLUGDEV GROUP 'sudo gpasswd -a <USERNAME> plugdev'
Setting up openrazer-daemon (2.0.0-0ubuntu1~ubuntu16.04.1) ...
Setting up openrazer-doc (2.0.0-0ubuntu1~ubuntu16.04.1) ...
Setting up python3-openrazer (2.0.0-0ubuntu1~ubuntu16.04.1) ...
Setting up openrazer-meta (2.0.0-0ubuntu1~ubuntu16.04.1) ...
Processing triggers for shim-signed (1.32~16.04.1+0.9+1474479173.6c180c6-1ubuntu1) ...
-Blade-Stealth:~$ modprobe razerkbd
modprobe: ERROR: could not insert 'razerkbd': Operation not permitted

z3ntu, 2017-10-08:
Please run
```
wget -nv https://download.opensuse.org/repositories/hardware:razer/xUbuntu_17.04/Release.key -O Release.key
sudo apt-key add - < Release.key
sudo apt-get update
```
to get rid of the GPG error at `apt update`.
Then run (and post the output of)
```
sudo dkms status
sudo modprobe razerkbd
```

whiteknights2007, 2017-10-08:
I get a different error message. here is what I see. 

-Blade-Stealth:~$ wget -nv https://download.opensuse.org/repositories/hardware:razer/xUbuntu_17.04/Release.key -O Release.key
2017-10-08 11:12:08 URL:http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_17.04/Release.key [1084/1084] -> "Release.key" [1]
-Blade-Stealth:~$ sudo apt-key add - < Release.key
[sudo] password for : 
WhitSorry, try again.
[sudo] password for : 
OK
-Blade-Stealth:~$ sudo apt-get update
Hit:1 http://ca.archive.ubuntu.com/ubuntu xenial InRelease
Get:2 http://security.ubuntu.com/ubuntu xenial-security InRelease [102 kB]     
Hit:3 http://ppa.launchpad.net/cyanogen/uchroma/ubuntu xenial InRelease        
Hit:4 https://dl.winehq.org/wine-builds/ubuntu xenial InRelease                
Get:5 http://ca.archive.ubuntu.com/ubuntu xenial-updates InRelease [102 kB]    
Ign:6 http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_17.04  InRelease
Hit:7 http://ppa.launchpad.net/lah7/polychromatic/ubuntu xenial InRelease      
Hit:8 http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_17.04  Release
Get:9 http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_17.04  Release.gpg [481 B]
Hit:10 http://ppa.launchpad.net/openrazer/stable/ubuntu xenial InRelease       
Get:11 http://ca.archive.ubuntu.com/ubuntu xenial-backports InRelease [102 kB] 
Ign:12 http://ppa.launchpad.net/realender/winconn/ubuntu xenial InRelease      
Hit:13 http://ppa.launchpad.net/ubuntu-desktop/ubuntu-make/ubuntu xenial InRelease
Ign:14 http://ppa.launchpad.net/realender/winconn/ubuntu xenial Release        
Ign:15 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main amd64 Packages
Ign:16 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main i386 Packages
Ign:17 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main all Packages
Ign:18 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main Translation-en_CA
Ign:19 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main Translation-en
Ign:20 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main amd64 DEP-11 Metadata
Ign:21 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main DEP-11 64x64 Icons
Ign:15 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main amd64 Packages
Ign:16 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main i386 Packages
Ign:17 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main all Packages
Ign:18 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main Translation-en_CA
Ign:19 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main Translation-en
Ign:20 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main amd64 DEP-11 Metadata
Ign:21 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main DEP-11 64x64 Icons
Ign:15 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main amd64 Packages
Ign:16 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main i386 Packages
Ign:17 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main all Packages
Ign:18 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main Translation-en_CA
Ign:19 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main Translation-en
Ign:20 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main amd64 DEP-11 Metadata
Ign:21 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main DEP-11 64x64 Icons
Ign:15 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main amd64 Packages
Ign:16 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main i386 Packages
Ign:17 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main all Packages
Ign:18 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main Translation-en_CA
Ign:19 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main Translation-en
Ign:20 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main amd64 DEP-11 Metadata
Ign:21 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main DEP-11 64x64 Icons
Ign:15 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main amd64 Packages
Ign:16 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main i386 Packages
Ign:17 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main all Packages
Ign:18 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main Translation-en_CA
Ign:19 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main Translation-en
Ign:20 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main amd64 DEP-11 Metadata
Ign:21 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main DEP-11 64x64 Icons
Err:15 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main amd64 Packages
  404  Not Found
Ign:16 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main i386 Packages
Ign:17 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main all Packages
Ign:18 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main Translation-en_CA
Ign:19 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main Translation-en
Ign:20 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main amd64 DEP-11 Metadata
Ign:21 http://ppa.launchpad.net/realender/winconn/ubuntu xenial/main DEP-11 64x64 Icons
Fetched 307 kB in 9s (33.9 kB/s)                                               
Reading package lists... Done
W: The repository 'http://ppa.launchpad.net/realender/winconn/ubuntu xenial Release' does not have a Release file.
[... cut ...]
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch http://ppa.launchpad.net/realender/winconn/ubuntu/dists/xenial/main/binary-amd64/Packages  404  Not Found
E: Some index files failed to download. They have been ignored, or old ones used instead.
-Blade-Stealth:~$ sudo dkms status
razer_chroma_driver, 1.0.0, 4.10.0-35-generic, x86_64: installed
-Blade-Stealth:~$ sudo modprobe razerkbd
modprobe: ERROR: could not insert 'razerkbd': Required key not available

z3ntu, 2017-10-08:
First of all, you seem to have added a PPA (`realender/winconn`) which isn't available for Ubuntu 16.04. You should be able to remove it by running `sudo add-apt-repository --remove ppa:realender/winconn`

The issue with `Required key not available` is solved by disabling Secure Boot in your UEFI.

z3ntu, 2017-10-08:
Also please format your terminal outputs with triple backticks (that character: `). I edited all your comments to be formatted like that.

whiteknights2007, 2017-10-08:
Sorry for the late reply . 

I disabled the secure boot . it worked !!

Thank you so much for your help

whiteknights2007, 2017-10-08:
Btw do I reEnable secure Boot ?

z3ntu, 2017-10-08:
No, leave secure boot off otherwise you'll hit the same issue again. Now have fun with OpenRazer :)

closed 2017-10-08 by z3ntu

whiteknights2007, 2017-10-08:
will do . Thank you for your help

whiteknights2007, 2017-10-18:
Hey there . So I experienced a problem and had to reinstall Ubuntu 16.04 .. Im trying to install polychromatic controller now but Im experiencing problems . The error message I get is "Daemon is not working" the modprobe is working though.

z3ntu, 2017-10-18:
A fix for some issues under Ubuntu 16.04 is currently under work/testing (see #358 ). Please try the daily branch on launchpad tomorrow and test if it works then.

whiteknights2007, 2017-10-18:
ok thank you