openrazer/openrazer #408: Red hat support

dventura11, 2017-10-10:
Hi I am a red hat 7 user and I want to use my razer keyboard, but there is not support for it.

Can you help me please?

z3ntu, 2017-10-10:
Is #399 good for you?

dventura11, 2017-10-13:
No, I have a dependency exception:
```
--> Processing Dependency: gobject-introspection(x86-64) >= 1.46.0 for package: python3-gobject-base-3.24.1-1.el7.x86_64
--> Finished Dependency Resolution
Error: Package: python3-gobject-base-3.24.1-1.el7.x86_64 (elmooz)
           Requires: gobject-introspection(x86-64) >= 1.46.0
           Installed: gobject-introspection-1.42.0-1.el7.x86_64 (@anaconda/7.3)
               gobject-introspection(x86-64) = 1.42.0-1.el7
           Available: gobject-introspection-1.36.0-4.el7.x86_64 (release-rhel-x86_64-workstation-7-r03)
               gobject-introspection(x86-64) = 1.36.0-4.el7
```

z3ntu, 2017-10-13:
I can just say that #399 was made with EPEL, but can't help you further than that.

dventura11, 2017-10-13:
I downloaded the missing dependencies from this repo: http://ftp.stu.edu.tw/Linux/CentOS/7/os/x86_64/Packages/

But now the keyboard does not work, I entered to /sys/bus/hid/devices/<device_id> and run `echo -n "1" > matrix_effect_wave` It works, but I can't write nothing.

moozhub, 2017-10-13:
No idea what you're doing, but gobject-introspection is available in the base centos7 repo...

Installed Packages
Name        : gobject-introspection
Arch        : x86_64
Version     : 1.50.0
Release     : 1.el7
Size        : 815 k
Repo        : installed
From repo   : base
Summary     : Introspection system for GObject-based libraries
URL         : https://wiki.gnome.org/Projects/GObjectIntrospection
License     : GPLv2+, LGPLv2+, MIT
Description : GObject Introspection can scan C header and source files in order to
            : generate introspection "typelib" files.  It also provides an API to examine
            : typelib files, useful for creating language bindings among other
            : things.

z3ntu, 2017-10-20:
I'm going to close this issue as we don't support CentOS/RedHat.

closed 2017-10-20 by z3ntu