openrazer/openrazer #411: Layout changing since install

Alkeryn, 2017-10-16:
Before install it was in azerty
but now it is always set in qwerty after i use setxkbamp in my autostart script
and i have to use setxkbap again
also sometime it fail shutdown and i have to force shutdown
and quiting X make it crash

Alkeryn, 2017-10-16:
Though sometime it boots in azerty and other times in qwerty so wtf

z3ntu, 2017-10-20:
I am pretty sure that our kernel driver doesn't interfere with keyboard layout... Do you have any logs you could show us?

Alkeryn, 2017-11-03:
Well i couldn't really fixed it so i wrote a small bash script to set to fr if it isn't at boot time
though it occur randomly so i don't really know why ^^

z3ntu, 2017-11-03:
🤷‍♂️ 

(if anybody other experiences the same issue, please write a comment here 🙂)

closed 2017-11-03 by z3ntu