openrazer/openrazer #413: Cannot control Logo LED on Razer Blade Pro 2017

KazWolfe, 2017-10-18:
The logo LED on the Blade Pro 2017 seems to act a bit oddly.

Writing `0` to `/sys/bus/hid/drivers/razerkbd/[...]/logo_led_state` will set the LED to on. 

Writing `1` will result the following log entry in dmesg, and will set the LED state to on similar to `0`:

    razerkbd: Command not supported. Start Marker: 05 id: ff Num Params: 03 Reserved: 03 Command: 02 Params: 01040100000000000000000000000000 .

Writing `2` will result in the LED pulsating similar to a sleep mode.

KazWolfe, 2017-10-18:
Looks like we can't turn off the LED right now for whatever reason. It also looks like the Blade Pro isn't on the inverted LED state.

Adding it there breaks the pulse though. Interesting...

KazWolfe, 2017-10-18:
Packet capture results from Synapse:

From Synapse

| Start | ID | Reserved | Num Params | Reserved | Command | Sub Cmd |  Params  | Note |
|---|---|---|---|---|---|---|---|---|
| `00` | `3f` | `00 00 00` | `03` | `03` | `00` | `01` | `04 00 00` | Logo Light Off (synapse) |
| `00` | `3f` | `00 00 00` | `03` | `03` |   `00` | `01` | `04 01 00` | Logo Light On (synapse) |

From razerkbd (with #414's patch)

| Start | ID | Reserved | Num Params | Reserved | Command | Sub Cmd |  Params  | Note |
|---|---|---|---|---|---|---|---|---|
| `00` | `ff` | `00 00 00` | `03` | `03` | `02` | `01` | `04 01 00` | Logo Light Off (razerkbd) |
| `00` | `ff` | `00 00 00` | `03` | `03` | `02` | `01` | `04 00 00` | Logo Light Off (razerkbd) |

Does this mean 414 is unnecessary/bad? It looks like different commands are being used now, even...

terrycain, 2017-10-18:
Hmm, Ok well I believe pulse is an artifact of copying and pasting firmware, if it works, I dont see a reason not to allow it ;-) as then we can claim #BetterThanSynapse.

Do you have the actual pcaps, for those 2 captures? I have a script that will grab all that with the updated headers

PaleoCrafter, 2017-10-24:
I'm currently having issues with this as well. I'd simply like to turn off the logo LED completely. Is there some option to get the raw (correct) packet to the device?

KazWolfe, 2017-10-24:
As of now, @PaleoCrafter, you're stuck using Synapse. If you're feeling a bit more adventurous, you may be able to use `hidraw` but I would have no idea how to do that effectively.

I know a couple people are investigating this bug, so we will hopefully have a fix pushed out soon.

PaleoCrafter, 2017-10-24:
Great to know that. I've looked into using `hidraw` but it appears a little too risky to me. In my eternal wiseness, I of course completely dumped Windows before installing Arch. I guess it's time to whip up a VM, assuming I can manipulate the settings from there.

KazWolfe, 2017-10-24:
You can, of course, use Virtualbox to send commands to your system. That's how we get packet captures. :)

I would advise you grab another keyboard though. When you bind the Razer device to your VM, the VM *will* take exclusive control over the keyboard.

KazWolfe, 2017-10-24:
Confirmed with Razer Blade Stealth (Late 2017) as well. See above linked issue.

PR #423 (2017-10-24): Fix LED status on certain Razer laptops
- Remove Razer Blade Pro 2017 from inverted LEDs list
- Update code so that laptops use state instead of effect for
  lights.

Should fix #413.

**Note:** Was not tested yet!
closed 2017-10-24 by terrycain