openrazer/openrazer #416: Heatmap feature seems missing.

songandsilence, 2017-10-18:
First used keyboard driver when it was from pez2011. Heatmap was my favorite effect. Seems to be missing. Ubuntu 17.10, Blackwidow Chroma.

Hope I'm not being stupid.

z3ntu, 2017-10-19:
I am not aware of any heatmap but I have joined the project after it was split from the pez2001 repo. Could you explain what you mean?

lah7, 2017-10-19:
Two things come to mind, I think it's the former:

* The original custom effect where pressing keys caused the colours to move to that area... sort of like a heatmap.
* **Or...** a heatmap feature to see how much a key is used, but I don't remember either projects actually implementing this.

z3ntu, 2017-11-23:
@songandsilence bump

songandsilence, 2017-12-19:
I'm talking about the effect, from the pez2001 repo.

terrycain, 2017-12-19:
@songandsilence There's a reimplementation of that effect in a test branch though it's not ready

songandsilence, 2018-01-01:
How can i try it out?

closed 2018-05-31 by stale[bot]