openrazer/openrazer #417: Reversing The 2 Finger Scroll Direction

calebfavor, 2017-10-19:
I have installed all of the open razor driver on my 2017 razor blade stealth (13.3) and everything is working great, except I cannot change the 2 finger scroll direction. The system trackpad option 'Reverse scroll direction' doesn't seem to do anything. I have also tried a couple random solutions I found in this thread, https://askubuntu.com/questions/91426/reverse-two-finger-scroll-direction-natural-scrolling. None of them seem to work. 

Basically I am looking to setup 'natural' 2 finger scrolling, any ideas? I am running Ubuntu 16.04.

Cheers!

KazWolfe, 2017-10-19:
I believe this is a driver-level issue, not an issue with Razer itself.

I may be wrong, but as best as I can tell we can't fix this -- contact the kernel devs.

calebfavor, 2017-10-19:
You were right it is just using the regular synaptics driver. After doing some digging I was able to reverse the scroll direction with this command (you can use any negative value): synclient VertScrollDetla=-28

Thanks!

closed 2017-10-19 by calebfavor