openrazer/openrazer #419: No module named Daemonize

joecatarata, 2017-10-21:
I am using Ubuntu 16.04.3 LTS

Running openrazer-daemon produces an import error:
openrazer-daemon
Traceback (most recent call last):
  File "/usr/bin/openrazer-daemon", line 14, in <module>
    from daemonize import Daemonize
ImportError: No module named 'daemonize'

Any suggestions?

z3ntu, 2017-10-21:
Fixed in fc63ceccf8c468131834c16f09328ed0d44cf3db. Should be included in the next daily build. Thanks for reporting.

z3ntu, 2017-10-21:
Most likely you won't be able to start the daemon with python3-daemonize installed because we need at least version 2.4 but even the newest versions of Ubuntu only includes version 2.3.1. We will have to publish our own updated version of python3-daemonize in the PPA and on OBS.

lah7, 2017-10-21:
`python3-daemonize` (2.4.7) has been uploaded to the daily/stable PPA :+1:

joecatarata, 2017-10-21:
Updated my python3-daemonize to the one in the PPA (2.4.7) albeit I still could not get the daemon to start.
File "/usr/bin/openrazer-daemon", line 153, in <module>
    run()
  File "/usr/bin/openrazer-daemon", line 141, in run
    install_example_config_file()
  File "/usr/bin/openrazer-daemon", line 105, in install_example_config_file
    os.makedirs(os.dirname(CONF_FILE), exist_ok=True)
AttributeError: module 'os' has no attribute 'dirname'

z3ntu, 2017-10-22:
Fixed in 1efabb4. Sorry for not testing the merge properly but that's what the daily branch is for ;)

joecatarata, 2017-10-22:
Thanks! Working all properly now. :)

z3ntu, 2017-10-22:
Great 👍

closed 2017-10-22 by z3ntu