openrazer/openrazer #420: Razer Stealth with Core i7 8th Generation - question

tiry, 2017-10-22:
I just received a Razer Stealth with Core i7 8th Generation and installed Ubuntu 17.10 right away.

I installed `openrazer-meta` using the ubuntu package:

 - install worked fine
 - add to disable secured boot

Keyboard USB ID does not match what is in the Readme

    lsusb | grep 1532
    Bus 001 Device 004: ID 1532:0232 Razer USA, Ltd

May be as a side effect, the `razerkbd` kernel module is not loaded automatically.
Should I use the installation steps provided in the [wiki page](https://github.com/openrazer/openrazer/wiki/Setting-up-the-keyboard-driver) ?

tiry, 2017-10-23:
By the way, I did try to apply the steps provided in the wikipage, but no luck.
I was able to find the device ID and do the unbind, but I am not able to run the bind on the razerkbd 

    root@tiry-Blade-Stealth:/sys/bus/hid/drivers/razerkbd# echo -n "0003:1532:0232.0003" > bind
    bash: echo: write error: No such device

I believe the ID I use is correct:

    root@tiry-Blade-Stealth:/sys/bus/hid/drivers/razerkbd# ll /sys/bus/hid/devices/
    lrwxrwxrwx 1 root root 0 Oct 22 20:39 0003:04F3:2379.0004 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/0003:04F3:2379.0004/
    lrwxrwxrwx 1 root root 0 Oct 22 20:39 0003:1532:0232.0001 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:0232.0001/
    lrwxrwxrwx 1 root root 0 Oct 22 20:39 0003:1532:0232.0002 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:0232.0002/
    lrwxrwxrwx 1 root root 0 Oct 22 20:39 0003:1532:0232.0003 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0232.0003/
    lrwxrwxrwx 1 root root 0 Oct 22 20:39 0018:06CB:8323.0005 -> ../../../devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-1A586753:00/0018:06CB:8323.0005/

KazWolfe, 2017-10-23:
This is a new device. We don't have support for it available just yet, which is why nothing is working. The driver just literally does not know about the device

However, I may have an idea for you. Give me a while to get a patch ready for you to try. You'll need to apply the patch, and compile the driver yourself.

KazWolfe, 2017-10-24:
Please try [this branch](https://github.com/KazWolfe/openrazer/tree/rz-bladestealth-2017) over on my repository. You'll need to compile and build support yourself.

Do this by running the following command from the root folder of the project:

    ./scripts/build_debs.sh

Then, install using:

    sudo dpkg -i dist/*.deb
    sudo apt -f install

Make sure your user is in the `plugdev` group. You *could* play with binding and unbinding, but I'd recommend you just reboot.

***Note:*** I did not test this at all (lack of a device)! You'll be a guinea-pig on bleeding edge software that may or may not do nasty things. Don't blame me if your laptop kills your parrot somehow.

Things like `polychromatic` *should* be able to control your device with this. Please try as many of the lighting effects you can, and let us know what is/isn't working. My guess is that everything but Ripple should work.

tiry, 2017-10-24:
Hi,

Thank you.
I did a build from the repository and branch you provided.

Build is ok and after restart I see that the razerkdb driver is loaded:

    tiry@tiry-Blade-Stealth:~/dev/github/third-party/openrazer$ lsmod | grep razer
    razerkbd               57344  0
    hid                   118784  5 i2c_hid,hid_generic,usbhid,hid_multitouch,razerkbd

However, when I try to run `polychromatic` or `razercommander` both report "no device" found.

I do understand that since you do not have the device to test on your side you are basically blind: please do not hesitate to tell what I can do on my side to provide useful info ...

I try to look at 

    ll /sys/bus/hid/drivers/razerkbd

But there seems to be no bound device.

Trying to do the `unbind` from `hid-generic` and the `bind` in `razerkbd`, did not raise any error this time, however, even after restarting the deamon it still does not work.

Looking at openrazer/logs/razer.log

    2017-10-24 12:19:17 | razer                          | INFO     | Found device.0: 0003:1532:0232.0003
    2017-10-24 12:19:17 | razer                          | CRITICAL | Could not access /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0232.0003/device_type, file is not owned by plugdev

My user is part of the plugdev env.

I naively did a `chown -R root:plugdev` on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0232.0003/

Now both polychromic and razercommander report the deamon to not be running, looking at the logs, not sure what is the issue ...

    2017-10-24 12:28:18 | razer                          | INFO     | Found device.0: 0003:1532:0232.0003
    2017-10-24 12:28:18 | razer.device0                  | INFO     | Initialising device.0 RazerBladeStealthLate2017
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.misc.getSerial method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDriverVersion method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.misc.suspendDevice method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.misc.getRazerUrls method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.misc.getVidPid method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceMode method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.misc.setDeviceMode method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.misc.resumeDevice method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.misc.hasDedicatedMacroKeys method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.misc.getFirmware method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.misc.getMatrixDimensions method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.misc.hasMatrix method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceName method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceType method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.lighting.brightness.getBrightness method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.lighting.brightness.setBrightness method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setWave method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setStatic method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setSpectrum method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setReactive method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setNone method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathRandom method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathSingle method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathDual method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setCustom method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setKeyRow method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.lighting.custom.setRipple method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.lighting.custom.setRippleRandomColour method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.lighting.logo.getLogoActive method to DBus
    2017-10-24 12:28:18 | razer.device0                  | DEBUG    | Adding razer.device.lighting.logo.setLogoActive method to DBus

Running in commandline `openrazer-daemon -Fv` does not give more information, it dumps the same log in the console and exists.

KazWolfe, 2017-10-24:
Woo, progress! We're finding the device. 

I slightly messed up the commit and forgot to add udev rules. Go ahead and pull the latest version of my branch.

You will need to reboot again to make sure that the udev rules take effect, and that your system discovers things.

Also, if Polychromatic/et al are still being annoying, try [manually sending commands](https://github.com/openrazer/openrazer/wiki/Using-the-keyboard-driver).

tiry, 2017-10-24:
Hi,

Driver is now automatically loaded and "bound"

    tiry@tiry-Blade-Stealth:~$ ll /sys/bus/hid/drivers/razerkbd
    ....
    lrwxrwxrwx 1 root root    0 Oct 24 14:01 0003:1532:0232.0001 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:0232.0001/
    lrwxrwxrwx 1 root root    0 Oct 24 14:01 0003:1532:0232.0002 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:0232.0002/
    lrwxrwxrwx 1 root root    0 Oct 24 14:01 0003:1532:0232.0003 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0232.0003/
    --w------- 1 root root 4096 Oct 24 14:01 bind
    lrwxrwxrwx 1 root root    0 Oct 24 14:03 module -> ../../../../module/razerkbd/
    --w------- 1 root root 4096 Oct 24 14:03 new_id
    --w------- 1 root root 4096 Oct 24 14:01 uevent
    --w------- 1 root root 4096 Oct 24 14:03 unbind

Keyboard is now detected as a Razer Tartarus Chroma

Testing the effects:

 - Spectrum: ok
 - wave: ok
 - Recative: ok
 - Breath: ok
 - Ripple: does not do anything

I can change the keys colors.
I can adjust keys brightness.

Change to the "logo" does not seem to produce any result.

Thank you very much for your help: this is great to not have to reboot under windows to change settings!

tiry, 2017-10-24:
Should I close the issue or let you do it when the code is merged in master?

KazWolfe, 2017-10-24:
As a Tartarus? Wut. Is that in your GUI or the raw result of `cat`ting `device_type`?

The logo is the thing on the back of your laptop (the Razer icon that lights up), though I wouldn't be surprised if it's not working. Razer seems to have changed the protocol a bit.

Also, it's good to know Ripple isn't working. We'll look at that next, as that means no software effects are working.

I'll close after we validate the Tartarus issue, and after it's merged in.

tiry, 2017-10-24:
The device name is now correct, thank you again !

tiry, 2017-10-24:
Using polychromatic to change the logo does not have any effect, but it kind of works using command line:

    /sys/bus/hid/drivers/razerkbd/0003:1532:0232.0003$ echo 0 > logo_led_state

 - 0 does not seem to do anything
 - 1 makes the logo "on"
 - 2 makes the logo blink

KazWolfe, 2017-10-24:
Alright, awesome! That's about what I expected. You're running into issue #413.

KazWolfe, 2017-10-24:
Alright, @tiry, can you please test out [this branch](https://github.com/KazWolfe/openrazer/tree/rz-inverted-led-fix) for the LED fix?

You'll have to pull the latest version of the `rz-bladestealth-2017` branch, and then merge that branch on top of it.

tiry, 2017-10-24:
Here is what I did:

    tiry@tiry-Blade-Stealth:~/dev/github/third-party/openrazer$ git branch
       master
     * rz-bladestealth-2017
    tiry@tiry-Blade-Stealth:~/dev/github/third-party/openrazer$ git pull
    tiry@tiry-Blade-Stealth:~/dev/github/third-party/openrazer$ git merge origin/rz-inverted-led-fix

Trying to build the modules fails with

    ...
    :: Compiling OpenRazer kernel modules (Launchpad)
    ========================================
    make -C /lib/modules/4.13.0-16-generic/build SUBDIRS=/tmp/tmp.2N7BJKbDUQ_deb_build_tmp/razer/driver modules > /dev/null 2>&1
    Makefile:60: recipe for target 'lp_driver' failed
    make[2]: *** [lp_driver] Error 2
    ...

Basically, it seems that:

 - `rz-bladestealth-2017` branch build
 - `rz-inverted-led-fix` branch does not build

Let me know if I messed up something ...

KazWolfe, 2017-10-24:
No, that was me. I messed up. Pushed a fix that actually builds now.

tiry, 2017-10-24:
Sorry to ask, but what branch should I now use?

 - `rz-bladestealth-2017` branch does not exist anymore in your repository
 - `rz-inverted-led-fix` branch exists but does not seem to include the fixes you did in the `rz-bladestealth-2017` branch

KazWolfe, 2017-10-24:
Give me a moment. Git is hard.

Most likely, you'll just be able to pull from `master` on the main repo after #424 goes through.