openrazer/openrazer #426: chroma v2 on gentoo linux:  echo -n "int" > matrix_brightness sometimes sets value to zero

tklebanoff, 2017-10-29:
Most of the time, this command (with an int between 0 and 255) works as expected.  Sometimes, however, it sets the value to zero.  
app-misc/razer-drivers-9999

z3ntu, 2017-10-29:
Anything new in the kmsg when that happens?

tklebanoff, 2017-11-01:
No, not that I can tell.  Since this post, however, I've started experiencing some other problems with the system that are calling into question my kernel configuration.  I am working with a new, but customized rt build.  I am closing this for now, because it is not clear to me that the problem is not just something strange on my end.  Once I get the system to a state where everything else is working smoothly, I will reinvestigate this issue and reopen if necessary.

Thanks for your help!

closed 2017-11-01 by tklebanoff