openrazer/openrazer #427: Razer DeathAdder black edition (1532:0029) support

bdmayes, 2017-10-29:
I would love for my mouse to be supported by openrazer! Here is the output from Ubuntu. I'll try to find a Windows system (or setup a VM) where I can gather screenshots of the Synapse options.

```
$ lsusb -d 1532:0029 -v

Bus 003 Device 005: ID 1532:0029 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0029 
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      73
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

$ ls -lah /dev/input/by-id/ | grep "Razer"
lrwxrwxrwx 1 root root   9 Oct 29 09:47 usb-Razer_Razer_DeathAdder-event-mouse -> ../event2
lrwxrwxrwx 1 root root   9 Oct 29 09:47 usb-Razer_Razer_DeathAdder-mouse -> ../mouse0

```

z3ntu, 2017-10-29:
Yes, screenshots from Synapse would be great. And also please send us the product & model number of the device for documentation reasons please (are probably on the underside of the mouse).

bdmayes, 2017-10-30:
Model: RZ01-0015
Product: RZ01-00152400

For packet captures, are you saying I should fire up something like Wireshark on a host system and then capture the packets being sent to the VM as the mouse moves and buttons are pressed?

terrycain, 2017-10-30:
For packet captures, are you saying I should fire up something like Wireshark on a host system and then capture the packets being sent to the VM as the mouse moves and buttons are pressed?

Yep, less so when the mouse moves, more when you click things on synapse.

bdmayes, 2017-10-31:
Well I'm not sure there is much I can do in the form on synapse collection. It appears that this mouse is not supported by synapse. I guess that's because it doesn't have any LEDs on it (which is precisely the reason that I bought it). However, it does have configurable DPI. In fact, the razercfg project supports emulated profiles, frequency switching, and DPI switching. I found the DPI switching to be incredibly handy, so I guess that's all I was really looking for here. In any case, here is there site and github repo:

https://github.com/mbuesch/razer
https://github.com/mbuesch/razer

z3ntu, 2017-10-31:
There should be an application by Razer that allows you to change the DPI from Windows. I've seen  screenshots from some really old looking configuration utilities that were used before Synapse was born. Maybe you can find the one that supports the DA Black Edition.

z3ntu, 2017-10-31:
But actually https://www2.razerzone.com/au-en/gaming-mice/razer-deathadder-black lists "Razer Synapse 2.0 enabled" so it should be supported. Are you sure you passed through the USB device to the VM?

bdmayes, 2017-10-31:
D'oh! You are correct -- I failed to select it in Virtualbox as being passed through. The VM recognized a mouse was present, but obviously didn't know the full details.

Oddly enough, I went here and I do not see this mouse as supported (NOTE: under the "Hardware configurator" section there is a link for "Synapse enabled devices" that opens a modal):

https://www.razerzone.com/synapse#synapse-enabled

Let me collect some info and I'll repost. Thanks for all of your immediate responses! :100:

bdmayes, 2017-11-14:
Sorry -- was out of town for work and I've been trying to get back to this but I'm having a hard time. Virtualbox (or the VM itself?) is behaving very strangely. The mouse works fine until I pass the Razer through to the VM. Then the cursor itself in Windows gets stuck in one spot, even though as I move the mouse, I can see it highlight things and I can click things. Hitting right+CTRL doesn't release the mouse back to the host as it's supposed to, and yes, I have the guest additions installed.

I might have to simply wipe out this system and start over? I'll keep trying and hopefully post something once I have it. In wireshark, am I simply capturing packets from my network interface, or is there some special interface to capture that is related directly to the VM?

z3ntu, 2017-11-14:
Well if you have passed through the USB mouse, then you can't get out of it by RightCtrl as the VM has basically physical access to the USB bus and Linux can't access it anymore. 
Have you tried plugging in a second mouse and does the mouse cursor work then? Because it's actually preferable if you wouldn't move the Razer mouse at all as that would just generate a lot of noise in the packet captures.

bdmayes, 2017-11-30:
Sorry, I'm rather unresponsive about this at the moment. I just fired up wireshark and captured some output, mainly me just clicking around the different buttons for this mouse and then hitting cancel (to leave the default values assigned to them). I see quite a bit of noise from many other things going on with my network.

Would it preferable (or possible) to install wireshark on my VM and then monitor from there? I'm not sure if that will even provide the data you are looking for. Alternatively, is there a specific interface you're looking for data from? I guess it's not clear to me if you're trying to view USB data or if you're trying to see what data gets sent to Razer over the internet as I click around in Synapse.

Let me know what generates the cleanest output and I'll get it shortly.

terrycain, 2017-11-30:
Wiresharking usb traffic going to the vm. What gets sent to razer is of no concern to me, and thats probably all https so would be even more hassle to get.

bdmayes, 2017-12-05:
Ok attached is a gzipped version of the captured traffic. I hope I did this correctly. I basically just listened on all USB interfaces. Also, I went effectively started from the top-left and clicked down through everything. First hitting the dropdown menu for the profiles, closing it, hitting "+", then deleting my new profile, then hitting link and cancel. Then I clicked through buttons 2-7. Hopefully this is what you needed.

[image]

[wireshark-usb-deathadder-black.pcapng.gz](https://github.com/openrazer/openrazer/files/1529523/wireshark-usb-deathadder-black.pcapng.gz)

z3ntu, 2017-12-10:
Key rebinding isn't yet supported with the daemon. But DPI etc should be doable without much work. A screenshot of the "Performance" tab and packet captures of e.g. DPI changing are needed.

bdmayes, 2019-03-16:
Oh wow...I completely forgot about this until recently. Well, I hope this is still useful. I was able to get wireshark to capture usbmon output. Unfortunately both the mouth I was using, as well as the DeathAdder were both on bus 3, so there is probably a bunch of noise just from moving my other mouse around to click on things. I'm not sure if there is a way I can redirect the kernel to pin a device to a different bus. In any case, I tried. I hope this includes what you need. I didn't save any of the information but I click click around and move the sliders up and down.

[image]

[usbmon.pcapng.gz](https://github.com/openrazer/openrazer/files/2974038/usbmon.pcapng.gz)

z3ntu, 2019-07-29:
Hi, as the DeathAdder 3.5G is now supported, I've made a commit to the [`feature_deathadder3.5g_black` branch](https://github.com/openrazer/openrazer/tree/feature_deathadder3.5g_black) which makes this device use the same code as the normal 3.5G. Could you please test that? If you need any help with getting it set up, just ask.

pschloenzke, 2020-04-22:
Hi! I'm chipping in ... I also use this mouse and was happy to find this thread!

I installed openrazer alongside polychromatic on Fedora 31. (I installed kernel-devel before and added the user to plugdev afterwards).
Unfortunately polychromatic does not detect my mouse. I then looked in the openrazer-wiki for "Using the mouse driver" to get it working without a GUI. The article starts with "Inside the folder of the device binded to the driver there will be [...]",  but I don't know, where to find that folder. Could you help me out here?

I also visited the Troubleshooting Guide; here is the output, I hope that helps:
```
$ dmesg
[...]
[    2.692988] usb 3-3: New USB device found, idVendor=1532, idProduct=0029, bcdDevice= 1.00
[    2.692990] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.692992] usb 3-3: Product: Razer DeathAdder
[    2.692993] usb 3-3: Manufacturer: Razer
[    2.695195] input: Razer Razer DeathAdder as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/0003:1532:0029.0001/input/input6
[    2.695302] hid-generic 0003:1532:0029.0001: input,hidraw0: USB HID v1.11 Mouse [Razer Razer DeathAdder] on usb-0000:00:14.0-3/input0
[    2.701386] fuse: init (API version 7.31)
```
```
# journalctl -b | grep -i razer_mount
[no output]
```
```
# dkms install openrazer-driver/2.7.0
Module openrazer-driver/2.7.0 already installed on kernel 5.5.17-200.fc31.x86_64/x86_64
```
```
# modprobe razerkbd
[no output]
```
```
$ openrazer-daemon -Fv
Unable to lock on the pidfile.
```
~/.local/share/openrazer/logs/razer.log
[razer.log](https://github.com/openrazer/openrazer/files/4515737/razer.log)

z3ntu, 2020-04-23:
@pschloenzke You can follow https://github.com/openrazer/openrazer/wiki/Building-a-package#fedora and follow the "To build from custom sources, pause here to edit openrazer.spec:" section and use the latest commit from the [`feature_deathadder3.5g_black` branch](https://github.com/openrazer/openrazer/tree/feature_deathadder3.5g_black) (currently `26b93646ffd18f8d31ca5dd8ec8ac05af47078db`). Then when you have the new packages installed, just reboot and hopefully the device should show up.

pschloenzke, 2020-04-23:
Thanks! Building went fine; polychromatic now recognizes my mouse as "Razer DeathAdder 3.5G", but there are some problems:

[image]
[image]

- When I touch the dpi setting it flashes: "There was a problem communicating with the daemon for this device."
The resolution setting goes from 200 to 16000, while this mouse supports a maximum of 3500 dpi. Also I'm not sure if this mouse supports variable dpi settings or is limited to stepped settings like 1800/3500 dpi.
Changes do not take effect.
- When touching the lighting setting the same error message flashes, which makes sense since this mouse is all black :)
- Polling rate setting allows for 125/500/1000 Hz. I checked it with [evhz](https://gitlab.com/iankelling/evhz) -> it works fine!
- There are no settings for the buttons or even macros (which would be luxurious).
- And of course the displayed model and serial number are wrong. On my mouse it says:
S/N: PM1236015B 02938
MODEL NO.: RZ01-0015
PRODUCT NO.: RZ01-00152400

I went through the Troubleshooting Guide again, here's the output:
```
$ dmesg
[...]
[    2.689994] usb 3-3: New USB device found, idVendor=1532, idProduct=0029, bcdDevice= 1.00
[    2.689997] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.689998] usb 3-3: Product: Razer DeathAdder
[    2.690000] usb 3-3: Manufacturer: Razer
[    2.690742] fuse: init (API version 7.31)
[    2.690774] *** VALIDATE fuse ***
[    2.690778] *** VALIDATE fuse ***
[    2.693523] input: Razer Razer DeathAdder as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/0003:1532:0029.0001/input/input6
[    2.696060] hid-generic 0003:1532:0029.0001: input,hidraw0: USB HID v1.11 Mouse [Razer Razer DeathAdder] on usb-0000:00:14.0-3/input0
[...]
[   25.171442] razermouse: loading out-of-tree module taints kernel.
[   25.171538] razermouse: module verification failed: signature and/or required key missing - tainting kernel
[   25.203888] input: Razer Razer DeathAdder as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/0003:1532:0029.0001/input/input12
[   25.204103] razermouse 0003:1532:0029.0001: input,hidraw0: USB HID v1.11 Mouse [Razer Razer DeathAdder] on usb-0000:00:14.0-3/input0
```
```
# journalctl -b | grep -i razer_mount
Apr 23 12:20:21 dora razer_mount[1077]: Driver razermouse
Apr 23 12:20:21 dora razer_mount[1079]: Device_ID 0003:1532:0029.0004
Apr 23 12:20:21 dora razer_mount[1080]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0029.0004/
Apr 23 12:20:21 dora razer_mount[1087]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0029.0004/
```
```
# dkms install openrazer-driver/2.7.0
Module openrazer-driver/2.7.0 already installed on kernel 5.5.17-200.fc31.x86_64/x86_64
```
```
# modprobe razerkbd
[no output]
```
```
$ openrazer-daemon -Fv
Unable to lock on the pidfile.
```
~/.local/share/openrazer/logs/razer.log
[razer.log](https://github.com/openrazer/openrazer/files/4522387/razer.log)

z3ntu, 2020-04-23:
@pschloenzke  So the device doesn't have any LEDs?

pschloenzke, 2020-04-23:
Exactly!
[image]

... regarding the razer.log file: I might have uploaded the logfile before testing all the options in polychromatic. So if there is something missing, that might be the reason - just let me know. I am on a different machine now and cannot upload a new one for some days.

z3ntu, 2020-04-27:
The picture looks more or less like a regular deathadder with the led off :wink: 

Anyways, can you please try the latest dev build of polychromatic (from the `dev-1.0.0` branch)? If your mouse uses the same DPI mechanism as the DeathAdder 3.5G, then it should hopefully work with that.

pschloenzke, 2020-04-27:
I tried, I failed

I followed [these](https://polychromatic.app/download/manual/) Instructions under "System-wide Install":
```
[f@dora ~]$ git clone https://github.com/polychromatic/polychromatic.git
Klone nach 'polychromatic' ...
remote: Enumerating objects: 397, done.
remote: Counting objects: 100% (397/397), done.
remote: Compressing objects: 100% (201/201), done.
remote: Total 5574 (delta 219), reused 311 (delta 160), pack-reused 5177
Empfange Objekte: 100% (5574/5574), 10.60 MiB | 830.00 KiB/s, Fertig.
Löse Unterschiede auf: 100% (3592/3592), Fertig.
[f@dora ~]$ git checkout stable
fatal: Kein Git-Repository (oder irgendeines der Elternverzeichnisse): .git
[f@dora ~]$ cd polychromatic/
[f@dora polychromatic]$ ls
CHANGELOG  docs     locale    polychromatic-controller   README.md
data       install  Makefile  polychromatic-tray-applet  source
debian     LICENSE  man       pylib                      tools
[f@dora polychromatic]$ git checkout stable
Branch 'stable' folgt nun Remote-Branch 'stable' von 'origin'.
Zu neuem Branch 'stable' gewechselt
[f@dora polychromatic]$ meson build

ERROR: Neither directory contains a build file meson.build.
[f@dora polychromatic]$ ninja -C build
ninja: Entering directory `build'
ninja: error: loading 'build.ninja': No such file or directory
```

I then followed the instructions under "Development Mode".
Running ./polychromatic-controller asked for "lessc" to be installed. I ran 'sudo gem install less'. It then asked for "therubyracer" to be installed, which worked after installing "g++". However I couldn't get any further than this:
```
[f@dora ~]$ git clone https://github.com/polychromatic/polychromatic.git -b master
Klone nach 'polychromatic' ...
remote: Enumerating objects: 397, done.
remote: Counting objects: 100% (397/397), done.
remote: Compressing objects: 100% (201/201), done.
remote: Total 5574 (delta 219), reused 311 (delta 160), pack-reused 5177
Empfange Objekte: 100% (5574/5574), 10.60 MiB | 843.00 KiB/s, Fertig.
Löse Unterschiede auf: 100% (3592/3592), Fertig.
[f@dora ~]$ cd polychromatic/
[f@dora polychromatic]$ ls
CHANGELOG  docs     locale    polychromatic-controller   README.md
data       install  Makefile  polychromatic-tray-applet  source
debian     LICENSE  man       pylib                      tools
[f@dora polychromatic]$ ./polychromatic-controller 
/usr/include/c++/9/bits/stl_vector.h:1042: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = v8::Handle<v8::Value>; _Alloc = std::allocator<v8::Handle<v8::Value> >; std::vector<_Tp, _Alloc>::reference = v8::Handle<v8::Value>&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Failed to compile LESS file! Please install 'lessc' and run:
lessc source/data/controller.less data/pages/controller.css
[f@dora polychromatic]$ lessc source/data/controller.less data/pages/controller.css
/usr/include/c++/9/bits/stl_vector.h:1042: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = v8::Handle<v8::Value>; _Alloc = std::allocator<v8::Handle<v8::Value> >; std::vector<_Tp, _Alloc>::reference = v8::Handle<v8::Value>&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Abgebrochen (Speicherabzug geschrieben)
```

I'm sorry for the German system language, I hope that's not a hassle.

z3ntu, 2020-04-29:
> `[f@dora polychromatic]$ git checkout stable`

This should've been `git checkout dev-1.0.0`

> I'm sorry for the German system language, I hope that's not a hassle.

I'm from Austria so it's alright :wink: For temporarily setting the language to English you can do `export LC_ALL=C` (or prepend `LC_ALL=C` before each command)

pschloenzke, 2020-05-03:
Thanks! Und schöne Grüße ;)

I tried it again with the correct branch. In the end polychromatic starts with a Controller Error. 
This time I was on a different machine running Debian 10; here's the commandline output, screenshots and the openrazer logfile:
```
:~$ git clone https://github.com/polychromatic/polychromatic.git
Cloning into 'polychromatic'...
remote: Enumerating objects: 430, done.
remote: Counting objects: 100% (430/430), done.
remote: Compressing objects: 100% (218/218), done.
remote: Total 5607 (delta 239), reused 335 (delta 172), pack-reused 5177
Receiving objects: 100% (5607/5607), 10.60 MiB | 5.14 MiB/s, done.
Resolving deltas: 100% (3612/3612), done.

:~/polychromatic$ git checkout dev-1.0.0
Branch 'dev-1.0.0' set up to track remote branch 'dev-1.0.0' from 'origin'.
Switched to a new branch 'dev-1.0.0'

:~/polychromatic$ meson build
The Meson build system
Version: 0.49.2
Source dir: /home/dirk/polychromatic
Build dir: /home/dirk/polychromatic/build
Build type: native build
Project name: polychromatic
Project version: 0.3.12
Build machine cpu family: x86_64
Build machine cpu: x86_64
WARNING: Project targetting '>=0.40.0' but tried to use feature introduced in '0.46.0': Python Module
Using meson's python ['/usr/bin/python3']
Program sass found: YES (/usr/local/bin/sass)
Build targets in project: 4
WARNING: Project specifies a minimum meson_version '>=0.40.0' but uses features which were added in newer versions:
 * 0.46.0: {'Python Module'}
Found ninja-1.8.2 at /usr/bin/ninja

:~/polychromatic$ ninja -C build
ninja: Entering directory `build'
[1/1] Generating css with a custom command.

:~/polychromatic$ ./polychromatic-controller-dev
Compiling styling... (sass)
Initalising Window...
Initalising WebView...
WebView Initialised
Initialised Window
Page loaded
Version 0.5.0-dev
Getting device list...
Backend Ready
Application Ready. Showing window.
→ Controller: {"backend":"openrazer","uid":0,"request":"open_device"}
Failed to execute request: open_device with data: {'backend': 'openrazer', 'uid': 0, 'request': 'open_device'}
Traceback (most recent call last):
  File ’/home/dirk/polychromatic/pylib/controller.py’, line 75, in parse_request
    requests[request](data)
  File ’/home/dirk/polychromatic/pylib/controller.py’, line 202, in _open_device
    data = middleman.get_device(data[’backend’], data[’uid’])
  File ’/home/dirk/polychromatic/pylib/backends/middleman.py’, line 111, in get_device
    return BACKEND_ID[backend].get_device(uid)
  File ’/home/dirk/polychromatic/pylib/backends/openrazer.py’, line 176, in get_device
    dpi_x = rdevice.dpi[0]
  File ’/usr/lib/python3/dist-packages/openrazer/client/devices/mice.py’, line 65, in dpi
    dpi_x, dpi_y = self._dbus_interfaces[’dpi’].getDPI()
ValueError: not enough values to unpack (expected 2, got 1)

→ Controller: {"request":"quit"}
```
[image]
[image]
[razer.log](https://github.com/openrazer/openrazer/files/4570193/razer.log)

z3ntu, 2020-05-03:
It's indeed broken at the moment, see https://github.com/polychromatic/polychromatic/issues/209#issuecomment-623147553

pschloenzke, 2021-01-25:
Now that [polychromatic/polychromatic#209](https://github.com/polychromatic/polychromatic/issues/209) has been closed, I would like to try it out again.
I figured I would have to use `feature_deathadder3.5g_black` for building openrazer and `master` for building polychromatic. Is that correct?

I tried the above, but building openrazer fails:
```
$ spectool -g -R openrazer.spec 
warning: line 35: It's not recommended to have unversioned Obsoletes: Obsoletes: razer-kernel-modules-dkms
warning: line 55: It's not recommended to have unversioned Obsoletes: Obsoletes: razer-daemon
warning: line 84: It's not recommended to have unversioned Obsoletes: Obsoletes: python3-razer
[...]

$ rpmbuild -bb openrazer.spec
[...]
RPM build errors:
    line 35: It's not recommended to have unversioned Obsoletes: Obsoletes: razer-kernel-modules-dkms
    line 55: It's not recommended to have unversioned Obsoletes: Obsoletes: razer-daemon
    line 84: It's not recommended to have unversioned Obsoletes: Obsoletes: python3-razer
    Directory not found: /home/f/rpmbuild/BUILDROOT/openrazer-meta-2.9.0-1.fc33.x86_64/usr/src/openrazer-driver-2.9.0
```
~/rpmbuild/RPMS/ remained empty.

I attached the full terminal output.
[terminal_output.txt](https://github.com/openrazer/openrazer/files/5865745/terminal_output.txt)

z3ntu, 2021-01-25:
@pschloenzke I've rebased the branch, please try again now. The previous branch state was based on an older release than 2.9.0 but the rpm spec was for 2.9.0 so it failed with this error.

pschloenzke, 2021-01-25:
Okay, now after running `$ ./polychromatic-controller-dev` a Backend Error pops up with the following details:
```
Traceback (most recent call last):
  File ’/home/f/polychromatic/pylib/middleman.py’, line 160, in get_device
    device = module.get_device(uid)
  File ’/home/f/polychromatic/pylib/backends/openrazer.py’, line 208, in get_device
    dpi_x = rdevice.dpi[0]
  File ’/usr/lib/python3.9/site-packages/openrazer/client/devices/mice.py’, line 66, in dpi
    dpi_x, dpi_y = self._dbus_interfaces[’dpi’].getDPI()
ValueError: not enough values to unpack (expected 2, got 1)
```
After clicking [OK], Polychromatic still starts, but is unable to open the device. Terminal output seemed fine.
The dpi setting of my DeathAdder got reset during this process.

z3ntu, 2021-01-25:
@lah7 Could you check if the DeathAdder 3.5G (`razerdeathadder3.5g` fake driver) works fine with latest polychromatic dev?

lah7, 2021-01-25:
Yes, using the fake driver, there are no issues with the latest `master` branches:

[image]

The error above suggests the _real_ driver / D-Bus isn't passing a Y DPI value to the Python library:

```
  File ’/usr/lib/python3.9/site-packages/openrazer/client/devices/mice.py’, line 66, in dpi
    dpi_x, dpi_y = self._dbus_interfaces[’dpi’].getDPI()
ValueError: not enough values to unpack (expected 2, got 1)
```

Perhaps the library should be passing `0` or `None` for the `Y` value since this device only has a single DPI?

z3ntu, 2021-01-26:
Yep that's the case.. https://github.com/openrazer/openrazer/blob/master/driver/razermouse_driver.c#L1293-L1310 
The API design is definitely bad here (but guess who to blame) ^^

pschloenzke, 2021-07-12:
Is this a point where I should try it out again?
Using `feature_deathadder3.5g_black` for building openrazer and `master` for polychromatic?

z3ntu, 2021-07-25:
@pschloenzke If you (or someone else) doesn't, nothing will happen for this issue.

pschloenzke, 2021-08-06:
Okay, unfortunately it did not change.

I built openrazer using [commit c192cd08958c9968fff3a80dfd5f7a558e046078](https://github.com/openrazer/openrazer/commit/c192cd08958c9968fff3a80dfd5f7a558e046078) and polychromatic from master branch (About says: Application 0.6.0-248-g3836b42, Commit 3836b426ac3ea5da15f52aaec2fbf644540ca6eb) on Fedora 34.

As last time a Backend Error pops up:
```
Traceback (most recent call last):
  File ’/home/f/polychromatic/pylib/middleman.py’, line 165, in get_device
    device = module.get_device(uid)
  File ’/home/f/polychromatic/pylib/backends/openrazer.py’, line 216, in get_device
    dpi_x = rdevice.dpi[0]
  File ’/usr/lib/python3.9/site-packages/openrazer/client/devices/mice.py’, line 66, in dpi
    dpi_x, dpi_y = self._dbus_interfaces[’dpi’].getDPI()
ValueError: not enough values to unpack (expected 2, got 1)
```
razer.log
```
2021-08-06 16:25:06 | razer                          | INFO     | Initialising Daemon (v3.0.1). Pid: 3669
2021-08-06 16:25:06 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-08-06 16:25:06 | razer                          | INFO     | Found device.0: 0003:1532:0029.0001
2021-08-06 16:25:06 | razer.device0                  | INFO     | Initialising device.0 RazerDeathAdder3_5G
2021-08-06 16:25:06 | razer                          | INFO     | Serving DBus
```
[openrazer_build.log](https://github.com/openrazer/openrazer/files/6946247/openrazer_build.log)
[polychromatic_build.log](https://github.com/openrazer/openrazer/files/6946248/polychromatic_build.log)

lah7, 2021-08-06:
@pschloenzke Please could you apply this patch to the install https://github.com/lah7/openrazer/commit/b09bfa32872028010e06806065fd68d68f78023c and try again?

    wget https://github.com/lah7/openrazer/commit/b09bfa32872028010e06806065fd68d68f78023c.diff
    sudo patch -i b09bfa32872028010e06806065fd68d68f78023c.diff /usr/lib/python3/site-packages/openrazer/client/devices/mice.py

(Not 100% sure what the path is on Fedora)

This patch _should_ resolve the error, since it seems the DPI retrieved from D-Bus (→ the driver) is only the X value, but the Python library expects X,Y.

pschloenzke, 2021-08-07:
Okay, the path for me was `/usr/lib/python3.9/site-packages/openrazer/client/devices/mice.py`.

The error message changed to:
```
Traceback (most recent call last):
  File ’/home/f/polychromatic/pylib/middleman.py’, line 165, in get_device
    device = module.get_device(uid)
  File ’/home/f/polychromatic/pylib/backends/openrazer.py’, line 216, in get_device
    dpi_x = rdevice.dpi[0]
  File ’/usr/lib/python3.9/site-packages/openrazer/client/devices/mice.py’, line 67, in dpi
    return int(dpi_x), 0
TypeError: int() argument must be a string, a bytes-like object or a number, not ’dbus.Array’
```

lah7, 2021-08-07:
Oops, I made a mistake, sorry. Try again with this patch on top of the previous one: https://github.com/lah7/openrazer/commit/7f5510667ab98c9a5e3c19f86559f162dccf8325
```
wget https://github.com/lah7/openrazer/commit/7f5510667ab98c9a5e3c19f86559f162dccf8325.diff
sudo patch -i 7f5510667ab98c9a5e3c19f86559f162dccf8325.diff /usr/lib/python3.9/site-packages/openrazer/client/devices/mice.py
```

Hopefully it'll be successful this time 🤞🏼

pschloenzke, 2021-08-07:
Now Polychromatic starts without errors and my mouse is recognised as Razer DeathAdder 3.5G.
DPI setting does not work though :(
I am offered 450, 900, 1800 and 3500 dpi, which should be correct. After selecting one of these, the sensitivity does not change and stays on what I believe is 3500 dpi. No errors.
The polling rate setting works fine and changes between 125, 500 and 1000 Hz correctly.

I attached the terminal output after playing around a bit, I do not see anything exciting though. Is there something else I can do or something else you need?

[image]
[polychromatic_terminal_output.txt](https://github.com/openrazer/openrazer/files/6949122/polychromatic_terminal_output.txt)
[razer.log](https://github.com/openrazer/openrazer/files/6949124/razer.log)

lah7, 2021-08-07:
Are there any silent errors by the kernel or driver?

    journalctl -b | grep -i razer_mount
    sudo dmesg | grep razer

It could be a silent error at a daemon/pylib level (like, should it be `450,-1` instead of `450,0`?). You could try a tool like **D-Feet** to inspect the session bus for `org.razer` and try `setDPI` with parameters directly to the driver.

An alternate is to create a Windows VM and [capture pcaps](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) of just changing the DPI  via Razer Synapse in case of the protocol is different. To note, there is a [pcap from a couple of years ago](https://github.com/openrazer/openrazer/issues/427#issuecomment-473545784) for this Black model.

Looking at the Synapse screenshot in https://github.com/openrazer/openrazer/issues/427#issuecomment-473545784, is this model supposed to accept **any** DPI between 100 and 3500? From what I know, a device like DeathAdder 3.5G only accepted specific values and only the X axis, hence why Polychromatic presents a dropdown and not a slider.

pschloenzke, 2021-08-07:
```
$ journalctl -b | grep -i razer_mount
Aug 07 19:49:35 dora razer_mount[1126]: Driver razermouse
Aug 07 19:49:35 dora razer_mount[1128]: Device_ID 3-2
Aug 07 19:49:35 dora razer_mount[1130]: Modprobing razermouse
Aug 07 19:49:35 dora razer_mount[1177]: Modprobed razermouse
Aug 07 19:49:35 dora razer_mount[1181]: Driver razermouse
Aug 07 19:49:35 dora razer_mount[1182]: Device_ID 3-2:1.0
Aug 07 19:49:35 dora razer_mount[1185]: Driver razermouse
Aug 07 19:49:35 dora razer_mount[1187]: Device_ID 0003:1532:0029.0001
Aug 07 19:49:35 dora razer_mount[1202]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0029.0001/
Aug 07 19:49:35 dora razer_mount[1206]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0029.0001/
```

```
$ sudo dmesg | grep razer
[   21.311171] razermouse: loading out-of-tree module taints kernel.
[   21.311260] razermouse: module verification failed: signature and/or required key missing - tainting kernel
[   21.340103] razermouse 0003:1532:0029.0001: input,hidraw0: USB HID v1.11 Mouse [Razer Razer DeathAdder] on usb-0000:00:14.0-2/input0
[   84.559121] razermouse: DPI requires 2 bytes
[   86.300665] razermouse: DPI requires 2 bytes
[   88.092586] razermouse: DPI requires 2 bytes
[   90.609142] razermouse: DPI requires 2 bytes
[   93.015793] razermouse: DPI requires 2 bytes
[   94.844502] razermouse: DPI requires 2 bytes
[   96.733699] razermouse: DPI requires 2 bytes
[   98.630620] razermouse: DPI requires 2 bytes
[  100.658432] razermouse: DPI requires 2 bytes
```
**D-Feet**: Like so?
[image]
I tried out different set-ups like `450,1` `450,450` `-1,1800` and so on. I did not notice any change in mouse sensitivity for any combination. If there is a negative number in it, the method output will be like in the screenshot. Otherwise it says `This method did not return anything`.
`availableDPI()` returns `[450, 900, 1800, 3500]`
`getDPI()` returns `[3500]`
`maxDPI()` returns `3500`

I had used this mouse in the past on Windows Xp and Windows 7 with the pre-Synapse driver. I am quite sure there were fixed values like 1800, 3500 to select for the dpi setting. But there was also some advanced setting, where one could change the sensitivity for x- and y-axis independently on a scale from 1 to 10 or so ... I will look into it.

z3ntu, 2021-10-01:
@pschloenzke Thanks, indeed possible that Windows 10 does the same in software. This device uses a legacy protocol, so you could hack this device next to some of the other legacy protocol devices like Orochi 2011 and some others (you'll probably see it in razermouse_driver.c).

codefiles, 2022-07-25:
@bdmayes @pschloenzke @kierun
Can any of you test #1859?

codefiles, 2022-07-27:
Excellent job testing and reporting back.

DPI is initialized to `1800,1800` which appears to be causing the daemon to fail on start since DPI X only devices need a Y value of 0.
https://github.com/openrazer/openrazer/blob/4ecf7c1650ea171b226d3fe4a16a2d8d8f7a3a9f/daemon/openrazer_daemon/hardware/device_base.py#L96

Patch the file given above using the same process as you have done before in this issue with the commit https://github.com/codefiles/openrazer/commit/083051f242ebda51511bf396b3541657fb03deaa and report back.

codefiles, 2022-07-28:
Poll rate initialization fails due to successive DBus calls when it follows DPI initialization. Patch the file `/usr/lib/python3.10/site-packages/openrazer_daemon/hardware/device_base.py` with the commit https://github.com/codefiles/openrazer/commit/ef336b957a428a3e8f05fedefa83b913d9b298b8.

bdmayes, 2022-07-29:
> @bdmayes @pschloenzke @kierun Can any of you test #1859?

Apologies. I am not going to be able to offer any further assistance with this. I reported this nearly 5 years ago. I have since switched to a deathadder v2 pro, which is already perfectly supported by openrazer. I'm not even sure what I did with my old mouse. I may have simply donated it. :upside_down_face: 

Thank you for forging ahead with this, however!

codefiles, 2022-07-29:
I suspected that may be the case.

It is possible that the mouse you currently have is also impacted by the issue that was last discussed here and that #1864 addresses. Would you please check for your device if poll rate is being properly initialization when the daemon starts? It would be much appreciated! What I'm looking for is pretty simple. You would use [evhz](https://gitlab.com/iankelling/evhz) after a reboot to see if the poll rate is the value it was last set to. Then change the poll rate to a different value, reboot, and check again.

PR #1859 (2022-07-31): Add support for the DeathAdder 3.5G Black
Closes #427

Requires #1863 and #1864. I'm not sure I followed proper etiquette by making those separate pull requests. My thinking is they address issues that may impact other devices and should have individual reviews.
closed 2022-07-31 by z3ntu