openrazer/openrazer #428: Not following sysfs rules

ukyg9e5r6k7gubiekd6, 2017-10-30:
Thanks for this driver!

However the sysfs files exposed by the driver break a lot of the kernel's rules for sysfs files, as documented in Documentation/filesystems/sysfs.txt or following existing kernel customs, especially:

- One value per file
- ASCII text files (not eg big-endian unsigned shorts)

Also, with the mouse X and Y DPI files, userspace writes binary data, but reads back out ASCII text in the format %d:%d, which is an asymmetry that really bugs me for some reason.

Also, it would be much more concise, and less racy, to use attribute groups, as detailed here:

http://kroah.com/log/blog/2013/06/26/how-to-create-a-sysfs-file-correctly/

Would you accept patches to fix these problems?

Would I need to maintain ABI compatibility (I think so)? I suppose I'd have to only add new files, so as to avoid changing any existing ones. But in many cases, the obvious good names are already taken, so I'm not sure what I'd call the new files.

terrycain, 2017-10-30:
I agree :smile: but were slowly undergoing moving most of the functionality in the driver to userspace so I feel it will be mostly a wasted effort.

z3ntu, 2017-10-30:
And accepting patches: yes, if they improve the code ;)
And of course the daemon has to be able to understand the format you change the driver to.

ukyg9e5r6k7gubiekd6, 2017-11-13:
Well, if you're intending to contribute the kernel-mode parts upstream to kernel.org, then some of these cleanups will need to be done at some stage, in order for the code to be accepted into mainline.

Otherwise, if they're not going to be merged into mainline Linux, then I think you guys should deprecate them now, so people like me don't waste time on them, and so you can begin the process of eventually deleting them.

Whatever you do, you don't want to be carrying around out-of-kernel-tree drivers for ever more.

ukyg9e5r6k7gubiekd6, 2017-11-13:
The most unfortunate, but necessary, change will be to convert all of the spaces to tabs, which will basically introduce an epoch into the version control history - a point in time which it will be difficult to look earlier than.

terrycain, 2017-11-13:
Yeah, I've foreseen this. But hopefully when we've moved most if not all of the effects part into userspace, a few of the drivers should be made completely redundant and whats left should be about a page of code so I don't think it'll be much of a problem, just were a while off that yet

z3ntu, 2017-11-13:
We won't be upstreaming our driver due to various reasons discussed in other issues. And spaces->tabs is least of our problem right now.

closed 2018-04-12 by stale[bot]