openrazer/openrazer #429: Failed Install on Arch

bcmmbc122333, 2017-10-30:
Hey, for some reason I cant seem to get openrazer to install correctly. I have tried using yaourt, and installing it manually and nothing works. I cant seem to get the dependencies to work, especially python-openrazer. Any suggestions?

z3ntu, 2017-10-30:
Please post the console output from yaourt

bcmmbc122333, 2017-10-30:
==> Downloading openrazer-meta PKGBUILD from AUR...
x .SRCINFO
x PKGBUILD
x openrazer-daemon.install
x openrazer-driver-arch.install
x openrazer-driver-dkms.install
Aryvandaar commented on 2017-03-12 19:56			 
I had the same dependency problem. If I recall correctly I installed python-razer before I installed the meta package.

GabMus commented on 2017-05-05 23:00			 
I approve of the merge request into the package razer-drivers.

timourf commented on 2017-09-01 17:11			 
It looks like the source file (v1.1.15.tar.gz) is failing to validate

PhantomWhiskers commented on 2017-09-20 01:51			 
Latest update won't update/install with pacaur, giving me the message `fatal: unable to access 'https://aur.archlinux.org/razer-drivers.git/': The requested URL returned error: 504`

Polychromatic will now give me the message "Daemon Not Installed" after attempting to update this package with `pacaur -Syu` and `pacaur -S openrazer-meta`

z3ntu commented on 2017-09-20 14:55			 
@PhantomWhiskers: You have to update openrazer-meta (+dependencies) to version 2.0.0-1 and polychromatic to 0.3.11.1-1 because there were some breaking changes.

About the 504 error: That's caused by the AUR and not by me...

openrazer-meta 2.0.0-1  (2017-10-21 11:25)
( Unsupported package: Potentially dangerous ! )
==> Edit PKGBUILD ? [Y/n] ("A" to abort)
==> ------------------------------------
==> n

==> WARNING: This PKGBUILD describes a split package.
==> Specific package options are unknown
==> razer-drivers dependencies:
 - make (already installed) [makedepend]
 - python (already installed) [makedepend]
 - python-setuptools (already installed) [makedepend]

==> Continue building openrazer-meta ? [Y/n]
==> ----------------------------------------
==> y

==> Building and installing package
==> Making package: razer-drivers 2.0.0-1 (Mon Oct 30 10:13:39 CDT 2017)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Downloading v2.0.0.tar.gz...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   127    0   127    0     0    127      0 --:--:-- --:--:-- --:--:--   335
100  162k    0  162k    0     0   162k      0 --:--:-- --:--:-- --:--:--  425k
==> Validating source files with sha256sums...
    v2.0.0.tar.gz ... Passed
==> Extracting sources...
  -> Extracting v2.0.0.tar.gz with bsdtar
==> Entering fakeroot environment...
==> Starting package_python-openrazer()...

:: Installing OpenRazer python library
=====================================================
python3 setup.py install --prefix=/usr --root=/tmp/yaourt-tmp-bcm/aur-openrazer-meta/pkg/python-openrazer
running install
running build
running build_py
creating build
creating build/lib
creating build/lib/openrazer
copying openrazer/__init__.py -> build/lib/openrazer
creating build/lib/openrazer/client
[... 339 lines cut ...]
(1/2) checking package integrity                   [######################] 100%
(1/2) loading package files                        [######################] 100%
(2/2) checking for file conflicts                  [######################] 100%
(2/2) checking available disk space                [######################] 100%
:: Processing package changes...
(1/2) installing dkms                              [######################] 100%
Optional dependencies for dkms
    linux-headers: build modules against the Arch kernel [installed]
    linux-lts-headers: build modules against the LTS kernel
    linux-zen-headers: build modules against the ZEN kernel
    linux-hardened-headers: build modules against the HARDENED kernel
(2/2) installing openrazer-driver-dkms             [######################] 100%
********************************************
* To complete installation, please run:    *
* # sudo gpasswd -a <yourUsername> plugdev *
*                                          *
* Please note, that you have to have the   *
* kernel headers for your current kernel   *
* installed for dkms to work correctly!    *
********************************************

:: Running post-transaction hooks...
(1/2) Install DKMS modules
==> dkms install razer_chroma_driver/1.0.0 -k 4.13.9-1-ARCH
(2/2) Arming ConditionNeedsUpdate...

z3ntu, 2017-10-30:
@simonszu yaourt is still broken, so the warning on the website will be re-added

@bcmmbc122333 Please use pacaur, or another AUR helper, because yaourt just can't handle our package for some reason...

bcmmbc122333, 2017-10-30:
@z3ntu thanks for letting me know, ill retry it with pacaur.

z3ntu, 2017-11-18:
Closing as there was no response > 1 week. If you still have the same problem, just comment here.

closed 2017-11-18 by z3ntu

neverfox, 2018-03-11:
Fails in the same with with pikaur.

z3ntu, 2018-03-11:
@neverfox The pikaur can't properly resolve dependencies. pacaur can handle it so use that for now (and report the issue upstream (with pikaur) please, if there's not already an issue for dependency resolution).

simonszu, 2018-03-12:
I am still not sure about the whole "yaourt is failing" thing. I use openrazer since march 2017 and yaourt since february 2017, and everything works well.

z3ntu, 2018-03-12:
@simonszu You could try uninstalling all openrazer packages and python-notify2 and then trying to install `openrazer-meta` with it.

ei-ke, 2018-05-06:
Also fails with trizen. aurman does handle your package without problems.

c1Koriginal, 2020-10-03:
pacman fails, yaourt fails, yay fails, and pretty much all of the other ones too. At this point it's just funny. I have my python-notify2 installed with pip and it works perfectly in code and anywhere else, however for some reason when installing openrazer specifically, it's just not there.
I think I know why it's just doesn't work. Because python-notify2 was removed from AUR, therefore it will always be getting error no matter which package manager you using. Please, fix this or find a way around it. 
https://aur.archlinux.org/pkgbase/python-notify2/

z3ntu, 2020-10-04:
@c1Koriginal python-notify2 was moved to the [extra] repo so you don't have to manually build that package anymore. And also please don't comment on years old closed issues, please open a new one if you still have problems.