openrazer/openrazer #430: Group membership ID error

r3pek, 2017-11-04:
Right now openrazer-daemon checks for group membership for the user running the daemon by parsing the file /etc/group. This is wrong and should not be done since there are other ways to specify group membership for a user (ldap, sssd, etc), in which case, the group membership isn't correctly identified.
(I personally use pam_sssd for group membership and I was already in the plugdev group, but the daemon kept saying that I wasn't member)

terrycain, 2017-11-04:
Calm down, it uses the built in module to check that. Wether that is right or wrong is here not there. You could of just asked can the group membership lookup be expanded, rather than typing a whole paragraph

r3pek, 2017-11-04:
I normally tend to explain the issue and the usecase, not just the problem itself. :)

z3ntu, 2017-11-04:
@r3pek 
1. I don't find any mention of `/etc/group` in our repo.
2. I am not at all familiar with LDAP, SSSD etc
3. If you can provide us with another way / better way of checking group membership than what we currently have, feel free.

r3pek, 2017-11-06:
Hi @z3ntu 
1. True. gr_mem just returns members from the file /etc/group (nothing else)
2. No need to ;)
3. change https://github.com/openrazer/openrazer/blob/master/daemon/openrazer_daemon/daemon.py#L183-L184 to
`return grp.getgrnam('plugdev').gr_gid in os.getgroups()`

That should do it (totally untested)

z3ntu, 2017-11-23:
Please try https://github.com/z3ntu/openrazer/commit/c708d285630e16c1e77dcb61ea8075569d307987

r3pek, 2017-11-23:
@z3ntu works.

PR #467 (2017-12-07): Fix #430: Improve the plugdev group check
Fixes #430
commit 6f7bdb22 (2017-12-07): Fix #430: Improve the plugdev group check
closed 2017-12-07 by z3ntu