openrazer/openrazer #431: openrazer-daemon fails to start; "org.freedesktop.DBus.Error.ServiceUnknown"

thunderrd, 2017-11-05:
Using the app-misc/openrazer package from vifino's Gentoo overlay, along with polychromatic, I get this message when starting polychromatic:

"org.freedesktop.DBus.Error.ServiceUnknown: The name :1.48 was not provided by any .service files"

And when using a terminal:
```
thunderrd@Q6600:~$ openrazer-daemon -Fv
2017-11-01 21:44:38 | razer                          | INFO     | Initialising Daemon (v2.0.0). Pid: 13092
2017-11-01 21:44:38 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: 

2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderChroma ------------------- (1532:0043)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderElite -------------------- (1532:005C)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerExpert ----------------- (1532:0202)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerFirefly ---------------------------- (1532:0C00)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerImperator -------------------------- (1532:002F)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerKrakenClassic ---------------------- (1532:0501)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wired --------------------- (1532:0024)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wireless ------------------ (1532:0025)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerMambaTE ---------------------------- (1532:0046)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerNaga2014 --------------------------- (1532:0040)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerNagaChroma ------------------------- (1532:0053)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerNagaHex ---------------------------- (1532:0041)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexRed ------------------------- (1532:0036)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexV2 -------------------------- (1532:0050)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerNostromo --------------------------- (1532:0111)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaver -------------------------- (1532:0113)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaverChroma -------------------- (1532:0207)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerOrnata ----------------------------- (1532:021F)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerOrochi2011 ------------------------- (1532:0013)
2017-11-01 21:44:38 | razer                          | DEBUG    | Loaded device specification: RazerOrochi2013 ------------------------- (1532:0039)
[... cut ...]
Traceback (most recent call last):
  File "/usr/lib64/python3.6/site-packages/openrazer_daemon/hardware/device_base.py", line 421, in __del__
    self.close()
  File "/usr/lib64/python3.6/site-packages/openrazer_daemon/hardware/device_base.py", line 334, in close
    if not self._is_closed:
AttributeError: 'RazerBlackWidowUltimate2016' object has no attribute '_is_closed'
thunderrd@Q6600:~$
```

It appears the daemon is not starting at all, and exiting with the above message.  Any advice on how to troubleshoot this?

z3ntu, 2017-11-05:
Can you please try applying the following patch to the file `/usr/lib/python3.6/site-packages/openrazer_daemon/hardware/device_base.py` and restarting the daemon?

```
diff --git a/daemon/openrazer_daemon/hardware/device_base.py b/daemon/openrazer_daemon/hardware/device_base.py
index f6acf70..1771229 100644
--- a/daemon/openrazer_daemon/hardware/device_base.py
+++ b/daemon/openrazer_daemon/hardware/device_base.py
@@ -331,10 +331,13 @@ class RazerDevice(DBusService):
         """
         Close any resources opened by subclasses
         """
-        if not self._is_closed:
-            self._close()
+        try:
+            if not self._is_closed:
+                self._close()
 
-            self._is_closed = True
+                self._is_closed = True
+        except AttributeError:
+            self.logger.info("_is_closed doesn't exist. Ignoring.")
 
     def register_observer(self, observer):
```

thunderrd, 2017-11-05:
After patching:

`thunderrd@Q6600:~$ openrazer-daemon -Fv
2017-11-05 19:43:30 | razer                          | INFO     | Initialising Daemon (v2.0.0). Pid: 4542
2017-11-05 19:43:30 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
thunderrd@Q6600:~$`

This indicates that a pid was created, but there is no matching pid in htop

z3ntu, 2017-11-05:
That doesn't make any sense... It seems that stderr isn't being printed with `openrazer-daemon -Fv`...

thunderrd, 2017-11-05:
Is there no debug flag/switch?

z3ntu, 2017-11-05:
What distro are you using? I checked and there was a regression after commit 8219aa7826e1bc909f07366086653069f36133ea (most likely introduced with the python-daemonize merge...)

thunderrd, 2017-11-05:
Using Gentoo, and the openrazer/polychromatic packages from here:
https://github.com/vifino/vifino-overlay/tree/master/app-misc

I reported this to the overlay maintainer first, and he directed me upstream.

z3ntu, 2017-11-05:
Is it possible to specify a git revision in an ebuild? If yes, please use 8219aa7826e1bc909f07366086653069f36133ea and install it from that revision.

z3ntu, 2017-11-05:
@whot It seems that stderr isn't outputted by python-daemonize. Do you know a way to not just get stdout?

whot, 2017-11-06:
See PR #432 for a quickfix, I've also filed https://github.com/thesharp/daemonize/issues/65

jkmac0, 2017-11-06:
I am also on gentoo but using python3.5. Just upgrading to openrazer from razer-drivers. The bug appears to affect the keyboard RazerBlackWidowUltimate2016 only, if I unplug it the daemon will start with my Taipan. With it plugged in I get the same error as above, after patching device_base.py I get the output below.

2017-11-06 07:51:41 | razer                          | INFO     | Initialising Daemon (v2.0.0). Pid: 2589
2017-11-06 07:51:41 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderChroma ------------------- (1532:0043)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderElite -------------------- (1532:005C)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerExpert ----------------- (1532:0202)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerFirefly ---------------------------- (1532:0C00)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerImperator -------------------------- (1532:002F)
2017-11-06 07:51:41 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
[... cut ...]
2017-11-06 07:51:41 | razer                          | INFO     | Found device.1: 0003:1532:0214.0013
Exception ignored in: <object repr() failed>
Traceback (most recent call last):
  File "/usr/lib64/python3.5/site-packages/openrazer_daemon/hardware/device_base.py", line 424, in __del__
    self.close()
  File "/usr/lib64/python3.5/site-packages/openrazer_daemon/hardware/device_base.py", line 340, in close
    self.logger.info("_is_closed doesn't exist. Ignoring.")
AttributeError: 'RazerBlackWidowUltimate2016' object has no attribute 'logger'

thunderrd, 2017-11-06:
Yeah, that's identical to my OP, with the same KB, but I'm not sure if that's a coincidence or if it's significant.

jkmac0, 2017-11-06:
Try changing line 702 in site-packages/openrazer_daemon/hardware/keyboards.py from

```
super(RazerBlackWidow2016, self).__init__(*args, **kwargs)

```
to

```
super(RazerBlackWidowUltimate2016, self).__init__(*args, **kwargs)
```

works for me

z3ntu, 2017-11-06:
Fixed in https://github.com/openrazer/openrazer/commit/552cb204101181c2a2a1db1e49137d9cf725c54b, sorry

PR #432 (2017-11-09): daemon: if we're running on foreground mode, supply our own logger
daemonize initializes a logger during start(). Any exception in the 'action'
argument will be logged to that logger - which defaults to the syslog.

Inside run_daemon() we set up our own logger for the daemon itself and catch
any exceptions by logging to that. However, if an exception occurs during the
daemon's __init__(), that exception was only caught by daemonize and thus
logged to syslog only.

Avoid this by using another logger in case we're using -F

The proper fix is likely to have a classmethod that returns the logger f [... cut]
commit 809612ea (2017-11-09): daemon: if we're running on foreground mode, supply our own logger

daemonize initializes a logger during start(). Any exception in the 'action'
argument will be logged to that logger - which defaults to the syslog.

Inside run_daemon() we set up our own logger for the daemon itself and catch
any exceptions by logging to that. However, if an exception occurs during the
daemon's __init__(), that exception was only caught by daemonize and thus
logged to syslog only.

Avoid this by using another logger in case we're using -F

The proper fix is likely to have a classmethod that returns the logger  [... cut]
closed 2017-11-09 by terrycain