openrazer/openrazer #434: Packages unavailable on yakkety (Ubuntu 16.10)

oliverseal, 2017-11-06:
The sites say that 16.04 and newer are supported, but the Packages on the ppa for 16.10 are empty. They're available in zesty and and xenial, but no love for 16.10. Is this intentional?

z3ntu, 2017-11-06:
Ubuntu 16.10 is EOL already.

closed 2017-11-06 by z3ntu

terrycain, 2017-11-06:
@lah7 is yakkety still on laumchpad? No harm turning it on if it's there

z3ntu, 2017-11-06:
Am not Luke, but not that I see 
[image]

terrycain, 2017-11-06:
Ahh yeah am on phone otherwise I'd look. Yeah main reason is laumchpad isn't offering us 16.10.

oliverseal, 2017-11-06:
k. Well, at least maybe I'll help someone searching on google.

lah7, 2017-11-06:
16.10 Yakkety is indeed EOL, and Launchpad won't be building for it anymore.

However, you could edit the relevant source file in your `/etc/apt/sources.d/` directory and change "yakkety" to "zesty" or "artful".. You can also do this from the **Software & Updates** program. Either way, you'll be able to install the packages from another (supported) release.