openrazer/openrazer #435: Manjaro 17.0.6 Gellivara - Razer Naga Chroma

ShaolinWarrior, 2017-11-07:
Still new to all of this, so please forgive me if I'm in the wrong spot, or if I'm posting the wrong information. I tried to do as much troubleshooting on my own, and, it may be that I have done everything right, but that the issue is elsewhere unrelated to this project. I'll let you be the judge of that, and hopefully we can get my issue resolved together.

Issue: Installed (what feels like everything), however I can't open the RazerCommander program to modify the side buttons on my Razer Naga Chroma. According to [OpenRazer](https://openrazer.github.io/) and [RazerCommander](https://gabmus.github.io/razerCommander/), after I have installed everything, I should be able to configure my mouse - right? Maybe? Polychromatic and Razergenie both seem to be recognizing my device without issue.
When I say it 'won't open, I mean exactly that. I click the icon, and it acts like it's going to open for just a moment, then it goes away. Below is my system details, installs regarding this, and logs gathered up from other people's questions - hopefully they help.

My setup:

Manjaro 17.0.6 Gellivara
x86_64 Linux 4.9.60-1-MANJARO
KDE 5.39.0 / Plasma 5.11.2
Razer Naga Chroma

Installed headers using:
`pacman -S linux49-headers`

plugdev:
`sudo gpasswd -a meihua plugdev`

Rebooted

Installed:
Polychromatic
Razercommander-git
Razergenie-git

rebooted

Output of:
sudo dkms install razer_chroma_driver/1.0.0
`Module razer_chroma_driver/1.0.0 already installed on kernel 4.9.60-1-MANJARO/x86_64`

Output of:
python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'razer'
```

Output of:
sudo modprobe razermouse && lsmod | grep razer
```
razermouse             65536  0
hid                   114688  5 i2c_hid,razermouse,hid_generic,usbhid,hid_multitouch
usbcore               208896  9 razermouse,uvcvideo,usbhid,ehci_hcd,xhci_pci,rtsx_usb,btusb,xhci_hcd,ehci_pci
```

Output of:
sudo udevadm test /sys/bus/hid/devices/0*:1532:0053.*
```
calling: test
version 235
This program is for debugging only, it does not run any program
specified by a RUN key. It may show incorrect results, because
some values may be different, or not available at a simulation run.

=== trie on-disk ===
tool version:          235
file size:         8988132 bytes
header size             80 bytes
strings            1930108 bytes
nodes              7057944 bytes
Load module index
Found container virtualization none.
timestamp of '/etc/systemd/network' changed
timestamp of '/usr/lib/systemd/network' changed
Ignoring /usr/lib/systemd/network/80-container-host0.network, because it's not a regular file with suffix .link.
Ignoring /usr/lib/systemd/network/80-container-vz.network, because it's not a regular file with suffix .link.
Ignoring /usr/lib/systemd/network/80-container-ve.network, because it's not a regular file with suffix .link.
Parsed configuration file /usr/lib/systemd/network/99-default.link
[... 1199 lines cut ...]
Nov 07 04:25:38 FuguKDE razer_mount[6135]: Driver razermouse
Nov 07 04:25:38 FuguKDE razer_mount[6136]: Device_ID 0003:1532:0053.0009
Nov 07 04:25:38 FuguKDE razer_mount[6137]: Unbinding 0003:1532:0053.000A from hid-generic
Nov 07 04:25:38 FuguKDE razer_mount[6139]: Unbinding 0003:1532:0053.0009 from hid-generic
Nov 07 04:25:38 FuguKDE razer_mount[6142]: Driver razermouse
Nov 07 04:25:38 FuguKDE razer_mount[6143]: Device_ID 0003:1532:0053.000B
Nov 07 04:25:38 FuguKDE razer_mount[6144]: Unbinding 0003:1532:0053.000B from hid-generic
Nov 07 04:25:38 FuguKDE razer_mount[6145]: Binding 0003:1532:0053.000A to razermouse
Nov 07 04:25:38 FuguKDE razer_mount[6146]: Binding 0003:1532:0053.0009 to razermouse
Nov 07 04:25:38 FuguKDE razer_mount[6570]: Binding 0003:1532:0053.000B to razermouse
Nov 07 04:25:38 FuguKDE razer_mount[6659]: Finished binding 0003:1532:0053.000A
Nov 07 04:25:38 FuguKDE razer_mount[6660]: Finished binding 0003:1532:0053.0009
Nov 07 04:25:38 FuguKDE razer_mount[6661]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0053.000A/
Nov 07 04:25:38 FuguKDE razer_mount[6662]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0053.0009/
Nov 07 04:25:38 FuguKDE razer_mount[6665]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0053.000A/
Nov 07 04:25:38 FuguKDE razer_mount[6667]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0053.0009/
Nov 07 04:25:38 FuguKDE razer_mount[6672]: Finished binding 0003:1532:0053.000B
Nov 07 04:25:38 FuguKDE razer_mount[6673]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0053.000B/
Nov 07 04:25:38 FuguKDE razer_mount[6675]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0053.000B/
```

Output of:
sudo ~/.local/share/openrazer/logs/razer.log

`sudo: /home/meihua/.local/share/openrazer/logs/razer.log: command not found`

z3ntu, 2017-11-07:
So you say RazerCommander isn't working, but RazerGenie and Polychromatic works? And have you tried rebooting?

ShaolinWarrior, 2017-11-07:
@z3ntu Yes sir, and to be doubly sure, I have rebooted twice more after submitting the above issue.

ShaolinWarrior, 2017-11-08:
Was looking through the comments on some other posts an ran along these steps which I had not presently done before:

`git clone https://aur.archlinux.org/razer-drivers-git.git`

`cd razer-drivers-git`

`curl https://gist.githubusercontent.com/z3ntu/28cb958ce9a778de0975e54c54b77318/raw/0afe285366c35799924717b3286f460425e69bd6/PKGBUILD.nagachroma -o PKGBUILD`

`makepkg -sr`

^^This last one wasn't working at all initially, was outputting an error:
_install file (razer-driver-dkms-git.install) does not exist or is not a regular file._

Ended up finding this step:
`git reset --hard origin/master`

Which fixed that above issue. I then ran:
`sudo pacman -U python-razer-git-*.tar* razer-daemon-git-*.tar* razer-driver-dkms-git-*.tar*`

Which is presently giving me the following feedback:

```
λ ~/razer-drivers-git/ master* sudo pacman -U python-razer-git-*.tar* razer-daemon-git-*.tar* razer-driver-dkms-git-*.tar*
zsh: no matches found: python-razer-git-*.tar*
```

Also, I performed the following:

`λ ~/ sudo dkms remove razer_chroma_driver/1.0.0 --all`

I edited the dkms.conf file at   /usr/src/razer_chroma_driver-1.0.0/dkms.conf, changed this line
`MAKE="KERNELDIR=/lib/modules/${kernelver}/build make driver"`
to
`MAKE="KERNELDIR=/lib/modules/${kernelver}/build make driver_verbose"`

Then I ran
`sudo dkms install razer_chroma_driver/1.0.0`

Here is the output:

```
λ ~/ sudo dkms install razer_chroma_driver/1.0.0     

Creating symlink /var/lib/dkms/razer_chroma_driver/1.0.0/source ->
                 /usr/src/razer_chroma_driver-1.0.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
KERNELDIR=/lib/modules/4.9.60-1-MANJARO/build make driver_verbose...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.9.60-1-MANJARO (x86_64)
Consult /var/lib/dkms/razer_chroma_driver/1.0.0/build/make.log for more information.
```

Her is the contents of said log file:

```
DKMS make.log for razer_chroma_driver-1.0.0 for kernel 4.9.60-1-MANJARO (x86_64)
Wed Nov  8 10:02:47 EST 2017
make: *** No rule to make target 'driver_verbose'.  Stop.
```

I tried runnig RazerCommander once more, but same issue as previously mentioned.

Below is a screenshot of Polychromatic working on my system, and recognizing my device.

[image]

z3ntu, 2017-11-08:
**Please** do not follow random advices we have given to others over the months! Most problems you have now (eg dkms stuff, package stuff etc) are becaused you ran random stuff. 

As you can see in Polychromatic, our daemon is working absolutely fine. 

For the reason why razerCommander isn't opening, you have to ask @GabMus in his repository [GabMus/razerCommander](https://github.com/GabMus/razerCommander). I'll close this issue now as it's "not our problem".

closed 2017-11-08 by z3ntu