openrazer/openrazer #436: Naga Hex V2 side keys not working while openrazer-damon active

dssouza-ti, 2017-11-08:
Hello,

Whenever the openrazer-damon is active, I can configure DPI, the color scheme, the polling rate... but the 7 numeric side keys doesn't work. If I kill the daemon, the keys start working right away. So I need to start the daemon in order to configure, but have to kill the daemon in order to use the mouse. Don't know if it's something similar to issue #375. I tried to do what is told there, but the scripts didn't work for my mouse, probably because it's a different model. How can I collect more information or logs to help identify what is the problem here?

Regards,

Diego Souza

hyph3n, 2017-11-22:
+1 

The whole reason I bought this mouse was to map shortcuts to the side buttons, but they don't work when the daemon is running (and the device is recognized). At first I thought maybe I needed a GUI tool (along the lines of Polychromatic) to configure the side buttons via the daemon, but haven't found anything that works yet. 

Appreciate any info, or glad to help troubleshoot if necessary!

z3ntu, 2018-01-28:
I guess this is fixed in the latest release. If not please leave a comment.

closed 2018-01-28 by z3ntu